Arduino-IRremote/Arduino-IRremote #654: Attempted a Sanyo Send without success

EasyG0ing1, 2019-10-01:
**Board:** ARDUINO NANO
**Library Version:** 2.2.3
**Protocol:** Sanyo

I have a Cisco brand cable box, and when the codes get decoded, they are identified as Sanyo codes. I've written a rather large sketch that holds almost every button code from two different remotes, so I stored the codes in an array in PROGMEM. 

The problem is, I had to go with RAW on the Cisco device because I can't get the library to send the hex encoded commands... here is how I created the Sanyo send routine in ir_Sanyo.cpp:

**Code Block:**
```c
#define SANYO_BITS                   12
#define SANYO_HDR_MARK	           3500  // seen range 3500
#define SANYO_HDR_SPACE	            950  // seen 950
#define SANYO_ONE_MARK	           2400  // seen 2400
#define SANYO_ZERO_MARK             700  // seen 700
#define SANYO_DOUBLE_SPACE_USECS    800  // usually ssee 713 - not using ticks as get number wrapround
#define SANYO_RPT_LENGTH          45000

#if SEND_SANYO
void  IRsend::sendSanyo (const unsigned long data,  int nbits)
    {
        enableIROut(38);

        mark(SANYO_HDR_MARK);
        space(SANYO_HDR_SPACE);

        for (unsigned long  mask = 1UL << (nbits - 1);  mask;  mask >>= 1) {
            if (data & mask) {
                mark(SANYO_ONE_MARK);
                space(SANYO_HDR_SPACE);
            } else {
                mark(SANYO_ZERO_MARK);
                space(SANYO_HDR_SPACE);
            }
        }
    }

#endif

```

When I send any of the hex codes that I decoded I can see the infrared LED lighting up when I look at it through my cell camera, so I know its doing SOMETHING ... it just has no effect on the receiver.

If I use the RAW codes, it works fine, but they consume a lot of memory which puts me into the warning range when I compile my sketch, so I would really like to use the hex codes instead.

Is there any way of really breaking down the protocol of this Cisco remote somehow so that all of the timing parameters can be properly incorporated into a send command? I understand how IR works CONCEPTUALLY but I'm a little lost with the marks and spaces and the timing of them and how that all works to create an intelligible signal ... but if there was some way to decode the commands from the remote and get all of the specifics of the remote so that I can make a send command that will work ... that would be awesome, I just wouldn't know where to start.

Thank you,

Mike Sims

AnalysIR, 2019-10-01:
I don't recall Cisco boxes using the SANYO protocol (but could be wrong).

It could be possible that you are getting a false detection by IRremote.

As a workaround, You could try using the sketch in our blog post, which shows how to send long signals from FLASH (=save SRAM)
[Sending long AC Signals from Flash with IRremote](https://www.analysir.com/blog/2016/04/11/sending-long-ac-signals-flash-irremote/)

_**TIP:** When opening a new issue users are prompted to provide a list of information, which usually helps in providing better responses._

bengtmartensson, 2019-10-02:
First of all, you should give the exact model of the Cisco brand cable box, Cisco makes several... The reason is that IR signals can often be found on the internet.

Secondly, despite of the common meaning (?), the IRremote library is a very blunt instrument for analyzing IR signals. Likely, the box uses a protocol IRremote does not recognize. So, you can post your raw codes, you can search for codes in the internet, or you can paste the codes into IrScrutinizer (or AnalysIR).

> Is there any way of really breaking down the protocol of this Cisco remote somehow so that all of the timing parameters can be properly incorporated into a send command? I understand how IR works [...]

Try IrScrutinizer.

EasyG0ing1, 2019-10-02:
I've searched the Internet for codes from this cable box. The model number of the cable box is 8640HDC2

And I have downloaded and installed IrScrutinizer but I was not able to figure out how to take the raw codes and have the software do any kind of analysis with them ... should this be possible using the raw numbers in the format below?

Here are just some of the raw codes ... if these aren't Sanyo codes, how would I find out which protocol they use, assuming the irRemote library got it wrong?

`/** 1 PgDn **/	{3300,3350,800,2500,800,2550,800,900,750,2600,750,2600,750,2550,800,2550,750,2600,750,900,750,2600,750,2600,750,900,750,900,750,2600,750,900,750,850,850,850,800,850,800,900,750,2550,800,850,800,900,800}

/** 2 Guide **/	{3250,3350,800,2550,800,2600,750,900,750,2500,850,2600,750,850,800,2600,750,2550,750,950,750,800,850,900,750,900,750,900,750,2600,750,850,800,950,750,2550,750,850,850,900,750,2550,800,2550,750,2550,800}

/** 3 Menu **/	{3250,3350,800,2550,800,2550,750,900,750,2550,800,2600,750,2500,850,900,750,900,750,2600,750,2550,800,2550,800,900,750,900,750,2600,750,850,800,900,750,900,750,2600,750,2600,750,900,750,900,750,900,800}`

EasyG0ing1, 2019-10-02:
I take that back ... I pasted the page up raw code into the software and this is what it gave me.
[image]

EasyG0ing1, 2019-10-02:
Well now i tried a different version of the software and it gave me something useful perhaps
[image]

bengtmartensson, 2019-10-03:
OK, now we are getting somewhere.  https://irdb.globalcache.com/ has a set "Cisco 8640HD, 8642HD, 8650HD, & 8652 HD Cable". So I got that, and imported it into IrScrutinizer. It there decodes as "Panasonic_Old".  Exported as raw for Arduino, [this Gist](https://gist.github.com/bengtmartensson/ef7ed66c64f15f9c779e1b68fbc384d5) (yes,  using PROGMEM). 

Notable is the modulation frequency, given as 58000Hz. This also explains why IrScrutinizer 2.1.0 does not accept it. (You can set the frequency tolerance Options -> Protocol Parameters -> Frequency Tolerance (Hz). Set it to -1 to disable frequency tests. Then you signals will decode, as Panasonic_Old.) 

Back to the original question: sending this format in "cooked" (opposite of raw) format.  Obviously the protocol is not "Sanyo", it is another protocol, that someone decided to call Panasonic_Old. Its IRP form is `{57.6k,833}<1,-1|1,-3>(4,-4,D:5,F:6,~D:5,~F:6,1,-44m)*[D:0..31,F:0..63]`.  You can relatively easily write C code for this. I would suggest  that you try with both 58kHz and 38kHz and see which one works (best). Fortunately, it is easier to send "funny" frequencies than to receive them.

Let us know how this works out, or if something is not understandable.

(If you want to try older IrScrutinizer versions, try the latest 1.*.*, which is 1.4.3.)

EasyG0ing1, 2019-11-06:
As luck would have it, this ended up working for me:

```
const unsigned long PanasonicAddress = 0x4004;

const unsigned long ciscoCmds[] PROGMEM = {
/**  0 PgUp     **/     0X36D924,
/**  1 PgDn     **/	0X37D904,
/**  2 Guid     **/	0X36C127,
/**  3 Menu     **/	0X373918,
/**  4 Info     **/	0X36213B,
/**  5  OK      **/	0X366133,
/**  6  Up      **/	0X36812F,
/**  7 Down     **/	0X37A10B,
/**  8 Left     **/	0X37810F,
/**  9 Right    **/	0X364137,
/**  10  A      **/	0X37E902,
/**  11  B      **/	0X36193C,
/**  12  C      **/	0X37191C,
/**  13  D      **/	0XF7E10,
/**  14 Exit    **/	0X366932,
/**  16 Rew     **/	0X37291A,
/**  17 Play    **/	0X37990C,
/**  18 FFWD    **/	0X36293A,
/**  19 Stop    **/	0X365934,
/**  20 Pau     **/	0X374117,
/**  21 1       **/	0X36113D,
/**  22 2       **/	0X37111D,
/**  23 3       **/	0X36912D,
/**  24 4       **/	0X37910D,
/**  25 5       **/	0X365135,
/**  26 6       **/	0X375115,
/**  27 7       **/	0X36D125,
/**  28 8       **/	0X37D105,
/**  29 9       **/	0X363139,
/**  30 0       **/	0X373119,
/**  31 Rec     **/	0X375914,
/**  32 Last    **/	0X36E123,
/**  33 DVR     **/	0X36C926,
/**  34 Back    **/	0X37C906,
/**  35 DayU    **/	0XFE603,
/**  36 DayD    **/	0XE163D,
/**  37 Iput    **/	0XFF601,
/**  38 Aspt    **/	0X36B928,
/**  39 Fav     **/	0X37F101,
/**  40 Live    **/	0X36B129,
/**  41 Pipon   **/	0X37B908,
/**  42 PPSwp   **/	0X367930,
/**  43 PPMv    **/	0X377910,
/**  44 PPCUp   **/	0X36E922,
/**  45 PPCDn   **/	0X37F900,
/**  46 VSrc    **/	0X376113,
/**  47 Rept    **/     0XFFFFFF};

    void sendCommand(int index) {
        int ciscoSize = (sizeof(ciscoCmds)/sizeof(ciscoCmds[0]));
            if (index < ciscoSize) {
                irsend.sendPanasonic(PanasonicAddress, pgm_read_dword_near(&ciscoCmds[index]));
        }
    }

```

EasyG0ing1, 2019-11-12:
@bengtmartensson 

> OK, now we are getting somewhere. https://irdb.globalcache.com/ has a set "Cisco 8640HD, 8642HD, 8650HD, & 8652 HD Cable". So I got that, and imported it into IrScrutinizer. It there decodes as [...]

This export uses a library called **IrSenderPwm.h** that does not exist in IRremote ... any idea where I can find this library and its dependent libraries? I did a google search and found many different libraries with that name, tried a couple of them without success.

> Notable is the modulation frequency, given as 58000Hz. This also explains why IrScrutinizer 2.1.0 does not accept it. (You can set the frequency tolerance Options -> Protocol Parameters -> Frequency [...]

That worked, thanks!  The info given in version 2 is more detailed than other versions ... which is great if you know what to do with the info.  :)

> Back to the original question: sending this format in "cooked" (opposite of raw) format. Obviously the protocol is not "Sanyo", it is another protocol, that someone decided to call Panasonic_Old. [...]

Herein lies the problem ... I don't even know where to begin to write this code ... the IRP form seems to me to be a good place to start in that process but it's Greek to me ... 

I posted above, some code that was working ... I recall using that code **successfully** not long after I started this ticket, but now, for some reason, it's not working an I'm not sure why ... so I came back here to see if I overlooked something or perhaps to see if I could somehow figure out where to start in terms of writing my own code that can send the signal in "cooked" format as you called it, but I'm at a loss. My wielding of C is spotty at best ... I enjoy Java but learning C has been a relationship that is functional at times ... frustrating at times ... and this project has a good balance of both. I've struggled to understand IR codes and have yet to find a good, clear document that explains how they work without assuming that the reader has at least intermediate knowledge of concepts that I obviously don't have or else I'd be all over it by now.

For example, you refer to this IRP format ... I have no idea how to break that down much less how to convert that format into code that would produce the desired result... and when I look at existing IR sending code, I cannot see the relationship between the code and that format so I'm out here in the desert ... so to speak.

bengtmartensson, 2019-11-12:
> This export uses a library called IrSenderPwm.h that does not exist in IRremote ... any idea where I can find this library and its dependent libraries?

You did not read the comments in the sketch... It supports the libraries IrRemote (the current one), my own Infrared4Arduino, as well as cyborg5's IRLib. Default is Infrared4Arduino. This can be downloaded from the Arduino library manager, using the name "Infrared". To use IrRemote instead, comment out `#define INFRARED4ARDUINO` and "comment-in" `#define IRREMOTE` (as is described in the comments). Infrared4Arduno contains `IrSenderPWM.h`. 

Infrared4Arduino has many advantages over IrRemote, in particular:  it is supported, which hardly can be said about IRremote...

BTW, you should find out the difference between include files, classes, and libraries :-).

> I don't even know where to begin to write this code ... the IRP form seems to me to be a good place to start in that process but it's Greek to me ...

For the time being, I think you can use the sketch and let the Greek remain Greek. Jamas!

EasyG0ing1, 2019-11-13:
@bengtmartensson 

> You did not read the comments in the sketch... 

It's a bad habit ... 

> Infrared4Arduino has many advantages over IrRemote, in particular: it is supported, which hardly can be said about IRremote...

Then I will definitely look it over.
 
> BTW, you should find out the difference between include files, classes, and libraries :-).

Well ... include files are usually libraries, and almost always contain classes ... this much I know. I've even written my own library that provides a more refined, simpler and linguistically friendly way to use non-blocking timers.  I'll be publishing it soon.

> For the time being, I think you can use the sketch and let the Greek remain Greek. Jamas!

I'm not OK with that ... I've been reading up on IRP notation today already ... found a decent document that is ... thorough ...

☺

EasyG0ing1, 2019-11-13:
@bengtmartensson 

I remember now, I looked at your library months ago, and the one thing that kept me from using it, was that you stripped out the ability to flash an LED while a user is pressing a button on an IR remote. In the application that I am almost done developing, having that feedback is important for the user (my 75 year old mother ☺ ) and I couldn't figure out how to add it back into your library.

bengtmartensson, 2019-11-13:
> I looked at your library months ago, and the one thing that kept me from using it, was that you stripped out the ability to flash an LED while a user is pressing a button on an IR remote.

Why don't you just turn on a LED before sending and turn it off when stopped sending? Outside of the library. The "do one thing only" principle... In the sketch

```
    void setup() {
          // ....
          pinMode(LED_BUILTIN, OUTPUT);
    }
    // ....
   void sendRaw(...) {
        // ...
       digitalWrite(LED_BUILTIN, HIGH);
       irsend->sendIrSignal(irSignal, times);
       digitalWrite(LED_BUILTIN, LOW);
        // ...
    }
```
I would go so far as to say that the IRremote way is a good example of bad design.

> include files are usually libraries, and almost always contain classes ... this much I know. 

This issue is not the right place to go into depth here, Let me just comment that as a Java programmer, you know that in Java a (public) class must be in a file with the same name as the class.  IMHO, this is a good principle to carry over to C++/Arduino, even though most Arduinauts do not do it that way. In C++, this means two files per class: one include file that declares the interface to the class (among other things...), and one file with the implementation. In some cases, the latter is left out. 

> I'm not OK with that ... I've been reading up on IRP notation today already ... found a decent document that is ... thorough ...

Again, this is not the right place to go into depth here.  Check out my upcoming Youtube channel "Ultra Infrared" :-). RSN...

closed 2020-07-02 by ArminJo