Arduino-IRremote/Arduino-IRremote #659: Using a string for IRsend.sendNEC(stringname, 32)

knil92, 2019-10-24:
Hey,

So I've been playing around with this library and visual studio and I've created an app so that I can control my home lights, that have an infrared remote, over the internet. It's quite basic, but currently the app receives the command to turn the lights on and then sends a command to the arduino over the serial port, and then there are a lot of if statements to check what code the app has sent, and then I use IRsend.sendNEC("0xFFF40A", 32) and change the hex code for the different hex statements. I was thinking I could send just the hex code from the app, over the serial port and then use something like;

string code == Serial.readString()
IRsend.sendNEC(code, 32) 

but when compiling, it says that IRsend can't use strings. I'm just wondering if there is a way around this, as I've managed to make the app side use a config file so I can set it up for different lights, but with the arduino side being basically hard coded. If I change my lights and they use a different remote, then I'll need to recode the arduino. Or if there isn't a way currently to get around this, would there be a possibility of adding in that functionality. With the if statements the arduino code is quite long, but if I'm able to use a string, I can achieve the same functionality with about 5 lines of code.

AnalysIR, 2019-10-24:
Yes it takes a 32 bit unsigned integer.

either convert the hex string to an integer before calling the send function or just Send & read it in from the app as a binary value over Serial.

knil92, 2019-10-24:
Hey,

Thanks for the really speedy reply, do you have an example of this? Arduino code isn't really my strongest. Or can you point me in the right direction, to some material that might shed some light on it? Also if I was to read it in as a binary value, would I have to convert it back to the hex code before sending? You've answered my question and its like a hydra, two more have popped up lol

AnalysIR, 2019-10-24:
Your query is not really related to issues with the Irremote  library code, more of a generic Arduino question.

I suggest you search online where there are likely 100s if not 1000s of examples to find.

If you try it yourself, you could also post your attempt to the Arduino forum and seek assistance there.

knil92, 2019-10-24:
yeah I thought that would be the response, thanks for the speedy replies. I'll come back here and post an example when I get a working version.

knil92, 2019-10-28:
So this is the solution I'm using. Instead of using the hex codes, I used this website to convert it into a decimal value https://www.rapidtables.com/convert/number/hex-to-decimal.html, and then I used the .toInt function on the Serial.readString(); and this turns it into an integer instead of a string and the irsend.SendNEC likes this better than the string. Here is what I'm using now;

  if (Serial.available()){
      irsend.sendNEC(Serial.readString().toInt(), 32);
  }

marcpouliot, 2020-01-11:
It seems that Sanyo and Mitsubishi do not have irsend. (irsend.Sanyo(code,nbits) implemented.
(I have a sanyo tv... )
thnks

closed 2020-07-04 by ArminJo