Arduino-IRremote/Arduino-IRremote #661: Issue with sending raw signal to Carbon Heater

QkrQudRjs, 2019-10-29:
Hello!

I'm trying to turn on the Carbon Heater with IRremote.
When I try to turn on the SamSung TV or LG Air purifier, it's working,
whlie the heater don't.

This is the Air purifier's result
**Code Block:**
```c
Code      : 81E2817E (32 bits)
Timing[67]: 
     +8850, -4350     + 600, -1600     + 550, - 550     + 550, - 550
     + 550, - 550     + 550, - 550     + 550, - 550     + 550, - 550
     + 550, -1650     + 500, -1650     + 550, -1650     + 550, -1650
     + 550, - 550     + 550, - 550     + 550, - 550     + 550, -1650
     + 550, - 550     + 500, -1700     + 500, - 550     + 550, - 550
     + 550, - 550     + 550, - 550     + 550, - 550     + 550, - 550
     + 550, -1650     + 550, - 550     + 550, -1650     + 550, -1650
     + 500, -1650     + 600, -1600     + 550, -1650     + 550, -1650
     + 550, - 550     + 550
unsigned int  rawData[67] = {8850,4350, 600,1600, 550,550, 550,550, 550,550, 550,550, 550,550, 550,550, 550,1650, 500,1650, 550,1650, 550,1650, 550,550, 550,550, 550,550, 550,1650, 550,550, 500,1700, 500,550, 550,550, 550,550, 550,550, 550,550, 550,550, 550,1650, 550,550, 550,1650, 550,1650, 500,1650, 600,1600, 550,1650, 550,1650, 550,550, 550};  // NEC 81E2817E
unsigned int  data = 0x81E2817E;
```
In this case, both sendNEC or sendRaw  code is working .

And this is the Heater's result.

**Code Block:**
```c
Encoding  : UNKNOWN
Code      : A32AB931 (32 bits)
Timing[23]: 
     +1300, - 350     +1300, - 400     + 450, -1200     +1300, - 350
     +1300, - 400     + 450, -1200     + 450, -1250     + 400, -1250
     + 450, -1200     + 450, -1250     + 450, -1200     +1300
unsigned int  rawData[23] = {1300,350, 1300,400, 450,1200, 1300,350, 1300,400, 450,1200, 450,1250, 400,1250, 450,1200, 450,1250, 450,1200, 1300};  // UNKNOWN A32AB931
```
When I use sendRaw code, it's not working. 

I try so many times, various distance between the heater and the Aduino UNO. 
I want to know what the problem is.

Lastly, I write the code ,sendRawDemo, I used.

Thanks.
**Code Block:**
```c

#include <IRremote.h> 

IRsend irsend;

void setup()
{
  Serial.begin(9600);
  
}
void loop() {
int khz = 38;// 38kHz carrier frequency for the NEC protocol
unsigned int On[] ={1250,400, 1250,400, 450,1250, 1250,400, 1250,400, 450,1250, 400,1250, 450,1200, 450,1250, 400,1250, 450,1200, 1300};  // Power 
 irsend.sendRaw(On,sizeof(On)/sizeof(int),khz);
delay(1000);
}

```

AnalysIR, 2019-10-29:
Some ideas...

- Leave a longer gap say 15 seconds (the heater may not like getting repeated signals evry 1 second. 
- Might be better to trigger off sending 'o' for ON over serial or wire up a switch to action signal sending..
- How do you know the carrier frequency is 38kHz (try 30, 33, 36, 40, 56kHz as well to see if they work)
- It may be very timing sensitive. Try the following sketch from our blog, to rocord with better accuracy.
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

The signal itself looks very proprietary & repetetive, with little info within. 

...post any results, if relevant.

QkrQudRjs, 2019-10-30:
Hi!,
I appreciate your kind reply, but..
 1 . I tried changing the delay to 5,15, but it still hasn't worked.
 2 . It didn't work with trigger sending or using button.
 3. I tried all kHz (including 38kHz), and changed  number of times a signal sent , but it didn't work.
 4 . I read your blog well, but I don't think that blog contents are in the power cord of my heater. because My heater's power code is shorter than the blog.

I wanted to tell you the good news, but I'm sorry that the results aren't coming out well.
Now I'm going to buy another Aduino Uno use cable, 
one cable connects my ir recive, the other to the ir send, and make sure that my sendraw code is transmitting the correct code.
If there's a better way, let me share it.
Thank you.

QkrQudRjs, 2019-11-02:
Hi,
Like I said before, I bought a cable, and I used my recieve and send at the same time.
The result was....no error

I've got a strangely accurate recoding.
I don't think there was a problem with my send chip and code.

So the last thing I want to ask you is this.
is there a case where irremote doesn't work?
If there's a case like that, my heater could be the same, so we need to change the model.

I don't know if it helps, but I'll write down my heater's model name and various information.

Model : SF-CB90R    
Name : SAFIRE STAND HEATER
Peak power output : AC 220V/60Hz,900W
Manufacturer : Cixi Kaile Electric Co.,Ltd
Made by China

AnalysIR, 2019-11-02:
Have you recorded the original signal using the sketch on our blog yet? If so, please post the signal captures here..

I didn't fully understand your last post. 

If you sent a signal from one Arduino to another Arduino, then you will likely get good results.

However, IRremote records signals to +/- 50uSecs accuracy and IR receivers can distort signals by +/- 200 uSecs (wort case). The sketch I linked on our blog will provide a more accurate recording and capture some signals that IRremote will not (using defualt settings)

Unfortunatley, the model number does not help as it is likey not very common with IRremote users. But it is useful to know the model number for future reference.

QkrQudRjs, 2019-11-03:
Hello!
I've got some really good news.
You referred to the sketch again in the last answer, and I try to check your blog again.
I don't think I read the blog carefully. I read the blog again and use your sketch. and then
unsigned int  On2[] ={1228, 452, 1232, 448, 416,  1272, 1232,  448, 1232,  448, 392,  1296, 392,  1296, 392,  1292, 396,  1292, 392,  1296, 388,  1296, 1232,  7208, 1228,  452, 1228,  452, 392,  1296, 1228,  452, 1232,  448, 392,  1296, 388,  1300, 388,  1296, 392,  1296, 388,  1296, 392,  1296, 1228,  7264, 1228,  452, 1228,  452, 392,  1296, 1228,  452, 1228,  452, 392,  1296, 388,  1296, 392,  1296, 388,  1300, 388,  1296, 388,  1300, 1228,  7208, 1228,  452, 1228,  456, 388,  1296, 1228,  452, 1228,  452, 392,  1296, 388,  1300, 388,  1296, 388,  1300, 388,  1296, 388,  1300, 1224,  7264, 1228,  456, 1224,  456, 384,  1300, 1228,  452, 1228,  456, 384,  1300, 388,  1300, 384,  1300, 388,  1300, 384,  1300, 388,  1300, 1224,  7212, 1224,  456, 1224,  456, 388,  1300, 1224,  456, 1224,  456, 388,  1324, 360,  1324, 360,  1328, 360,  1324, 364,  1324, 360,  1324, 1200,  7288, 1200,  460, 1224,  456, 384,  1324, 1200,  456, 1224,  484, 356,  1328, 360,  1324, 360,  1328, 360,  1324, 360,  1328, 360,  1324, 1200,  7236, 1200,  480, 1200,  480, 360,  1328, 1200,  480, 1200,  480, 360,  1324, 360,  1328, 360,  1328, 356,  1328, 360,  1324, 364,  1324, 1208}; 
  
It was different from the previous code, I applied this code to the irsend... and it works!!!

Thank you so much for your help, and I'm glad to be able to continue with my project.

AnalysIR, 2019-11-03:
Great News!

I am not sure why IRremote wouldn't capture this for you other than the length of signal (191 timings).

Here is a screenshot of what the signal looks like.

[image]

It looks like it is repeating many times. So you may be able to just send the first part a small number of times or just once.

PS: If you want to clean up the timings _(for even beter performance)_, try using the following 
560
1120

The longer gaps can be left as is ~@7280

closed 2020-07-04 by ArminJo