Arduino-IRremote/Arduino-IRremote #664: irrecvdumpv2 no unsigned int data found

ahmedlouati, 2019-11-14:
hello there i need help ! i have a smal remote that i want to clone and when i read it code with the
"IRrecvDumpV2" it gives me this result : 

EncEncoding  : UNKNOWN
Code      : E0984BB6 (32 bits)
Timing[23]: 
     +1300, - 350     +1300, - 350     + 500, -1200     +1250, - 400
     +1300, - 350     + 450, -1250     + 450, -1200     + 450, -1200
     +1300, - 400     + 450, -1200     + 450, -1200     + 500
unsigned int  rawData[23] = {1300,350, 1300,350, 500,1200, 1250,400, 1300,350, 450,1250, 450,1200, 450,1200, 1300,400, 450,1200, 450,1200, 500};  // UNKNOWN E0984BB6

and when i re send the same hexa code with the "  irsend.sendNEC(code, 32); " code i  gives me this result :

Encoding  : NEC
Code      : E0984BB6 (32 bits)
Timing[67]: 
     +8900, -4450     + 550, -1650     + 600, -1600     + 600, -1650
     + 550, - 550     + 600, - 550     + 550, - 500     + 600, - 550
     + 600, - 500     + 600, -1600     + 600, - 550     + 600, - 500
     + 600, -1600     + 600, -1650     + 600, - 500     + 550, - 550
     + 600, - 500     + 600, - 550     + 600, -1600     + 600, - 550
     + 550, - 550     + 600, -1600     + 600, - 500     + 600, -1650
     + 600, -1600     + 600, -1600     + 600, - 550     + 600, -1600
     + 600, -1650     + 600, - 500     + 600, -1600     + 600, -1650
     + 550, - 550     + 600
unsigned int  rawData[67] = {8900,4450, 550,1650, 600,1600, 600,1650, 550,550, 600,550, 550,500, 600,550, 600,500, 600,1600, 600,550, 600,500, 600,1600, 600,1650, 600,500, 550,550, 600,500, 600,550, 600,1600, 600,550, 550,550, 600,1600, 600,500, 600,1650, 600,1600, 600,1600, 600,550, 600,1600, 600,1650, 600,500, 600,1600, 600,1650, 550,550, 600};  // NEC E0984BB6
unsigned int  data = 0xE0984BB6;

as u see the Timing diffrent and the rawData too and this made my cloned remote not contoling the actual device please help me here <3

AnalysIR, 2019-11-14:
its not NEC 

The encoding is  flaggedas UNKNOWN. When this occurs I believe the hex value is a 32 bit hash of the signal

so use sendRAW to send the signal.

On AnalysIR it is detected as a 12 bit LASKO AC signal. SO i suspect the capture is good.
_(The name of your AC may differ)_

ahmedlouati, 2019-11-14:
@AnalysIR thank you sir i'll look into it <3

closed 2019-11-14 by ahmedlouati