Arduino-IRremote/Arduino-IRremote #666: SparkFun Pro Micro - IRSend leaves IR LED in on state - Onboard led blinks at defined intervals

noenthu, 2019-12-02:
**Board:** SparkFun Pro Micro 5
**Library Version:** 2.2.3
**Protocol:** Any

**Code Block:**
```c

#include <IRremote.h>

IRsend irsend;

void setup()
{
}

void loop() {
	for (int i = 0; i < 3; i++) {
		irsend.sendSony(0xa90, 12);
		delay(40);
	}
	delay(5000); //5 second delay between each signal burst
}

```

I have followed some of the recommendations when using the Spark Fun Pro Micro (5v/16Hz) board. Including..
- Updated to use timer3
- Updated to use pin 5 for output

What I observe when running above code, the oboard red LED blinks 3 times and then repeats every 5 seconds. The IR LED stays in an "on" state (verified by using cellphone camera)

noenthu, 2019-12-03:
Lucky #666

ArminJo, 2020-07-04:
Did you set the pin before to OUTPUT or INPUT_PULLUP, or connect IR-LED to VCC?

closed 2020-08-03 by ArminJo