Arduino-IRremote/Arduino-IRremote #668: atmega 328 frequency problem/question

nedoskiv, 2019-12-10:
Hello,

I wrote remote control (receiver)  sketch on arduino UNO all works fine, then decided to move whole thing on just an atmega 328p processor using his internal clock , on 8MHz nothing works, on 1MHz few buttons are recognized.
My question is does processor frequency affect stability of irreceiver? If yes, I can use an external escilator, what frequency is  considered stable?

AnalysIR, 2019-12-10:
If possible, use a crystal or resonator @ 16 or 8 or 4 Mhz (higher is better)

Without a crystal, it is possibe to calibrate the internal clock...I would stick with 8Mhz for this.
Do a search on how to calibrate the internal oscillator.

I expect you will see lots of issues @ 1MHz with or without a crystal.

nedoskiv, 2019-12-10:
On other devices when uart is needed i use 18.432MHz and work fine, gonna try it here too and post results

AnalysIR, 2019-12-10:
FYI: IRremote depends on the Arduino timers. e.g. Timer0, Timer1or2, micros() etc

I don't believe 18.432MHz is supported by Arduino for UNO.

**Conclusion**: I would be surprised if it worked without modifications.

nedoskiv, 2019-12-11:
just attached 18.432 MHz external oscilator and everything works fine

AnalysIR, 2019-12-11:
Very interesting result indeed ...  can you post the code you use to record signals and the code you use send signals.(along with the oscillator used)

What BAUD rate are you using for Serial using the "18.432 MHz external oscillator" ?

Have you modified the Arduino Board definitiions or core for that crystal speed. If the UNO is configured for 16Mhz that would represent a ~15% deviation in system clocks. 

Did you use a bootloader to upload the firmware?

It is feasible that IR remote control could work with a 15% inaccuracy, but from memory Serial can only tolerate up to ~+/-3% deviation (usually, depending on rate).

I guess if you recorded a signal with a 15% faster clock and resent it with the same 15% faster clock it would be similar to the way an accurate clock would work (for pulse timings but not carrier freq which is usually internal to the IR Reciver only.). However, serial BAUD rate would be a dfifferent matter. Similarly, sending or receiveng a signal that is 15% out in terms of timings and carrier frequency....could also work out at the cost of reduced performance & range. I can see how some protocols could work but some more challenging protocols woud not work.

**TIP:** In general, users should stick with standard configurations when using IRremote. (e.g. a standard 16MHz crystal or resonator costs <US$0.10)

nedoskiv, 2019-12-11:
well I try it with 16MHz, works fine too
I do not send in my code, only receive. A little bit long, but here it is:

`// custom remote control by v_armenchev@yahoo.com

#define btn1 0xFFA25D
#define btn2 0xFF629D
#define btn3 0xFFE21D
#define btn4 0xFF22DD
#define btn5 0xFF02FD
#define rpt 0xFFFFFFFF

#define volup 0xFF18E7
#define volup1 0xFF5AA5
#define voldown 0xFF4AB5
#define voldown1 0xFF10EF
#define but0 0xFF9867

#define ON HIGH
#define OFF LOW

#define volumelenght 4
#define maxvolume 15
#define minvolume 0
int volume=9; // default volume
const byte  volumepattern[16][4] PROGMEM = {
 {ON,ON,ON,ON},
 {OFF,ON,ON,ON},
 {ON,OFF,ON,ON},
 {OFF,OFF,ON,ON},
 {ON,ON,OFF,ON},
 {OFF,ON,OFF,ON},
 {ON,OFF,OFF,ON},
 {OFF,OFF,OFF,ON},
 {ON,ON,ON,OFF},
 {OFF,ON,ON,OFF},
 {ON,OFF,ON,OFF},
 {OFF,OFF,ON,OFF},
 {ON,ON,OFF,OFF},
 {OFF,ON,OFF,OFF},
 {ON,OFF,OFF,OFF},
 {OFF,OFF,OFF,OFF}       
};
const byte  volumepins[4] PROGMEM = {19,18,17,16};

#define inputlenght 5
int input=0; // default volume
const byte  inputpattern[5][5] PROGMEM = {
 {ON,OFF,OFF,OFF,OFF},
 {OFF,ON,OFF,OFF,OFF},
 {OFF,OFF,ON,OFF,OFF},
 {OFF,OFF,OFF,ON,OFF},
 {OFF,OFF,OFF,OFF,ON}
};
const byte  inputpins[5] PROGMEM = {13,12,11,10,9};

const int powerpin=15;
int power=ON;

const int mutepin=14;
int mute;

unsigned long repeatmillis=0;
unsigned long repeatpreviousmillis=0;
unsigned int repeatrate=250;
unsigned long int remotecode;
unsigned long int lastremotecode=0;
#include <IRremote.h>

int RECV_PIN = 4;
[... 226 lines cut ...]
          //Serial.println(power);
          break;          
    }
         
      }
  else
      {
      // repeatpreviousmillis= millis();
      // repeatmillis=repeatpreviousmillis;
      // lastremotecode=0;
      }
  
//return;
 mute=digitalRead(mutepin);
 if (mute==ON)
  {
//  //Serial.println(F("Mute active!"));
  togglevolume(0,0);
  }
  else
   {
   togglevolume(volume,0);
   }
//delay(1000);
}`

AnalysIR, 2019-12-11:
OK  - thanks for that...

I note you have commented out all the Serial stuff.

As you are receiving only, the IR receiver takes care of the IR carrier - so crystal speed doesn't matter for this scenario. 

The signal's pulse timings will be ~15% off (shorter). However, Many IR protocol decoders will handle 15% innacuracies fine (introduced by the wrong crystal), but others will not.

(FYI: The IR receiver can distort pulse lenghts by up to +/-200uSecs in addition)

Nontheless, an interesting result and something to bear in mind for the future. 

However, IRremote users should stick with standard crystal/Arduino settings & configurations to avoid the 'potential' for obscure behaviour.

nedoskiv, 2019-12-11:
it works fine on arduino board, when processor is alone there is no serial, so I comment it.

closed 2020-07-01 by ArminJo