Arduino-IRremote/Arduino-IRremote #669: How to instal the modified version of library too

karansoi, 2019-12-11:
**Board:** ATmega8UNO
**Library Version:** 2.1.0
**Protocol:** NEC(if any)

**Code Block:**
```c

#include <boarddefs.h> 
my sketch required use of Timer2, So this was changed to Timer1 in boardefs.h and only NEC protocol was selected in IRremote.h. My sketch worked. compiled with Minicore in Arduino IDE
I want to maintain two libraries 1 the original IRremote and 
2 The one adapted for ATmega8 How can this be done
Guidance will be of much help
.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2019-12-11:
One way would be make some modifications in your local copy for your own use,  using conditional #defines #if #idef etc.

setting one or more  #define statements before you include IRremote.h in your sketch and then using it in your 'modified library to set the configuration(s) you need,

I havent tested this but it might work for your custom use. ???

karansoi, 2019-12-14:
I made a copy of your library(IRremote) as IR_remote.
On global basis changed all IRr... to IR_r... 
Made necessary changes in IR_remoteInt.h, board_defs.h.
and made the change in library.propertis file name=IR_remote.
Now zipped this folder and installed the library in Arduino IDE as a zip file.
All well.
However when I compile my sketch with this library the size of code nearly doubles and does not compile for Atmega8.
As per my thinking only IRremote.h file needs to be included but include from skethc>include library includes all the .h files. Is there a way of controlling this?

karansoi, 2019-12-15:
The issue may be closed. In IRremote.h All defines for various protocols was set to 0 except NEC. Now there is no memory overflow for Atmega8

closed 2019-12-15 by karansoi