Arduino-IRremote/Arduino-IRremote #673: send not work

DjMarinetti, 2020-01-22:
**Board:** ARDUINO nano
**Library Version:** 2.1.0
**Protocol:** NEC

**Code Block:**
```c

#include <IRremote.h>

IRsend irsend;

void setup()
{

}

void loop() {
  int khz = 38; // 38kHz carrier frequency for the NEC protocol
  unsigned int  irSignal[73] = {8950,4450, 650,1600, 700,550, 650,550, 650,1600, 650,1600, 700,550, 650,1600, 650,550, 650,550, 650,550, 650,550, 650,1600, 700,550, 650,550, 650,500, 700,500, 700,500, 700,500, 650,550, 650,550, 650,550, 650,550, 650,1600, 700,500, 700,500, 650,550, 700,500, 650,550, 650,1600, 700,500, 700,1600, 650,550, 650,550, 650,1600, 700,500, 700};  // NEC 9A10020A
  
  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz); //Note the approach used to automatically calculate the size of the array.

  delay(5000); //In this example, the signal will be repeated every 5 seconds, approximately.
}

dump whit irdumpv2:
unsigned int  irSignal[73] = {8950,4400, 650,1650, 650,500, 700,500, 700,1600, 700,1600, 650,550, 650,1600, 700,500, 700,500, 700,500, 650,550, 650,1650, 650,500, 700,500, 700,500, 700,500, 700,500, 650,550, 650,550, 650,550, 650,550, 650,500, 700,1600, 700,500, 650,550, 650,550, 650,550, 650,550, 650,1600, 700,500, 650,1650, 650,550, 650,550, 650,1600, 700,500, 700};  // NEC 9A10020A
unsigned int  irSignal[73] = {8950,4400, 700,1600, 650,550, 650,550, 650,1600, 700,1600, 650,550, 650,1650, 650,550, 650,500, 700,500, 700,500, 700,1600, 650,550, 650,550, 650,550, 650,500, 700,500, 700,500, 700,500, 700,500, 650,550, 650,550, 650,1600, 700,500, 700,500, 700,500, 700,500, 650,550, 650,1650, 650,500, 700,1600, 700,500, 700,500, 650,1650, 650,500, 700};  // NEC 9A10020A
unsigned int  irSignal[73] = {8950,4450, 650,1600, 700,500, 700,500, 650,1650, 650,1600, 700,500, 700,1600, 650,550, 650,550, 650,550, 650,550, 650,1600, 700,500, 700,500, 650,550, 650,550, 650,550, 650,500, 700,500, 700,500, 700,500, 700,500, 650,1650, 650,550, 650,550, 650,500, 700,500, 700,500, 650,1650, 650,550, 650,1600, 700,500, 700,500, 700,1600, 650,550, 650};  // NEC 9A10020A
unsigned int  irSignal[73] = {8950,4450, 650,1600, 700,500, 700,500, 650,1650, 650,1600, 700,500, 700,1600, 650,550, 650,550, 650,550, 650,550, 650,1600, 700,500, 700,500, 650,550, 650,550, 650,550, 650,550, 650,500, 700,500, 700,500, 700,500, 650,1650, 650,550, 650,550, 650,500, 700,500, 700,500, 700,1600, 650,550, 650,1600, 700,500, 700,500, 700,1600, 650,550, 650};  // NEC 9A10020A
unsigned int  irSignal[73] = {8950,4450, 650,1600, 700,550, 650,550, 650,1600, 650,1600, 700,550, 650,1600, 650,550, 650,550, 650,550, 650,550, 650,1600, 700,550, 650,550, 650,500, 700,500, 700,500, 700,500, 650,550, 650,550, 650,550, 650,550, 650,1600, 700,500, 700,500, 650,550, 700,500, 650,550, 650,1600, 700,500, 700,1600, 650,550, 650,550, 650,1600, 700,500, 700};  // NEC 9A10020A

DUMP WITH: https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

Press the button on the remote now - once only

Raw: (279) 9016, -4464, 676, -1656, 652, -556, 648, -556, 644, -560, 652, -1656, 644, -564, 648, -1660, 652, -552, 648, -560, 644, -560, 652, -556, 644, -1664, 648, -556, 656, -528, 672, -560, 652, -552, 652, -556, 644, -536, 676, -556, 648, -556, 656, -552, 648, -1660, 652, -552, 648, -560, 652, -552, 652, -552, 648, -560, 652, -556, 676, -1604, 676, -556, 676, -1632, 648, -556, 644, -564, 648, -1632, 672, -536, 672, -19964, 676, -1632, 676, -528, 676, -532, 680, -524, 676, -532, 672, -532, 680, -528, 672, -532, 680, -528, 672, -1636, 676, -528, 672, -536, 676, -528, 676, -532, 680, -1628, 672, -532, 680, -528, 672, -536, 676, -528, 676, -532, 668, -536, 676, -532, 672, -532, 680, -528, 672, -532, 680, -528, 676, -528, 672, -536, 676, -1628, 704, -1608, 700, -1608, 672, -532, 680, -45968, 9012, -4464, 676, -1632, 680, -552, 648, -560, 652, -1628, 672, -1636, 676, -556, 656, -1624, 676, -556, 656, -548, 652, -556, 648, -556, 656, -1628, 672, -560, 652, -556, 644, -560, 652, -556, 648, -556, 656, -552, 648, -556, 648, -560, 652, -556, 644, -1636, 676, -1632, 676, -556, 648, -556, 656, -552, 648, -560, 644, -560, 652, -1632, 680, -552, 648, -1632, 680, -552, 648, -560, 652, -1628, 672, -560, 652, -19956, 680, -1628, 672, -560, 652, -556, 648, -556, 656, -552, 648, -556, 648, -560, 652, -552, 648, -564, 648, -1628, 684, -552, 648, -556, 648, -556, 656, -552, 648, -1632, 680, -556, 644, -560, 652, -556, 648, -556, 656, -548, 652, -556, 648, -560, 652, -552, 648, -560, 652, -552, 652, -552, 648, -560, 652, -556, 648, -1632, 676, -1632, 680, -1628, 672, -1632, 708, 

Press the button on the remote now - once only

Raw: (279) 9012, -4488, 672, -1636, 644, -588, 624, -552, 648, -584, 620, -1688, 620, -588, 624, -1684, 628, -552, 652, -580, 620, -588, 624, -580, 620, -1660, 652, -580, 620, -588, 624, -584, 620, -560, 652, -576, 624, -584, 620, -584, 628, -552, 648, -588, 624, -1680, 620, -588, 624, -584, 620, -560, 652, -552, 648, -584, 620, -588, 624, -1684, 624, -556, 648, -1684, 624, -584, 620, -560, 652, -1656, 644, -560, 652, -19984, 672, -1612, 668, -564, 648, -556, 644, -560, 652, -556, 648, -560, 672, -532, 648, -560, 644, -560, 652, -1656, 652, -528, 696, -536, 644, -560, 652, -556, 648, -1656, 652, -556, 680, -524, 656, -524, 676, -532, 672, -560, 652, -552, 648, -560, 644, -560, 652, -556, 656, -552, 648, -556, 648, -560, 652, -552, 648, -1660, 652, -1656, 652, -1656, 644, -560, 652, -9956, 9016, -4460, 676, -1632, 680, -552, 648, -560, 652, -1628, 672, -1636, 676, -556, 656, -1624, 676, -560, 652, -552, 648, -560, 656, -548, 652, -1656, 652, -556, 648, -556, 656, -552, 652, -552, 648, -560, 652, -552, 648, -560, 656, -548, 652, -556, 644, -1636, 676, -1632, 680, -552, 648, -560, 652, -552, 652, -552, 648, -560, 652, -1628, 684, -552, 648, -1632, 680, -552, 648, -556, 648, -1632, 676, -560, 644, -19964, 672, -1636, 676, -556, 644, -564, 648, -556, 648, -560, 652, -552, 648, -560, 644, -560, 652, -556, 644, -1636, 676, -556, 644, -560, 656, -552, 648, -556, 644, -1636, 676, -560, 652, -552, 652, -556, 644, -560, 652, -556, 644, -564, 648, -556, 648, -560, 652, -552, 652, -556, 656, -552, 648, -556, 648, -560, 652, -1628, 672, -1636, 672, -1632, 680, -1628, 684, 

Press the button on the remote now - once only

Raw: (279) 9008, -4472, 668, -1636, 676, -532, 680, -524, 676, -532, 672, -1636, 672, -532, 672, -1636, 704, -504, 676, -528, 676, -532, 668, -536, 676, -1632, 680, -528, 672, -532, 672, -536, 676, -528, 672, -536, 676, -528, 704, -504, 680, -528, 704, -500, 680, -1628, 672, -536, 676, -528, 676, -532, 680, -524, 676, -532, 672, -536, 676, -1628, 680, -528, 676, -1632, 700, -504, 676, -532, 668, -1636, 708, -500, 700, -19936, 672, -1636, 672, -532, 672, -536, 676, -528, 672, -532, 680, -528, 676, -532, 680, -524, 676, -532, 704, -1604, 672, -536, 676, -528, 676, -532, 680, -524, 676, -1632, 680, -524, 676, -532, 680, -528, 676, -528, 704, -504, 696, -508, 676, -532, 680, -524, 676, -532, 700, -508, 676, -528, 672, -536, 676, -528, 676, -1632, 676, -1632, 732, -1576, 672, -532, 680, -54784, 9040, -4460, 680, -1628, 684, -552, 648, -532, 672, -1636, 672, -1632, 680, -528, 672, -1636, 676, -528, 672, -536, 676, -532, 672, -532, 680, -1628, 672, -532, 680, -528, 676, -528, 672, -536, 676, -556, 648, -556, 656, -552, 648, -560, 652, -552, 652, -1628, 680, -1628, 672, -560, 652, -556, 648, -556, 644, -564, 648, -556, 648, -1636, 676, -556, 644, -1636, 676, -556, 656, -552, 648, -1632, 680, -552, 648, -19964, 676, -1628, 684, -548, 652, -556, 648, -556, 656, -552, 648, -556, 648, -556, 656, -552, 648, -556, 656, -1628, 672, -560, 652, -552, 652, -556, 644, -560, 652, -1632, 680, -552, 648, -556, 648, -560, 652, -552, 648, -560, 652, -552, 652, -556, 644, -564, 648, -556, 648, -560, 652, -552, 652, -556, 656, -552, 648, -1628, 684, -1624, 676, -1632, 676, -1632, 680, 

Press the button on the remote now - once only

Raw: (279) 8984, -4568, 572, -1736, 576, -576, 624, -608, 592, -616, 596, -1712, 600, -580, 624, -1736, 572, -608, 596, -608, 604, -604, 596, -612, 588, -1692, 620, -612, 600, -580, 624, -584, 616, -588, 624, -608, 596, -584, 628, -580, 620, -584, 628, -580, 624, -1680, 620, -612, 600, -580, 620, -588, 624, -580, 624, -584, 680, -528, 620, -1684, 628, -580, 620, -1712, 600, -584, 648, -556, 624, -1684, 648, -532, 648, -20012, 628, -1652, 648, -588, 624, -580, 624, -584, 616, -564, 648, -556, 648, -612, 600, -604, 596, -612, 600, -1656, 644, -588, 624, -584, 620, -584, 628, -5 [... cut ...] 2, -552, 644, -19964, 676, -1632, 680, -552, 648, -560, 652, -528, 676, -556, 656, -552, 648, -556, 648, -560, 652, -552, 648, -1660, 652, -528, 676, -556, 656, -548, 652, -556, 648, -1660, 648, -556, 648, -560, 652, -556, 644, -560, 652, -528, 676, -556, 644, -564, 648, -556, 648, -560, 652, -552, 652, -556, 644, -560, 652, -556, 648, -1656, 652, -1628, 672, -1636, 676, -1632, 680, 

FOR READ USE TSOP34438
FOR EMISOR: SFH4546, MODULE OPEN SMART

SCRIPT WORK ON TV, AC AND DIRECTV BUT NOT AC ANWO

AnalysIR, 2020-01-22:
IRremote as implemented does not handle that AC signal properly. The longer ones with our sketch is much better. So forget the shorter signasl (73) and only use the longer ones(279), captured with the sketch from our blog.

They are detectred as Gree AC 128 bit signals on AnalysIR  (name is not important).

Here is a cleaned signal exported from AnalysIR, which may perform better: 

```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)

Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_0_0[] = {9000,4500,600,1690,600,600,600,600,600,600,600,1690,600,600,600,1690,600,600,600,600,600,600,600,600,600,1690,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1690,600,600,600,600,600,600,600,600,600,600,600,600,600,1690,600,600,600,1690,600,600,600,600,600,1690,600,600,600,19936,600,1690,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1690,600,600,600,600,600,600,600,600,600,1690,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1690,600,1690,600,1690,600,600,600,54784,9000,4500,600,1690,600,600,600,600,600,1690,600,1690,600,600,600,1690,600,600,600,600,600,600,600,600,600,1690,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1690,600,1690,600,600,600,600,600,600,600,600,600,600,600,1690,600,600,600,1690,600,600,600,600,600,1690,600,600,600,19964,600,1690,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1690,600,600,600,600,600,600,600,600,600,1690,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,600,1690,600,1690,600,1690,600,1690,600}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: GREE32_64AC, Key:  
```

TIP: When recording IR signals, hold the remote circa 1->2m away from the receiver and point it directly at the receiver.

DjMarinetti, 2020-01-22:
thanks a lot. work it.
how to i can cleaned the signal??? 

approximate values and positive numbers?

best regards

AnalysIR, 2020-01-23:
I am glad it worked out.

The clean feature is a part of our AnalysIr application (purchase required).

Alternatively, you can manually change each number individually to match the nearest value in the cleaned signal I sent you.

closed 2020-07-01 by ArminJo