Arduino-IRremote/Arduino-IRremote #674: Send raw code as HEX instead of int array

hackwin, 2020-02-01:
**Board:** ARDUINO UNO
**Library Version:** 2.2.3
**Protocol:** None (if any)

I would like to know if it's possible to send the raw code as HEX instead of an array of ints.

For example, "0xE0E040BF" instead of: "rawData[67] = {4500, 4400, 600, 1650, ... }"

**Code Block:**
```c

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [X] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [X] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2020-02-01:
The library supports sending signals in RAW format or as per a **supported** IR  protocol.

Otherwise, you can write you own function to send your custom protocol. Have a look at the NEC (or Samsung) implementation in the library.

We also have several examples of sending signals from HEX values on our [blog](https://www.analysir.com/blog/sitemap_analysir_posts/) & also sending longer RAW signals from FLASH, in case SRAM is in short supply.

maxholgasson, 2020-02-14:
Wondering a similar thing. I only need to transmit 3 bit and therefore a own protocol I guess.
Where do I find your mentioned file and wahts the name of it?

>  Have a look at the NEC (or Samsung) implementation in the library.

AnalysIR, 2020-02-14:
[ see NEC file in IRremote Library here](https://github.com/z3t0/Arduino-IRremote/blob/master/ir_NEC.cpp)

use the sendNEC function. You can specify the number of bits to use. (default is 32). I would just send all 32 bits and only use the 3 bits I needed at the other end.

In other words use any protocol and only ise the 3 bits you want ot use :)

maxholgasson, 2020-02-14:
Thanks! for a battery powered application I would like to keep the transmission as short as possible therefore my question about a short/fast protocol

AnalysIR, 2020-02-14:
OK.....Then just use NEC with the 3 bits. You will make a ~40% saving per signal sent using 3 bits vs 32 bits. _(Using RC6 with 3 bits you may be able to get closer to an 80% reduction per signal for 3 bits vs 32 bits with NEC)_

Note...with only a few bits and without a checksum the chances of false signals being detected increase.

FYI: we recently released a standard 45 button remote control which is 100% Arduino compatible and supports IRremote using the Arduino IDE (KontroLIR). As it is powered by 2xAA Alkaline batteries we looked into power budgets a lot & expect to get 1->2 Years+ on each set of batteries with an idle current of <1uAmp. I suspect getting the low-power/sleep modes optimised is the most important part for typical day-to-day domestic usage.

With IR signals & batteries the key thing is to always run the IR carrier @ 33.3% duty cycle (which IRremote does by default).  There is interesting maths behind this for anyone who wants to research further.

If you have a particularly challenging power budget, then there are other things you can do to extend battery life with IR signal design. It might only be worth the extra effort for space related projects or similar???.

maxholgasson, 2020-02-14:
Your remote control sounds great, I will have a look

I tried to use your example how to create a new protocol and get this error
Otherwise thanks for it's useful documentation!
 `/Users/holga/Documents/Arduino/libraries/IRremote/ir_Template.cpp: In member function 'bool IRrecv::decodeShuzu(decode_results*)':
/Users/holga/Documents/Arduino/libraries/IRremote/ir_Template.cpp:161:23: error: 'SHUZU_BITS' was not declared in this scope
  for (int i = 0;  i < SHUZU_BITS;  i++) {

                       ^
/Users/holga/Documents/Arduino/libraries/IRremote/ir_Shuzu.cpp: In member function 'bool IRrecv::decodeShuzu(decode_results*)':
/Users/holga/Documents/Arduino/libraries/IRremote/ir_Shuzu.cpp:161:23: error: 'SHUZU_BITS' was not declared in this scope
  for (int i = 0;  i < SHUZU_BITS;  i++) {`

My goal is a light barrier. It works with 4 38kHz pulse and a 35ms off time. I'm trying to replace the 4 constant pulses with a short burst that contains data with 3 bit length.
I was wondering if that's possible with only small additional power conumption

AnalysIR, 2020-02-14:
looks like you are  using the template example which is documented as a 'fantasy'..
so you could try adding in a define for the indicated error
#define SHUZU_BITS 3 //as an example example

> I was wondering if that's possible with only small additional power consumption.

Sure..sounds good. You need to test a lot under varying conditions particularly if outdoors.
I don't recall seeing too many battery powered light barriers before...given they are usually permanent installations.

For a light barrier the TSSP4038 from Vishay is the best IR receiver to use. Some other receivers can switch off with continuous streams of IR pulses.

maxholgasson, 2020-02-14:
Ok then I guess I need to have a closer look to understand the template

i actually couldnt also find a lot of battery powered light barriers but there are some e.g which runs a lot of months http://www.trailmaster.com/tm1550ps.php

I have here both vishay receiver TSSP4038 and TSOP4838. I figured out that when sending really short pulses the TSSP4038 doesnt detect them

[image]

[image]

AnalysIR, 2020-02-14:
The short burst you are are showing above  is called the IR carrier.(in this case ~44kHz)
You should be sending a burst of a 38kHz signal for a total of 500uSecs ON(ie a mark) and maybe 500uSecs->1000uSecs OFF (a space). That is the way to implement an IR barrier. (Carrier always on is another way)

By varying the pulse patterns you can mitigate against false triggers.

You will get lots of issues with bursts of ~100uSecs, especially outdoors.

Of course, with a large enough battery bank, operation time can be extended. Really depends on the characteristics of your project.

The TSSP4038 is the best off-the-shelf solution for IR Barriers (just need to drive it properly). Using a narrow angle IR Emitter with additional optics can deliver the same perfomance with less current used....effectively extending battery life.

maxholgasson, 2020-02-15:
Ok I see, I figured out the length of the burst by experimenting indoors how low I can go. Haven't tested it outdoors yet. 
Can you explain how varying the pulse patterns mitigate against false triggers? Sounds interesting. The commercial products which I had in my hand used two NE555 for creating the carrier and pattern which didn't change. Another product used a Attiny85, The commercial recieving device uses a 4838 in combination with a CD4017 which is clocked with 46Hz and detects a cut beam when 3 IR bursts are missing. The commercial IR barrier (sender) runs on 4 AA batteries  for at least 3 months

AnalysIR, 2020-02-15:
>  false triggers

By measuring the length of gaps _(not all the same length, but in a predetermined order)_, you can be more confident the barrier is intact. (MCU required at both ends. With a continuous beam an easy exploit would be to use a jammer pointed at the receiver.

> The commercial IR barrier (sender) runs on 4 AA batteries for at least 3 months

I would avoid battery operated light barriers, whenever possible. If it is not possible to avoid a battery, you can always add additional capacity & solar etc etc...

You can also have the emitter & receiver at the same side, usung the same power source, by reflecting the beam off a mirror or similar. _You could also just run a wire for power._

I am curious as to your project...any details?

FYI: https://wiki.analysir.com/index.php?title=BeamIR

AnalysIR, 2020-02-15:
> used two NE555

lots of drift with temperature.....might result in false triggers & varying operational range

maxholgasson, 2020-02-20:
Sorry for the delay.. that's an interesting concept. MCU is used at both ends. I calculated and measured power consumption and according to a battery calculator it should run with two 18650 cells for 8 month in low power mode or 16 months in high power. I tested it oudoors and reached around 15m in high power mode at ca 200mA. I'm around 83,54m% duty-cycle

My project is a ir beam based counter device for wildlife  monitoring

AnalysIR, 2020-02-20:
I think you may have the 8 & 16 month numbers reversed :)

Your percentage duty cycle figure is not clear?

The most efficient duty cycle for battery operation is 33.3333%, after that just vary the peak IR current as required.

Wildlife monittoring is probably the most challenging type of project having to cope with battery power, Sunshine, falling leaves..debris during storms etc etc...best of luck :)

However, your best shot for power saving, by far, is with the use of optics. With lenses, you can focus the IR beam and thus get the same performance/range with much less IR current....=> longer battery life.  There are relatively cheap weatherproof IR barrier systems avaialble from china with very good optics and could be repurposed for your needs.

Alternatively just hack some cheap 18650 powered torces with a beam focus. Just need ot get the focal point adjusted correctly (trial & error).

maxholgasson, 2020-02-20:
You're right with the numbers ;)
The 83,54m% is calculated by my logic analyser.  A duty-cycle like 33% would drain the batteris quite quickly. I have a burst of 5  38kHz periods and a pause of 18ms. The current draw is 0,3mA or 0,5mA. Yes it's quite challenging but companies like trailmaster or cognysis managed it quite well.  Optics is a good idea and I already tested one for IR light. The available beam brakes I mentioned however don't use them. Good idea to have a look at the chinese sets! 
I guess I need some kind of algorithm to detect any leaves etc. On the other side the old and reliable trailmaster camera trigger just uses NE555 timers etc

closed 2020-07-04 by ArminJo