Arduino-IRremote/Arduino-IRremote #677: NEC sometimes multiple receiving same code.

gavron04, 2020-02-25:
**Board:** ARDUINO Micro
**Library Version:** 2.1.0
**Protocol:** NEC

Hello. Im working with NEC protocol on Apple remote control with enabled holding key with "Decoded NEC: FFFFFFFF (0 bits)".

I have conntected 2 arduino (one which are displaying code on serial monitor, second - driver which Im programming).

`    
 if (irrecv.decode(&results))
  {
    if (results.decode_type == NEC)
    {
      currentButtonCode = results.value;
      if (currentButtonCode == 0xFFFFFFFF) {
            currentButtonCode = lastButtonCode;
            blokadaPilot=0;
        } else {
            lastButtonCode = currentButtonCode;
blokadaPilot=1;
        }

  if (results.value== Button3 && count == 0 && mute_pressed==0 && blokadaPilot==1)
      {
        vol = vol + 0.5;
        if (vol > 127)vol = 127;
      }

      else if (currentButtonCode == Button3 && count == 0 && mute_pressed==0 && blokadaPilot==0)
      {
        vol = vol + 5;
        if (vol > 127)vol = 127;
      }

      if (results.value == Button4 && count == 0 && mute_pressed==0 && blokadaPilot==1)
      {
        vol = vol - 0.5;
        if (vol < 0)vol = 0;
      }
 
      else if (currentButtonCode == Button4 && count == 0 && mute_pressed==0 && blokadaPilot==0)
      {
        vol = vol - 5;
        if (vol < 0)vol = 0;
      }
    }
  irrecv.resume();
}
`

Button3;4 - code from Apple remote, up down key

count - work only when count 0 is set (must be)

mute_pressed - work only, when mute_pressed is 0 (must be)

blokadaPilot=0 - work only, when you holding key (increase/decrease vol by 5) - added because Im looking for solve problem

blokadaPilot=1 - work only, when you are pressing key (increase/decrease bol by 0.5) - added because Im looking for solve problem

Problem is: sometimes when Im pressing (not holding) up or down key , it should be decrease/increase by 0.5 , but its not. Sometimes its increase/decrease by 5. But when Im looking at same time when was increased/decreased wrong (0,5+5) ib arduino uno, which have Irdump v1 or v2 - it was nothing wrong. So I think that NEC protocol has bug in software/library.

When Im holding key - always is increase/decrease by 5 - so its work ok. I have never seen that it was increase/decrease by 0,5 + 5

Thanks and have a good day to all

lpuser13, 2020-02-25:
I did not examine your code in depth...but...

```
  if (results.value== Button3 ....
   if (results.value == Button4 ....
```

should you not be using **_currentButtonCode_**  instead of **_results.value_**

gavron04, 2020-02-26:
But I need store last "good" code (not 0xFFFFFFF) before repeat code (0xFFFFFF). But, maybe is other way to solve this?

AnalysIR, 2020-02-27:
Did you consider using 2 variables if you need to store 2 things?

It seems your issue is more 'C' coding related and not really related to IRremote. It may be better to seek assistance via the Arduino forum or similar.

gavron04, 2020-02-29:
I think that main problem is in IRremote library. I think that Apple remote control sometimes is sending 0xFFF..FF when Im only clicking not holding button.

gavron04, 2020-03-01:
Serial monitor:
it was only clicking, not holding key. And as all can see - sketch read in one moment as holding key. I dont know its something wrong with Apple remote control or IRremote library. But to solve this problem I think that I need delay/space beetween each pressing key.

```
lastButtonCode
2011287627
isARepeat
0
lastButtonCode
2011287627
isARepeat
0
lastButtonCode
2011287627
isARepeat
0
lastButtonCode
2011287627
isARepeat
0
lastButtonCode
2011287627
isARepeat
1
```

gavron04, 2020-03-01:
Play/pause key on Apple remote, always send good code and 0xFFFF, but IRremote control on arduino uno with IRdump cant see second wrong code. (sometimes other buttons send good code and 0xFFF too)

Only pressing/clicking one time:

```
12:54:29.317 -> lastButtonCode
12:54:29.317 -> 2011265611
12:54:29.317 -> isARepeat
12:54:29.317 -> 0
12:54:29.420 -> lastButtonCode
12:54:29.420 -> 2011242571
12:54:29.420 -> isARepeat
12:54:29.420 -> 0
```

`12:54:29.350 -> 77E17A4B
12:54:29.350 -> Decoded NEC: 77E17A4B (32 bits)
12:54:29.384 -> Raw (68): 9050 -4400 650 -450 650 -1600 650 -1550 600 -1600 650 -500 650 -1550 650 -1600 600 -1600 600 -1600 650 -1600 600 -1600 600 -500 650 -500 650 -450 600 -550 600 -1600 650 -450 650 -1600 650 -1550 650 -1550 650 -1600 600 -500 600 -1650 600 -500 600 -500 650 -1600 600 -500 600 -500 650 -1600 650 -450 600 -1600 650 -1600 650 
`

(2011242571 its 0xFFFF)

holding:

```
12:56:12.963 -> lastButtonCode
12:56:12.963 -> 2011265611
12:56:12.963 -> isARepeat
12:56:12.963 -> 0
12:56:13.067 -> lastButtonCode
12:56:13.067 -> 2011242571
12:56:13.067 -> isARepeat
12:56:13.067 -> 0
12:56:13.136 -> lastButtonCode
12:56:13.136 -> 2011242571
12:56:13.136 -> isARepeat
12:56:13.136 -> 1
12:56:13.239 -> lastButtonCode
12:56:13.239 -> 2011242571
12:56:13.239 -> isARepeat
12:56:13.239 -> 1
12:56:13.342 -> lastButtonCode
12:56:13.342 -> 2011242571
12:56:13.342 -> isARepeat
12:56:13.342 -> 1
12:56:13.445 -> lastButtonCode
12:56:13.445 -> 2011242571
12:56:13.445 -> isARepeat
12:56:13.445 -> 1
12:56:13.546 -> lastButtonCode
12:56:13.546 -> 2011242571
12:56:13.546 -> isARepeat
12:56:13.546 -> 1
12:56:13.648 -> lastButtonCode
12:56:13.648 -> 2011242571
12:56:13.648 -> isARepeat
12:56:13.648 -> 1
12:56:13.751 -> lastButtonCode
12:56:13.751 -> 2011242571
12:56:13.751 -> isARepeat
12:56:13.751 -> 1
12:56:13.854 -> lastButtonCode
12:56:13.854 -> 2011242571
12:56:13.854 -> isARepeat
12:56:13.854 -> 1
```

```
12:56:12.980 -> 77E17A4B
12:56:12.980 -> Decoded NEC: 77E17A4B (32 bits)
[... 80 lines cut ...]
```
12:58:07.289 -> Decoded NEC: 77E1104B (32 bits)
12:58:07.324 -> Raw (68): 9050 -4400 600 -550 600 -1600 600 -1650 550 -1650 600 -500 600 -1650 600 -1600 600 -1650 550 -1650 600 -1600 600 -1650 550 -550 600 -500 600 -550 600 -500 600 -1650 550 -550 600 -550 550 -550 600 -1600 600 -550 600 -500 600 -550 550 -550 600 -500 600 -1650 600 -500 600 -550 550 -1650 600 -500 600 -1650 600 -1600 600 
12:58:08.388 -> 77E1104B
12:58:08.388 -> Decoded NEC: 77E1104B (32 bits)
12:58:08.422 -> Raw (68): 9050 -4450 550 -550 600 -1600 600 -1650 600 -1600 600 -550 550 -1650 600 -1600 600 -1650 550 -1650 600 -1650 550 -1650 600 -500 600 -550 600 -500 600 -550 550 -1650 600 -500 600 -550 600 -500 600 -1650 550 -550 600 -550 550 -550 600 -500 600 -550 550 -1650 600 -550 550 -550 600 -1600 600 -550 550 -1650 600 -1650 600 
12:58:09.527 -> 77E1104B
12:58:09.527 -> Decoded NEC: 77E1104B (32 bits)
12:58:09.562 -> Raw (68): 9050 -4450 600 -500 600 -1650 550 -1650 600 -1600 600 -550 600 -1600 600 -1650 550 -1650 600 -1600 600 -1650 600 -1600 600 -550 550 -550 600 -500 600 -550 600 -1600 600 -550 550 -550 600 -500 600 -1650 600 -500 600 -550 550 -550 600 -500 600 -550 600 -1600 600 -550 550 -550 600 -1600 600 -550 600 -1600 600 -1650 600 
12:58:19.506 -> 77E1104B
12:58:19.506 -> Decoded NEC: 77E1104B (32 bits)
12:58:19.540 -> Raw (68): 9050 -4400 600 -500 600 -1650 550 -1650 550 -1650 600 -500 600 -1650 550 -1650 550 -1650 600 -1600 600 -1600 600 -1600 600 -550 600 -500 600 -500 600 -550 550 -1650 600 -500 600 -550 550 -550 600 -1600 600 -500 600 -550 600 -500 600 -500 600 -550 550 -1650 600 -500 600 -500 600 -1650 550 -550 600 -1600 600 -1650 550 
12:58:20.676 -> 77E1104B
12:58:20.676 -> Decoded NEC: 77E1104B (32 bits)
12:58:20.709 -> Raw (68): 9000 -4450 550 -550 600 -1600 600 -1600 600 -1650 550 -550 600 -1600 600 -1600 600 -1650 550 -1650 600 -1600 600 -1600 600 -550 550 -550 600 -500 600 -550 550 -1650 600 -500 600 -500 600 -550 600 -1600 600 -500 600 -550 550 -550 600 -500 600 -550 550 -1650 550 -550 600 -500 600 -1650 550 -550 600 -1600 600 -1650 550 
12:58:21.978 -> 77E1104B
12:58:21.978 -> Decoded NEC: 77E1104B (32 bits)
12:58:22.013 -> Raw (68): 9000 -4400 600 -500 600 -1650 550 -1650 600 -1600 600 -550 550 -1650 550 -1650 600 -1600 600 -1600 600 -1650 550 -1650 600 -500 600 -550 550 -550 600 -500 600 -1650 550 -550 600 -500 600 -550 550 -1650 600 -500 600 -500 600 -550 600 -500 600 -500 600 -1650 550 -550 600 -500 600 -1650 550 -550 600 -1600 600 -1650 550 
12:58:23.070 -> 77E1104B
12:58:23.070 -> Decoded NEC: 77E1104B (32 bits)
12:58:23.105 -> Raw (68): 9050 -4450 550 -550 600 -1600 600 -1650 600 -1600 600 -550 550 -1650 600 -1600 600 -1650 600 -1600 600 -1650 550 -1650 600 -500 600 -550 600 -500 600 -550 550 -1650 600 -500 600 -550 600 -500 600 -1650 550 -550 600 -500 600 -550 600 -500 600 -550 550 -1650 600 -500 600 -550 600 -1600 600 -550 550 -1650 600 -1650 550 
12:58:24.446 -> 77E1104B
12:58:24.446 -> Decoded NEC: 77E1104B (32 bits)
12:58:24.481 -> Raw (68): 9100 -4400 600 -550 600 -1600 600 -1600 600 -1650 600 -500 600 -1650 550 -1650 600 -1600 600 -1650 600 -1600 600 -1650 550 -550 600 -500 600 -550 600 -500 600 -1650 550 -550 600 -500 600 -550 600 -1600 600 -550 550 -550 600 -500 600 -550 600 -500 600 -1650 550 -550 600 -500 600 -1650 600 -500 600 -1650 550 -1650 600 
```

gavron04, 2020-03-01:
With delay 30-50ms work perfect. But maybe it would be better change delay to millis?

AnalysIR, 2020-03-04:
Some comments:

- NEC sends the 32 bit signal and only sends a 'pseudo' Header to indicate repeats when holding the button down (saves battery)
- Many systems send 1 repeat on the first press. (pretty common)
- Search for "sbprojects IR NEC" .....to get a good explanation.
- IRremote returns 0xFFFFFFFF for each repeat NEC signal decoded.
- If you are printing to Serial you may miss some signal  parts.
- In your code you are increasing VOL by 5 for each repeat and 0.5 for the initial 32 bit signal. I would stick with the same value for both. As is, it seems you are increasing vol by a factor of 10 for each repeat vs a single signal without repeat.
- same issue for decreasing vol.
- Also, rather than use 0.5 with a range of 127...why not use 1 with a range of 255. That way you can use an 8 bit int, instead of a float. (more efficient)

gavron04, 2020-03-06:
Thanks for reply. 

Thats good point - **"Also, rather than use 0.5 with a range of 127...why not use 1 with a range of 255. That way you can use an 8 bit int, instead of a float. (more efficient)"**

Anyway. With delay (50) its working good. But I need to check with millis() (more efficient).

Other question. I would like to learn my driver codes from Apple remote (sometimes they have other codes between new pcs).

Original IRrecvDump:
```
/*
 * IRremote: IRrecvDump - dump details of IR codes with IRrecv
 * An IR detector/demodulator must be connected to the input RECV_PIN.
 * Version 0.1 July, 2009
 * Copyright 2009 Ken Shirriff
 * http://arcfn.com
 * JVC and Panasonic protocol added by Kristian Lauszus (Thanks to zenwheel and other people at the original blog post)
 * LG added by Darryl Smith (based on the JVC protocol)
 */

#include <IRremote.h>

/* 
*  Default is Arduino pin D11. 
*  You can change this to another available Arduino Pin.
*  Your IR receiver should be connected to the pin defined here
*/
int RECV_PIN = 11;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

void dump(decode_results *results) {
  // Dumps out the decode_results structure.
  // Call this after IRrecv::decode()
  int count = results->rawlen;
  if (results->decode_type == UNKNOWN) {
    Serial.print("Unknown encoding: ");
  }
  else if (results->decode_type == NEC) {
    Serial.print("Decoded NEC: ");

  }
  else if (results->decode_type == SONY) {
    Serial.print("Decoded SONY: ");
  }
  else if (results->decode_type == RC5) {
    Serial.print("Decoded RC5: ");
  }
  else if (results->decode_type == RC6) {
    Serial.print("Decoded RC6: ");
  }
  else if (results->decode_type == PANASONIC) {
    Serial.print("Decoded PANASONIC - Address: ");
    Serial.print(results->address, HEX);
    Serial.print(" Value: ");
  }
  else if (results->decode_type == LG) {
    Serial.print("Decoded LG: ");
  }
  else if (results->decode_type == JVC) {
    Serial.print("Decoded JVC: ");
  }
  else if (results->decode_type == AIWA_RC_T501) {
    Serial.print("Decoded AIWA RC T501: ");
  }
  else if (results->decode_type == WHYNTER) {
    Serial.print("Decoded Whynter: ");
  }
  Serial.print(results->value, HEX);
  Serial.print(" (");
  Serial.print(results->bits, DEC);
  Serial.println(" bits)");
  Serial.print("Raw (");
  Serial.print(count, DEC);
  Serial.print("): ");

  for (int i = 1; i < count; i++) {
    if (i & 1) {
      Serial.print(results->rawbuf[i]*USECPERTICK, DEC);
    }
    else {
      Serial.write('-');
      Serial.print((unsigned long) results->rawbuf[i]*USECPERTICK, DEC);
    }
    Serial.print(" ");
  }
  Serial.println();
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
}
```

my small sketch for tests:

```
#include <IRremote.h>
#define irPin 5
IRrecv irrecv(irPin);
decode_results results;

unsigned long Button1 = 0; //next in // prawo
unsigned long Button2 = 0x77E1104B; //prev in // lewo
unsigned long Button3 = 0x77E1D04B; //vol up //gora 
unsigned long Button4 = 0x77E1B04B; //vol down //dol
unsigned long Button5 = 0x77E1404B; //menu
unsigned long Button6 = 0x77E17A4B; //playpause//mute
unsigned long Button7 = 0x77E1BA4B; //ok

void setup()
{
  irrecv.enableIRIn();

    Serial.begin(9600);
}

void loop()
{
  if (irrecv.decode(&results))
  {
    if (results.decode_type == NEC)
    {
        Serial.println("button1 hex");
      Serial.println(results.value, HEX);
      Button1=results.value;
      delay(50);
      
    }
    
    irrecv.resume(); // Continue receiving
  }    
    Serial.println("button1");
    Serial.println(Button1, HEX);
}
```

and again, without delay(50) I got for center (OK) ; play/pause ; menu button one time OK code, next time , wrong:

with delay, all is good. 

How original IRrecvDump works ok without delay??

gavron04, 2020-03-06:
Also, which way is best for storing NEC codes in eeprom?

AnalysIR, 2020-03-06:
```
void loop()
{
  if (irrecv.decode(&results))
  {
    if (results.decode_type == NEC)
    {
        Serial.println("button1 hex");
      Serial.println(results.value, HEX);
      Button1=results.value;
      delay(50);
      
    }
    Serial.println("button1");
    Serial.println(Button1, HEX);

    irrecv.resume(); // Continue receiving
  }    

}
```
Try above instead (putting prints inside the "if" statement. Once yo udo that you may net need the delay!!!

> Also, which way is best for storing NEC codes in eeprom?

Refer to the many examples of writing to /reading from eeprom via google etc

gavron04, 2020-03-06:
Yes, there are many, many examples. But no examples with IR codes.

AnalysIR, 2020-03-08:
> Yes, there are many, many examples. But no examples with IR codes.

The "IR code" for NEC is just a 32-bit unsigned int.....which is made up of 4 8-bit unsigned ints.

> needed 77E1BA4B , sometimes get 77E1204B
Could be .....
- weak battery in remote control
- reflection off any surface (like a monitor screen, window, shiny metal etc etc)
- not pointing directly at the receiver (1->2m distance is usually good for capturing signals)
- Poor IR receiver
- Interference from ambient light sources (sunshine, TVs, monitors, light bulbs etc)
- proximity to WiFi source Tx
- bad wiring
- etc....

gavron04, 2020-03-08:
Ok.thanks for information.

Im using tsop4838 Vishay, its ok or there are better Ir?

AnalysIR, 2020-03-08:
Im using tsop4838 Vishay, its ok or there are better Ir?

TSOP4838 hould be fine,. Most Vishay receivers are top quality (other than very old ones that sre superceded). My general favourite from Vishay is TSOP34438. Vishay seem to release improved models every few years.

From the TSOP4838 data sheet:

> Best choice for (1) RC-5 (2) RC-6 (3) Panasonic (4) NEC (5) Sharp (6) r-step (7) Thomson RCA

gavron04, 2020-03-09:
Ok. I see that TSOP32438 has better angle (90, 4838 has 45).

```
#include <IRremote.h>
#define irPin 5
IRrecv irrecv(irPin);
decode_results results;

#include <EEPROM.h>

unsigned long Button1 = 0; //next in // prawo

#define menuVol A0

void setup()
{
  irrecv.enableIRIn();
  Serial.begin(9600);
  pinMode(menuVol, INPUT);
}

void loop()
{
  if (irrecv.decode(&results))
  {
    if (results.decode_type == NEC)
    {
      Serial.println("button1 read");
      Serial.println(results.value, HEX);
      Button1 = results.value;
      EEPROM.update(0, Button1);
      delay(50);
    }
    irrecv.resume(); // Continue receiving
  }

  if (digitalRead(menuVol) == HIGH)
  {
    Button1 = EEPROM.read(0);
    delay(100);
    Serial.println("button 1 power reset");
    Serial.println(Button1, HEX);
  }
}
```

with this I have only last 2 digits.

button1 read
77E1BA4B
button 1 power reset
4B

button1 read
77E1C010
button 1 power reset
10

same situation when im not using update, but "put".

AnalysIR, 2020-03-09:
+/-45 Dg = 90 deg

> with this I have only last 2 digits.

Yes. because update & write use only 1byte. 
 [Looks like you have not read the docs](https://www.arduino.cc/en/Reference/EEPROM), which is where I found he following information:
You need to make 4 writes/updates to store a 32-bit integer, using read/write/update.

However,  GET/PUT may work bettwer for you, but remember the address increases by 4, for every value you store/retrieve.

gavron04, 2020-03-09:
All values which I need to store in eeprom is 22 (7 of them are IR).

6x values 0-255 range
1x value 0-100 range
1x 1-6 range
1x 0-8 range
3x 0-1 range
3x 0-99 range
for this I think that the best way will be EEPROM.update

and 7x IR , HEX.

AnalysIR, 2020-03-09:
use read/update for the first list above (all fit into an 8 bit uint)

Then use PUT/GET for the 7xIR 32 bit values.

gavron04, 2020-03-11:
Ok. Done, but  now I have problem. Saved values is without leading "0x" and arduino cant work with this codes properly.

How to add "0x" before "77E1E04B" to get "0x77E1E04B" ?

Button1=(EEPROMReadlong(15),HEX)

gavron04, 2020-03-12:
Only Im thinking about cycle of r/w internal eeprom. its "only" 100 000. 

I have 24 values in eeprom. 

24  times read + 24 times write * 365 days (one year) = 17 520
3x R/W (72times) * 2y = 52 560
4x R/W * 3y = 105 120 - its higher than atmega can.
5x R/W * 4y = 175 200 - its higher than atmega can.

As I said before, write will be max 2-3 times per year. only read will be all time if power is reset

But I think, device will be "long life" if I will add external eeprom, which has 1M cycles.

tookitogo, 2020-04-21:
> Im using tsop4838 Vishay, its ok or there are better Ir?
> 
> TSOP4838 should be fine,. Most Vishay receivers are top quality (other than very old ones that are superceded). My general favourite from Vishay is TSOP34438. Vishay seem to release improved models every few years.
> 
> From the TSOP4838 data sheet:
> 
> > Best choice for (1) RC-5 (2) RC-6 (3) Panasonic (4) NEC (5) Sharp (6) r-step (7) Thomson RCA

Just a heads-up that that datasheet (like most of the Vishay IR receiver datasheets) is for the whole **series**. The numbers are footnotes denoting _which specific model_ is best for each protocol, not that there's one single model that is best for _all_ of those. The TSOP4838 has no footnote, meaning it's not the preferred model for _any_ protocol. In the case of the TSOP24/28/24/48 series, these footnotes mean the TSOP24/4436 is preferred for (1)(2)(3), TSOP24/4438 for (4)(5)(6), and TSOP24/4456 for (6)(7).

(The first numeral in the part number simply denotes the pinout.)

gavron04, 2020-04-21:
Yes, thanks to all. I have order TSOP34438 :)

tookitogo, 2020-04-21:
As for the Apple IR codes: eventually, it'd probably be good to create explicit support for Apple IR codes, since they're only NEC-ish, not entirely NEC, as I understand it. Wiki describes it as using NEC timing, but with a different data format. (Also, don't be surprised if two Apple remotes produce different output for the same button: Apple remotes have a device ID code, so that receivers can "pair" by ignoring any remotes other than the device ID learned. And Apple remotes allow the ID to be changed, should one randomly have two remotes with the same device ID.)

tookitogo, 2020-04-21:
The other thing to keep in mind is that the select button (center of navigation circle) and play/pause buttons need to send **two** IR codes, not one. For backward compatibility with the old white Apple Remote (where select and play/pause were the same button), both of those buttons send the same IR code, but first send a "prefix" code, so that newer devices can tell the buttons apart, while old devices simply ignore the prefix and see them both as the same thing.

gavron04, 2020-04-21:
> As for the Apple IR codes: eventually, it'd probably be good to create explicit support for Apple IR codes, since they're only NEC-ish, not entirely NEC, as I understand it. Wiki describes it as [...]

Yes, you're right. I have about 3pcs new Apple remote, all have different codes. Also, they can change codes after pressing "MENU" + "OK". Also, fake apple remote from aliexpress have this function,too

tookitogo, 2020-04-21:
> Yes, you're right. I have about 3pcs new Apple remote, all have different codes. Also, they can change codes after pressing "MENU" + "OK". Also, fake apple remote from aliexpress have this [...]

Really? I had no idea the knockoffs were that sophisticated! I assumed they’d be naive copies. Good to know!! :) 

Btw, I kept reading a bit more about the Apple protocol, and the biggest difference is that it’s 7-bit encoding, so if one wants to actually know the commands, the 8-bit decoding per NEC won’t work. And since NEC has an entire 8 bits of redundancy, it’s entirely likely that NEC decoders actually discard some Apple protocol data entirely. :/

closed 2020-08-29 by ArminJo