Arduino-IRremote/Arduino-IRremote #678: Unexpected codeType 11 FFFFFFFF DANTEX

Woodgrob, 2020-02-27:
I have a remote control from the Dantex air conditioner
Unexpected codeType 11
FFFFFFFF
Unexpected codeType 11
FFFFFFFF
Unexpected codeType 11
FFFFFFFF
Unexpected codeType 11
FFFFFFFF
Unexpected codeType 11
FFFFFFFF

Woodgrob, 2020-02-27:
I have the same problem "Unexpected codeType 11" I tried everything that was listed - RAWBUF value to 280 and change the definition of rawlen to unsigned long, TOLERANCE 35, - NOT ALL HELP EVERYTHING - "Unexpected codeType 11"
Try this sketch so we can have some more information. - Showed me -␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀ [... cut ...] ␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀

AnalysIR, 2020-02-27:
use the IRrecvDump and/or IRrecvDumpV2 sketches from the examples provided with the library.

Then try the output with the sendRAW sketch.

_(Again this has been covered many many times already here. Using the search facility is much faster!)_

Woodgrob, 2020-02-27:
FFFFFFFF
FFFFFFFF (0 bits)
Raw (194): 550 -1650 550 -1650 600 -1600 550 -1650 600 -1550 600 -1650 550 -1650 550 -1600 600 -550 550 -600 550 -550 550 -600 600 -500 600 -550 600 -500 600 -550 600 -1600 550 -1650 650 -1550 550 -1650 550 -1650 550 -1650 550 -1650 550 -1650 550 -550 600 -550 550 -550 600 -550 600 -500 600 -550 650 -500 550 -550 600 -1600 600 -1600 650 -1550 550 -1650 600 -1600 550 -1650 550 -1650 550 -1650 550 -600 600 -500 550 -600 550 -550 600 -550 550 -550 650 -500 550 -600 600 -1550 600 -1600 650 -500 550 -1650 600 -500 600 -1650 550 -550 550 -1650 600 -500 600 -550 650 -1550 550 -600 600 -1600 550 -550 650 -1550 550 -600 600 -1600 550 -1650 550 -550 550 -1650 600 -1600 550 -1650 550 -1650 600 -1600 550 -600 600 -500 550 -1650 600 -550 550 -550 600 -550 550 -550 600 -550 550 -550 600 -1650 550 -550 550 -1650 600 -500 600 -1650 600 -500 550 -600 600 -1600 550 -550 650 -1550 550 -600 600 -1600 550 -550 600 -1650 550 -1600 600
FFFFFFFF
FFFFFFFF (0 bits)
Raw (194): 600 -1600 650 -1550 550 -1650 650 -1550 550 -1650 550 -1650 550 -1650 550 -1650 600 -500 600 -550 550 -600 550 -550 550 -600 600 -500 600 -550 600 -500 600 -1650 550 -1600 600 -1600 600 -1650 500 -1650 600 -1600 600 -1600 550 -1650 650 -500 550 -550 600 -550 550 -600 550 -550 550 -550 600 -550 650 -500 550 -1650 600 -1550 600 -1650 600 -1600 550 -1600 600 -1650 500 -1650 600 -1600 650 -500 550 -600 600 -500 550 -600 550 -550 600 -550 550 -550 650 -500 550 -1650 600 -500 600 -550 650 -1550 550 -600 600 -1550 600 -550 650 -1550 550 -600 600 -1550 600 -1650 550 -550 550 -1650 600 -500 600 -1650 600 -500 550 -1650 600 -1600 600 -550 600 -1600 650 -1550 550 -1650 650 -1550 550 -1650 550 -550 600 -550 650 -1550 550 -600 600 -500 600 -550 600 -500 600 -550 600 -550 550 -1650 550 -550 600 -1600 600 -550 550 -1650 500 -600 600 -550 550 -1650 550 -600 550 -1600 600 -550 500 -1700 550 -600 500 -1650 600 -1600 650

AnalysIR, 2020-02-27:
OK

try this sketch from our blog instead..
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

Then test the results with sendRAW

Woodgrob, 2020-02-28:
Raw: (197) 6138, -7343, 647, -1564, 571, -1644, 565, -1620, 569, -1639, 594, -1592, 572, -1639, 566, -1619, 573, -1643, 544, -574, 620, -497, 572, -572, 594, -526, 572, -573, 546, -573, 622, -498, 572, -573, 546, -1639, 573, -1616, 569, -1638, 624, -1563, 571, -1644, 566, -1619, 568,
-1638, 623, -1565, 571, -572, 594, -526, 572, -548, 571, -573, 622, -498, 572, -573, 546, -573, 623, -497, 571, -1639, 621, -1570, 569, -1640, 591, -1592, 572, -1639, 541, -1645, 572, -1639, 546, -1639, 572, -548, 572, -572, 623, -497, 572, -573, 546, -573, 623, -497, 572, -573, 594,
-525, 572, -1639, 546, -573, 598, -523, 571, -1639, 621, -498, 573, -1642, 544, -574, 570, -1614, 571, -573, 622, -1565, 571, -1638, 621, -499, 572, -1639, 546, -573, 573, -1639, 546, -572, 623, -1564, 571, -1644, 595, -550, 568, -1639, 622, -1565, 571, -1639, 594, -1592, 572, -1643,
513, -604, 595, -523, 571, -1639, 623, -497, 572, -572, 568, -551, 573, -548, 571, -572, 623, -498, 571, -1639, 571, -548, 597, -1615, 546, -572, 624, -1567, 569, -574, 591, -525, 573, -1638, 546, -573, 573, -1614, 571, -572, 623, -1564, 571, -573, 542, -1643, 573, -1638, 546, -7402,
566,

Press the button on the remote now - once only

Raw: (197) 6184, -7323, 596, -1614, 597, -1614, 572, -1613, 597, -1591, 595, -1617, 591, -1595, 594, -1612, 596, -1592, 596, -546, 568, -552, 598, -547, 573, -546, 596, -525, 596, -548, 518, -601, 597, -525, 596, -1612, 597, -1595, 568, -1640, 618, -1566, 597, -1612, 516, -1671, 598,
-1612, 545, -1641, 598, -547, 572, -547, 595, -526, 597, -546, 546, -573, 598, -524, 597, -546, 593, -527, 597, -1613, 516, -1670, 599, -1612, 572, -1613, 598, -1595, 567, -1641, 567, -1619, 594, -1612, 596, -526, 597, -546, 545, -574, 598, -524, 596, -546, 593, -528, 597, -547, 572,
-547, 598, -1589, 597, -1613, 622, -498, 598, -1612, 567, -552, 598, -1613, 573, -547, 597, -1592, 594, -547, 594, -526, 598, -1612, 515, -605, 598, -1613, 572, -547, 595, -1592, 597, -546, 593, -1594, 597, -1613, 624, -523, 598, -1612, 540, -1646, 598, -1617, 491, -1693, 595, -1613,
572, -547, 596, -525, 597, -1613, 540, -580, 598, -546, 573, -546, 597, -524, 597, -546, 567, -553, 597, -1619, 543, -574, 594, -1591, 596, -546, 568, -1619, 598, -546, 518, -601, 598, -1594, 568, -574, 592, -1592, 597, -546, 516, -1670, 598, -547, 572, -1617, 568, -1618, 593, -7376,
568,

Press the button on the remote now - once only

Raw: (197) 6156, -7345, 619, -1595, 568, -1641, 591, -1592, 571, -1639, 541, -1646, 571, -1639, 545, -1645, 544, -1642, 566, -573, 597, -525, 595, -548, 517, -602, 599, -523, 570, -573, 594, -526, 597, -547, 573, -1613, 597, -1614, 546, -1639, 598, -1594, 567, -1641, 619, -1565, 596,
-1613, 597, -1590, 596, -548, 517, -602, 598, -524, 595, -547, 597, -524, 597, -547, 572, -547, 623, -498, 596, -1618, 564, -1622, 592, -1614, 594, -1593, 597, -1613, 540, -1646, 598, -1617, 544, -1640, 595, -524, 596, -547, 594, -526, 598, -546, 572, -547, 597, -525, 622, -521, 515,
-605, 598, -1612, 572, -547, 597, -524, 597, -1617, 535, -584, 594, -1614, 517, -602, 598, -1591, 595, -546, 596, -1592, 597, -1616, 509, -611, 594, -1613, 545, -574, 598, -1591, 596, -546, 595, -1596, 569, -1640, 622, -523, 597, -1613, 567, -1619, 598, -1612, 572, -1618, 569, -1642,
569, -546, 598, -523, 598, -1612, 592, -528, 598, -547, 572, -546, 596, -526, 596, -547, 516, -603, 599, -1613, 572, -546, 596, -1596, 568, -574, 591, -1593, 597, -547, 572, -547, 597, -1590, 596, -547, 596, -1591, 597, -547, 514, -1672, 597, -547, 573, -1617, 593, -1591, 596, -7376,
568,

Woodgrob, 2020-02-28:
I don’t understand why the values are different every time ?? The values did not immediately repeat !!!

Woodgrob, 2020-02-28:
Raw: (197) 6101, -7373, 587, -1644, 567, -1619, 541, -1666, 595, -1589, 544, -1664, 571, -1615, 567, -1641, 571, -1619, 540, -602, 568, -549, 568, -575, 544, -575, 569, -552, 567, -575, 571, -549, 568, -575, 544, -1641, 569, -1641, 568, -1617, 569, -1640, 570, -1619, 542, -1666, 542,
-1641, 569, -1640, 568, -551, 570, -551, 568, -574, 596, -524, 569, -574, 544, -576, 569, -551, 568, -574, 624, -1562, 569, -1643, 592, -1593, 566, -1640, 622, -1564, 569, -1639, 621, -1589, 571, -1613, 596, -524, 595, -549, 593, -526, 596, -548, 545, -573, 623, -498, 595, -548, 593,
-526, 596, -1613, 544, -575, 598, -547, 570, -1614, 623, -497, 595, -1617, 593, -508, 612, -1612, 621, -499, 596, -1613, 543, -1642, 622, -522, 543, -1642, 597, -546, 572, -1613, 622, -523, 571, -1613, 622, -1588, 621, -524, 598, -1612, 571, -1613, 598, -1612, 571, -1614, 623, -1587,
570, -548, 622, -499, 619, -1588, 622, -499, 621, -521, 543, -577, 597, -547, 571, -547, 623, -498, 596, -1612, 622, -498, 596, -1613, 569, -550, 597, -1615, 515, -603, 596, -546, 572, -1613, 597, -547, 571, -1613, 621, -499, 596, -1616, 564, -555, 593, -1613, 567, -1618, 597, -7369,
594,

Press the button on the remote now - once only
Press the button on the remote now - once only

Raw: (197) 6125, -7368, 652, -1586, 600, -1584, 595, -1621, 594, -1588, 620, -1587, 600, -1585, 624, -1586, 599, -1585, 623, -493, 627, -520, 543, -576, 624, -520, 599, -520, 648, -468, 626, -521, 593, -526, 624, -1586, 541, -1644, 624, -1586, 518, -1667, 623, -1586, 618, -1567, 624,
-1588, 535, -1650, 621, -520, 541, -578, 649, -469, 625, -520, 620, -499, 625, -519, 541, -578, 649, -496, 572, -1612, 648, -1534, 627, -1585, 649, -1567, 542, -1641, 620, -1585, 573, -1611, 624, -1587, 572, -546, 622, -497, 599, -545, 620, -499, 598, -547, 572, -546, 623, -497, 573,
-571, 620, -1572, 567, -575, 536, -577, 598, -1613, 518, -600, 623, -1562, 573, -571, 623, -1562, 572, -571, 621, -1565, 572, -1637, 622, -497, 573, -1637, 594, -525, 572, -1638, 518, -600, 624, -1561, 573, -1637, 600, -572, 546, -1644, 568, -1641, 542, -1637, 573, -1637, 519, -1666,
573, -571, 548, -571, 622, -1570, 567, -575, 615, -499, 573, -571, 519, -600, 598, -521, 573, -571, 622, -1564, 572, -571, 595, -1591, 572, -571, 519, -1666, 573, -571, 548, -571, 623, -1562, 573, -571, 622, -1563, 572, -572, 619, -1566, 572, -572, 541, -1643, 573, -1637, 519, -7451,
561,

Press the button on the remote now - once only

Raw: (197) 6157, -7340, 648, -1587, 572, -1612, 623, -1562, 598, -1611, 624, -1567, 568, -1643, 589, -1592, 595, -1612, 623, -496, 624, -520, 620, -499, 625, -519, 573, -546, 624, -495, 624, -520, 620, -499, 625, -1585, 592, -1598, 594, -1615, 513, -1667, 625, -1585, 542, -1643, 624,
-1586, 593, -1592, 624, -519, 518, -601, 625, -493, 625, -520, 622, -497, 625, -519, 519, -600, 625, -493, 625, -1586, 624, -1587, 572, -1611, 623, -1588, 572, -1612, 624, -1593, 546, -1636, 619, -1586, 573, -546, 622, -497, 623, -521, 541, -578, 625, -519, 573, -545, 623, -497, 624,
-520, 620, -1565, 623, -521, 541, -578, 624, -1586, 518, -600, 623, -1562, 598, -546, 621, -1563, 598, -546, 622, -1563, 624, -1586, 621, -498, 624, -1585, 593, -527, 624, -1586, 518, -600, 624, -1586, 548, -1637, 625, -546, 518, -1666, 599, -1611, 519, -1672, 568, -1640, 514, -1666,
599, -545, 573, -546, 623, -1562, 597, -546, 622, -498, 598, -545, 518, -602, 598, -546, 572, -546, 623, -1561, 599, -545, 622, -1563, 598, -546, 594, -1591, 623, -521, 518, -601, 598, -1618, 511, -606, 618, -1587, 572, -546, 623, -1588, 572, -546, 620, -1565, 598, -1618, 588, -7376,
588,

Press the button on the remote now - once only
Press the button on the remote now - once only

Raw: (197) 6136, -7349, 587, -1637, 622, -1563, 572, -1637, 623, -1562, 573, -1636, 624, -1567, 569, -1640, 618, -1562, 572, -571, 623, -497, 572, -572, 518, -600, 573, -546, 573, -571, 623, -497, 572, -571, 519, -1666, 573, -1637, 565, -1626, 568, -1640, 614, -1566, 573, -1636, 542,
-1643, 573, -1637, 593, -526, 573, -571, 573, -545, 624, -496, 572, -572, 621, -498, 573, -571, 573, -545, 624, -1587, 572, -1612, 624, -1561, 597, -1613, 598, -1612, 547, -1637, 598, -1619, 542, -1640, 593, -521, 573, -571, 623, -496, 573, -571, 518, -601, 573, -546, 572, -572, 623,
-496, 573, -1637, 621, -498, 572, -572, 547, -1644, 568, -575, 542, -1636, 574, -546, 572, -1637, 598, -522, 572, -1643, 589, -1594, 567, -572, 621, -1564, 572, -572, 542, -1643, 572, -571, 547, -1644, 568, -1640, 646, -495, 572, -1638, 595, -1590, 572, -1637, 621, -1570, 568, -1640,
590, -525, 572, -572, 546, -1638, 572, -547, 572, -572, 622, -498, 571, -572, 595, -524, 573, -548, 570, -1638, 572, -548, 571, -1643, 567, -551, 568, -1637, 623, -497, 571, -573, 540, -1644, 572, -572, 547, -1637, 573, -547, 571, -1638, 598, -522, 571, -1638, 623, -1562, 572, -7397,
569,

Press the button on the remote now - once only

Raw: (197) 6135, -7366, 569, -1638, 597, -1589, 571, -1637, 598, -1589, 570, -1642, 569, -1641, 542, -1638, 598, -1588, 571, -572, 624, -496, 571, -573, 594, -525, 572, -572, 546, -573, 597, -523, 571, -572, 595, -1590, 572, -1642, 590, -1594, 568, -1638, 595, -1590, 572, -1637, 621,
-1565, 571, -1638, 594, -525, 572, -572, 546, -572, 597, -523, 571, -572, 622, -498, 572, -572, 546, -573, 596, -1614, 546, -1638, 572, -1638, 546, -1638, 572, -1643, 543, -1641, 568, -1641, 537, -1643, 573, -572, 546, -572, 623, -497, 571, -572, 595, -524, 573, -572, 546, -572, 623,
[... cut ...]
-1644, 572, -1643, 512, -606, 593, -547, 547, -572, 622, -498, 572, -571, 542, -577, 598, -547, 547, -572, 621, -1564, 572, -1637, 623, -1563, 572, -1637, 623, -1568, 569, -1640, 617, -1563, 598, -1612, 623, -496, 599, -545, 519, -600, 625, -494, 599, -546, 622, -497, 624, -520, 519,
-600, 625, -1586, 572, -546, 624, -496, 624, -1586, 620, -499, 625, -1585, 565, -554, 625, -1590, 512, -606, 645, -1561, 519, -1666, 624, -493, 626, -1585, 624, -495, 625, -1591, 615, -503, 620, -1584, 593, -1593, 651, -519, 622, -1564, 622, -1593, 589, -1594, 621, -1585,

Woodgrob, 2020-02-28:
This is all the turn-on signal (when I turned off, I blocked the remote control)

lpuser13, 2020-02-28:
I tried one of your signals and it decoded as CHIGO 96 bit AC signal using AnalysIR.

Getting different timings for the same signal is normal for IR signals. IR receivers can distort the the timings by up to +/-200uSecs.

AnalysIR, 2020-02-28:
I tried one of your signals and it decoded as CHIGO 96 bit AC signal using AnalysIR.
(Names are not material as lots of suppliers rebrand ACs from other OEMs)

Getting different timings for the same signal is normal for IR signals. IR receivers can distort the the timings by up to +/-200uSecs.

Woodgrob, 2020-02-28:
I don’t understand how can I send such a signal?

AnalysIR, 2020-02-28:
I suggest you work your way thru all the examples provided with the IRremote library, using your TV first.

Once you have done that the answer to your question will become clearer.

Searching google, youtube & previous issues might also help.

Make sure to check out the sendRAW function.

Finally, it is usually a good idea to try things out yourself and to seek help online only when it doesn't work out.

closed 2020-07-01 by ArminJo