Arduino-IRremote/Arduino-IRremote #683: Break Beam

panafrican, 2020-03-23:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** 38kHz

I'm using the following code to make a simple IR break beam sensor.  This will be expanded to create a larger array of emitters/sensors to make a light beam barrier.  I am driving the emitter (TSHF6210) at 38kHz 50% duty cycle with an external signal generator.  The code below reads the receiver ( TSSP58038) and reports the status of the beam.  The emitter and receiver are 36" apart.

This almost works.  When the beam is unobstructed, the serial output is "Unbroken".  When the beam is obstructed the output is "Broken".  When the beam is then unobstructed, the output rapidly oscillates between these two states until it stabilizes on "Unbroken".  Note the time stamps.

21:39:22.856 -> Unbroken
21:39:26.571 -> Broken
21:39:27.256 -> Unbroken
21:39:27.291 -> Broken
21:39:27.291 -> Unbroken
21:39:27.291 -> Broken
21:39:27.291 -> Unbroken
21:39:27.325 -> Broken
21:39:27.325 -> Unbroken
21:39:27.325 -> Broken
21:39:27.359 -> Unbroken

The number of oscillations is variable which makes trouble shooting this more difficult.  

There is ambient light from a fluorescent lamp and light from two monitors.  The behavior still exists when I test it in a fully dark room.

Any thoughts on this would be welcome. 

`#include <IRremote.h>

#define PIN_DETECT 2
#define PIN_STATUS 13

int sensorState = 0, lastState=1;

IRsend irsend;

void setup()
{
  pinMode(PIN_DETECT, INPUT);
  pinMode(PIN_STATUS, OUTPUT);
  Serial.begin(9600);  
}

void loop() {
  digitalWrite(PIN_STATUS, !digitalRead(PIN_DETECT));
  sensorState = digitalRead(PIN_DETECT);

  if (sensorState && !lastState) {
    Serial.println("Broken");
  }
  if (!sensorState && lastState) {
    Serial.println("Unbroken");
  }

  lastState = sensorState;
}`

AnalysIR, 2020-03-23:
This is not really related to IRremote...the Arduino forum or our own [IRforum ](https://irforum.analysir.com/) might provide more assistance.

Essentially, your issue seems 'de-bounce' related, which can be solved in hardware and/or firmware. Research how to implement de-bounce on Arduinos :)

As you are trying to detect a break, the multiple triggers shouldnt matter as you only need to detect it once to indicate a beam break.

panafrican, 2020-03-23:
Thanks, it was indeed bounce and I implemented a simple fix in my code to make it work perfectly.  Sometimes all it takes is the right search term to find a solution and “bounce” was the magic word.

closed 2020-07-01 by ArminJo