Arduino-IRremote/Arduino-IRremote #684: IRremote on Webserver

cylon147, 2020-03-29:
I am trying to find an online web server that provides all the AC codes to use with a Home Automation system, I have searched everywhere but didn't come across anything related to this.

Basically I am looking for a Webserver with API to which I can Pass Brand and Remote model number to get All the IR codes in JSON formate.

Please help me out with this.

AnalysIR, 2020-03-29:
AFAIK....not available anywhere.
We have a lot of commercial users worldwide who use our AnalysIR application & IR modules to build their own database of signals for use in their commercial products.

cylon147, 2020-03-29:
Great. Can you please provide more information on Commercial Product and guide me how to build our own database.

AnalysIR, 2020-03-29:
[Contact via](https://www.analysir.com/blog/contact/)

...as it would be off-topic here

cylon147, 2020-03-29:
Request sent...

closed 2020-08-03 by ArminJo