Arduino-IRremote/Arduino-IRremote #686: Random Codes with same button push

steeeeeeeeeve, 2020-05-10:
I am using an arduino nano v3 and an IR receiver HX1838 and the IRremote library

I am using the simple receiving code: 
https://github.com/z3t0/Arduino-IRremote/blob/master/examples/IRrecvDemo/IRrecvDemo.ino 

and am receiving random codes (they are different every time I press the same button). I have tried multiple pins and multiple IR remotes. Any suggestions would be greatly appreciated!!

AnalysIR, 2020-05-10:
3 comments:

1. Those IR receivers are not great & cheap for a reason. Try to avoid sunlight and other sources of light if you can.
2. Post how you have wired it up. Have you verified the pinout. Have you defined the correct rx pin in the sketch?
3. Post the output you get from the sketch. The output is seldom identical every time, but the codes should be (relatively) consistent. (Some protocols send a different pattern on every other press, by design)

ArminJo, 2020-05-10:
Or try the AllProtocol example from the [Arduino IRMP library](https://github.com/ukw100/IRMP#allprotocol-example). With this example you can then better estimate the source of random.

steeeeeeeeeve, 2020-05-11:
Thanks for the responses!
I've sheltered from light sources and it still has the issue.

Wiring: Receiver signal to pin 3 (D3)
Pin defined in sketch: 3

Code:
#include <IRremote.h>

int RECV_PIN = 3;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  // In case the interrupt driver crashes on setup, give a clue
  // to the user what's going on.
  Serial.println("Enabling IRin");
  irrecv.enableIRIn(); // Start the receiver
  Serial.println("Enabled IRin");
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  delay(100);
}

Output when pressing UP button 10 times:
Enabling IRin
Enabled IRin
FF18E7
FFFFFFFF
FF18E7
D3B103C5
FF18E7
FF18E7
FF18E7
83B97C75
FF18E7
1D929A6E
15E247A7

[image]

AnalysIR, 2020-05-11:
the FF18E7 is the correct signal & the FFFFFFFF is the repeat signal when you keep the button pressed.

The other value are noise/errors, which should not occure.

Possible reasons:
- poor receiver as mentioned earlier
- loose wire connections (less likely)
- bad battery in remote. The remote in hte photo can have a short range compared to most remotes.
- Make sure the signal is not reflecting off any nearby shiny surface like mirrors/Displays/glass etc
- Make sure to hold the remote within ~1m of the receiver and point it directly at the receiver. The LED is above the 2 button and you should point that at he remote.
- Did you try with another TV remote?

Suggestion: Try the IRrecvDumpV2 example sketch and post the output. This should give a better picture.

**Workaround**: This remote is NEC based and is encoded in a particular way. FF18E7 is a 32 bit value 0x00FF18E7. These signals always have the same pattern. the first and 2nd byte (00+FF) equals FF and the 3rd and 4th bytes (18+E7) equals FF. You can use this rule for this remote to check the values decoded and ignore any signal received that does not match. Similarly you could buld up a switch statement with all of the signal values you need and just igore any other value decoded.

steeeeeeeeeve, 2020-05-12:
Thanks for the suggestions. I have checked the wire connections and they seem good and have made sure the remote is close and not reflecting. I am getting about a 25% success rate using the cheap remote pushing the up button. I have also tried an LG tv remote with similar success. Below is the suggested IRrecvDumpV2 results pushing the up button 8 times on the cheap remote. Are any clues in this data regarding what is wrong?

Encoding  : UNKNOWN
Code      : 6F224440 (32 bits)
Timing[71]: 
     +9100, -4400     + 650, - 550     + 600, - 550     + 600, - 550
     + 600, - 550     + 600, - 550     + 600, - 550     + 600, - 600
     + 600, - 550     + 650, -1550     + 600, -1650     + 700, -1500
     + 600, -1600     + 600, -1650     + 600, - 200     + 100, -1300
     + 600, -1650     + 600, -1600     + 600, - 550     + 600, - 550
     + 700, - 400     + 650, -1650     + 600, -1600     + 600, - 550
     + 600, - 550     + 600, - 600     + 600, - 150     + 150, -1300
     + 600, -1600     + 600, -1650     + 600, - 550     + 600, - 550
     + 600, -1650     + 550, -1650     + 600, -1600     + 600
unsigned int  rawData[71] = {9100,4400, 650,550, 600,550, 600,550, 600,550, 600,550, 600,550, 600,600, 600,550, 650,1550, 600,1650, 700,1500, 600,1600, 600,1650, 600,200, 100,1300, 600,1650, 600,1600, 600,550, 600,550, 700,400, 650,1650, 600,1600, 600,550, 600,550, 600,600, 600,150, 150,1300, 600,1600, 600,1650, 600,550, 600,550, 600,1650, 550,1650, 600,1600, 600};  // UNKNOWN 6F224440

Encoding  : UNKNOWN
Code      : C30F1C7C (32 bits)
Timing[73]: 
     +9150, -4400     + 600, - 550     + 600, - 550     + 600, - 600
     + 550, - 150     + 250, - 150     + 650, - 550     + 600, - 550
     + 600, - 550     + 600, - 550     + 600, -1600     + 750, -1500
     + 600, -1600     + 600, -1650     + 600, -1600     + 650, -1550
     + 600, -1650     + 650, -1550     + 600, - 550     + 600, - 150
     + 250, - 150     + 600, - 550     + 600, - 150     + 150, -1350
     + 600, -1600     + 600, - 550     + 600, - 550     + 600, - 600
     + 550, -1600     + 650, -1600     + 600, -1600     + 650, - 550
     + 600, - 550     + 600, -1600     + 600, -1650     + 600, -1600
     + 600
unsigned int  rawData[73] = {9150,4400, 600,550, 600,550, 600,600, 550,150, 250,150, 650,550, 600,550, 600,550, 600,550, 600,1600, 750,1500, 600,1600, 600,1650, 600,1600, 650,1550, 600,1650, 650,1550, 600,550, 600,150, 250,150, 600,550, 600,150, 150,1350, 600,1600, 600,550, 600,550, 600,600, 550,1600, 650,1600, 600,1600, 650,550, 600,550, 600,1600, 600,1650, 600,1600, 600};  // UNKNOWN C30F1C7C

Encoding  : NEC
Code      : FF18E7 (32 bits)
Timing[67]: 
     +9200, -4400     + 600, - 550     + 600, - 550     + 600, - 550
     + 600, - 550     + 600, - 550     + 650, - 550     + 600, - 550
     + 600, - 550     + 650, -1550     + 600, -1600     + 650, -1600
     + 600, -1650     + 550, -1650     + 600, -1600     + 600, -1650
     + 600, -1600     + 600, - 550     + 600, - 550     + 600, - 550
     + 600, -1650     + 600, -1600     + 600, - 550     + 600, - 550
     + 600, - 550     + 600, -1650     + 600, -1600     + 600, -1600
     + 600, - 600     + 550, - 600     + 550, -1650     + 600, -1600
     + 600, -1650     + 600
unsigned int  rawData[67] = {9200,4400, 600,550, 600,550, 600,550, 600,550, 600,550, 650,550, 600,550, 600,550, 650,1550, 600,1600, 650,1600, 600,1650, 550,1650, 600,1600, 600,1650, 600,1600, 600,550, 600,550, 600,550, 600,1650, 600,1600, 600,550, 600,550, 600,550, 600,1650, 600,1600, 600,1600, 600,600, 550,600, 550,1650, 600,1600, 600,1650, 600};  // NEC FF18E7
unsigned int  data = 0xFF18E7;

Encoding  : UNKNOWN
Code      : 878F8DCF (32 bits)
Timing[71]: 
     +9150, -4400     + 750, - 450     + 550, - 200     + 150, - 250
     + 750, - 100     + 900, - 550     + 600, - 550     + 600, - 550
     + 600, - 550     + 600, - 550     + 600, -1650     + 600, -1600
     + 600, -1600     + 600, -1650     + 650, -1550     + 600, -1650
     + 550, -1650     + 600, -1600     + 600, - 550     + 600, - 600
     + 550, - 600     + 600, -1600     + 600, -1600     + 600, - 250
     + 100, - 200     + 600, - 550     + 600, - 600     + 600, -1600
     + 600, -1600     + 600, -1650     + 600, - 550     + 600, - 550
     + 600, -1600     + 600, -1650     + 600, -1600     + 600
unsigned int  rawData[71] = {9150,4400, 750,450, 550,200, 150,250, 750,100, 900,550, 600,550, 600,550, 600,550, 600,550, 600,1650, 600,1600, 600,1600, 600,1650, 650,1550, 600,1650, 550,1650, 600,1600, 600,550, 600,600, 550,600, 600,1600, 600,1600, 600,250, 100,200, 600,550, 600,600, 600,1600, 600,1600, 600,1650, 600,550, 600,550, 600,1600, 600,1650, 600,1600, 600};  // UNKNOWN 878F8DCF

Encoding  : UNKNOWN
Code      : 4A414B63 (32 bits)
Timing[93]: 
     +9150, -4400     + 600, - 550     + 600, - 150     + 200, - 200
     + 800, - 400     + 550, - 550     + 650, - 550     + 600, - 550
     + 600, - 550     + 600, - 550     + 700, -1550     + 600, -1600
     + 600, - 150     + 150, -1300     + 600, - 150     + 100, -1400
     + 600, - 150     + 200, -1250     + 600, - 200     + 200, -1250
     + 600, - 150     + 200, - 150     + 250, - 100     + 150, - 600
     + 600, -1650     + 600, - 100     + 150, - 200     + 700, - 550
[... 25 lines cut ...]
     +9150, -4400     + 650, - 550     + 600, - 550     + 700, - 450
     + 600, - 550     + 600, - 600     + 600, - 550     + 600, - 550
     + 600, - 550     + 600, -1600     + 600, -1650     + 600, -1600
[... cut ...]
     + 600, - 550     + 600, -1650     + 600, -1600     + 600, -1650
     + 600, - 550     + 600, - 550     + 600, -1600     + 600, -1650
     + 600, -1600     + 600
unsigned int  rawData[67] = {9200,4400, 650,550, 600,550, 800,200, 750,550, 600,550, 600,600, 550,600, 600,550, 600,1600, 600,1650, 550,1650, 700,1500, 600,1650, 600,1600, 600,1650, 550,1650, 600,550, 600,550, 650,500, 600,1650, 600,1600, 600,550, 600,550, 600,550, 600,1650, 600,1600, 600,1650, 600,550, 600,550, 600,1600, 600,1650, 600,1600, 600};  // UNKNOWN 5BB845F5

AnalysIR, 2020-05-12:
It looks to me that the Receiver is suspect based on what you posted. I suggest trying a TSOP34438 as a quality unit.

See Image below. _(Captured from our AnalysIR Application)_

One solution may be to place a capacitor between the 5V & GND of the Receiver. Otherwise get a new receiver or use the software workaround I suggested earlier.

[image]
_(click image to enlarge)_

Check out the following post on our blog about your IR receiver.
[Infrared receiver showdown – TSOP34438 vs VS1838B](https://www.analysir.com/blog/2014/12/08/infrared-receiver-showdown-tsop34438-vs-vs1838b-winner-revealed/)

jdb-m, 2020-05-12:
Y

Ipad

steeeeeeeeeve, 2020-05-12:
Thanks! I agree the receiver seems suspect. I will order the one you recommended. Cheers!

dogatonic, 2020-05-13:
@steeeeeeeeeve 
### Same problem here. My Solution was new IR receiver.
This post caught my eye because I had the "exact" same problem as you, with lots of random or inconsistent code data, that would force the User to have to push buttons multiple times to get the correct response. Very frustrating.
I ordered a new IR diode from Amazon and it solved my problem.
### Or... Try Higher voltage?
In your photograph, you have the IR receiver powered by the Nano's 3v3 pin.
What happens if you use the 5v pin?
#### OK, my problem was not "exactly" the same...
I'm actually using a different library for ESP8266 with a Wemos D1 mini - and I had the same problems as you. In my case, when I tested my IR receiver using _this_ library here and an Arduino Uno using the 5v pin, it worked GREAT! Just like AnalysIR said it should.
But... When I hooked up to the 5v pin on my Wemos and the ESP8266 library, it did NOT work well. And when I tested the voltage off that pin it was only around 4.5v. So maybe that is the problem.
### New Diode to the Rescue
[image]
_"Uncle Owen, this R2 unit has a bad motivator, look!"_
I think the new diode works better at the lower voltage of my Wemos, and the old diode still works fine with a higher voltage.

AnalysIR, 2020-05-13:
> In your photograph, you have the IR receiver powered by the Nano's 3v3 pin.

@dogatonic  : Good spot, that would most likely cause the issues as seen in the image above.
The receiver should be powered from 5V, when the MCU is running @ 5V. 

_If the MCU is running @3V3 then the IR receiver should be powered from 3V3_

steeeeeeeeeve, 2020-05-31:
thanks for the tips eerybody. 
i tried the 5V and it didnt help. still had random codes.
i just got in the suggested TSOP34438 sensor and it totally solved the issue. works great!

closed 2020-05-31 by steeeeeeeeeve