Arduino-IRremote/Arduino-IRremote #687: Air conditioner remote control problem (model: MH19ZC2-07)

Ivan0x62, 2020-05-18:
i have a problem with samsung ARH-428 remote control

non riesco a ottenere i codici attraverso lo sketch (IRrecvDumpv2)

output IRrecvDumpv2:

Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
Timing[113]: 
     + 550, -1500     + 500, - 600     + 500, - 550     + 550, - 500
     + 500, - 500     + 550, - 500     + 550, - 500     + 550, - 500
     + 550, - 500     + 550, -1550     + 500, - 550     + 550, - 500
     + 500, - 550     + 550, - 500     + 500, -1550     + 500, -1500
     + 550, - 500     + 500, -1550     + 600, -1500     + 500, -1500
     + 500, -1550     + 500, -1550     + 500, -1550     + 550, -1500
     + 550, -1500     + 550, - 500     + 500, - 500     + 550, - 550
     + 500, - 500     + 550, -1550     + 550, -1450     + 550, - 550
     + 500, -1500     + 550, - 500     + 550, - 500     + 550, - 500
     + 550, -1500     + 550, - 550     + 550, - 450     + 550, -1550
     + 500, -1550     + 550, - 500     + 550, -1500     + 550, -1450
     + 550, - 550     + 500, - 500     + 550, - 550     + 550, - 500
     + 500, - 550     + 500, - 550     + 550, - 500     + 500, - 500
     + 550, -1500     + 550, -1500     + 550, -1500     + 550, -1500
     + 500
unsigned int  rawData[113] = {550,1500, 500,600, 500,550, 550,500, 500,500, 550,500, 550,500, 550,500, 550,500, 550,1550, 500,550, 550,500, 500,550, 550,500, 500,1550, 500,1500, 550,500, 500,1550, 600,1500, 500,1500, 500,1550, 500,1550, 500,1550, 550,1500, 550,1500, 550,500, 500,500, 550,550, 500,500, 550,1550, 550,1450, 550,550, 500,1500, 550,500, 550,500, 550,500, 550,1500, 550,550, 550,450, 550,1550, 500,1550, 550,500, 550,1500, 550,1450, 550,550, 500,500, 550,550, 550,500, 500,550, 500,550, 550,500, 500,500, 550,1500, 550,1500, 550,1500, 550,1500, 500};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;

LIBRARY VERSION: 2.3.3

you can help me?

thanks in advance 👍🏻

AnalysIR, 2020-05-18:
try this sketch from the post on our blog. It should help.

[Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

Ivan0x62, 2020-05-20:
> try this sketch from the post on our blog. It should help.
> 
> 
> 
> [Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

Thank you very much, I'll test the recommended sketch as soon as possible.

Ivan0x62, 2020-05-20:
> try this sketch from the post on our blog. It should help.
> 
> [Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

hi AnalysIR ,

i tested the IRremoteESP8266 library:

output IRrecvDumpv2:

Timestamp : 000051.892
Library   : v2.7.6

Protocol  : VESTEL_AC
Code      : 0xF00D9321FEC201 (56 Bits)
Mesg Desc.: Power: On, Mode: 0 (Auto), Temp: 25C, Fan: 13 (Auto Heat), Sleep: Off, Turbo: Off, Ion: Off, Swing: Off
uint16_t rawData[115] = {3036, 9044,  540, 1528,  520, 528,  516, 558,  518, 530,  526, 522,  520, 528,  516, 530,  514, 560,  514, 534,  520, 1548,  520, 528,  516, 532,  512, 560,  524, 522,  520, 1550,  518, 1524,  512, 560,  514, 1528,  518, 1550,  518, 1526,  542, 1554,  514, 1528,  518, 1550,  518, 1552,  516, 1554,  514, 532,  522, 526,  518, 530,  544, 528,  516, 1528,  520, 554,  520, 526,  518, 1526,  542, 1528,  520, 526,  516, 558,  518, 1526,  512, 536,  540, 534,  518, 1524,  514, 1556,  512, 536,  518, 1550,  518, 1526,  522, 552,  512, 534,  520, 528,  514, 532,  512, 536,  540, 534,  518, 528,  516, 532,  512, 1556,  522, 1522,  546, 1522,  514, 1582,  486};  // VESTEL_AC F00D9321FEC201
uint64_t data = 0xF00D9321FEC201; --> 67568919992254977

closed 2020-07-01 by ArminJo