Arduino-IRremote/Arduino-IRremote #691: irRemote irsend doesn't work

vgenerato, 2020-06-06:
Hi Guys, I'm trying to implant an IR sensor to control a led strip. I'm using a code integrated with blynk, the code works, but when I try to send a signal through the IR it simply locks the code and the TX light turns on in arduino uno. Can you help me?

Code here:

`#define BLYNK_PRINT Serial

#include <ESP8266_Lib.h>
#include <BlynkSimpleShieldEsp8266.h>
#include <IRremote.h>

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "4RDUlnc-dkjieGoPEUWDNTara4MjBMXX";

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "Internet Free";
char pass[] = "joao101215";

// or Software Serial on Uno, Nano...
#include <SoftwareSerial.h>
SoftwareSerial EspSerial(3, 2); // RX, TX

// Your ESP8266 baud rate:
#define ESP8266_BAUD 19200

ESP8266 wifi(&EspSerial);
IRsend irsend;

void setup()
{
  // Debug console
  Serial.begin(9600);
  delay(10);

  // Set pinMode
  pinMode(8, OUTPUT);

  // Set ESP8266 baud rate
  EspSerial.begin(ESP8266_BAUD);
  delay(10);

  Blynk.begin(auth, wifi, ssid, pass);
}

BLYNK_WRITE(V1){
  if (param.asInt() == 1){
    Serial.println("botão DOWN");
    irsend.sendNEC(0xF7C03F, 32);
  }
  else{
    Serial.println("botão UP");
  }
}

void loop()
{
  Blynk.run();
}`

ArminJo, 2020-06-06:
Do you have the same error using the [Arduino IRMP library](https://github.com/ukw100/IRMP)?

AnalysIR, 2020-06-06:
I see you are using an ESP8266 - try using the fork for ESP8266, also on Github.

Search for "github esp8266 irremote"

As you will note from the home page of this repository ESP8266 is not currently on the list of supported platforms for IRremote. [IRremote](https://github.com/z3t0/Arduino-IRremote)

PS: Don't forget to close this issue, when you are ready and for future reference, follow the instructions for placing code inside code tags **<>** from the menu above

vgenerato, 2020-06-06:
@AnalysIR When I try to use the esp8266 library it gives the following error

C:\Users\windows\Documents\Arduino\libraries\IRremoteESP8266-master\src\IRrecv.cpp:17:10: fatal error: algorithm: No such file or directory

 #include <algorithm>

          ^~~~~~~~~~~

I'm using an arduino uno with a esp8266 esp-01 board.

AnalysIR, 2020-06-06:
> @AnalysIR When I try to use the esp8266 library it gives the following error

You are not using this library.

That is probably why you should open an issue on that repository rather than this one.

closed 2020-06-06 by vgenerato