Arduino-IRremote/Arduino-IRremote #692: Sending not working for Lasko fan / UNKNOWN code

kylejohnson, 2020-06-11:
**Board:** ARDUINO UNO
**Library Version:** 2.2.3
**Protocol:** UNKNOWN

**Code Block:**
```c
#include <IRremote.h>

IRsend irsend;

void setup(){
}

void loop() {
  irsend.sendNEC(0xA32AB931, 32);
  delay(5000);
}

```

Hi everyone.  I've captured codes for my Lasko M16950 fan using a TSOP4838 and have gotten the following codes repeatably.  I'm attempting to send the POWER command, yet my fan doesn't turn on / off.

I am using a [KY-005](https://arduinomodules.info/ky-005-infrared-transmitter-sensor-module/) IR transmitter, with `signal` connected to `pwm 3`, `+` to `5v` and `-` to `gnd`.

```
# POWER
20:23:49.556 -> Encoding  : UNKNOWN
20:23:49.556 -> Code      : A32AB931 (32 bits)
20:23:49.590 -> Timing[23]: 
20:23:49.624 ->      +1350, - 300     +1350, - 300     + 500, -1150     +1350, - 300
20:23:49.692 ->      +1350, - 300     + 500, -1150     + 550, -1150     + 550, -1100
20:23:49.760 ->      + 550, -1150     + 550, -1150     + 500, -1150     +1350
20:23:49.827 -> unsigned int  rawData[23] = {1350,300, 1350,300, 500,1150, 1350,300, 1350,300, 500,1150, 550,1150, 550,1100, 550,1150, 550,1150, 500,1150, 1350};  // UNKNOWN A32AB931
20:23:49.997 -> 
# SPEED
20:23:50.472 -> Encoding  : UNKNOWN
20:23:50.506 -> Code      : 143226DB (32 bits)
20:23:50.540 -> Timing[23]: 
20:23:50.540 ->      +1350, - 300     +1350, - 300     + 500, -1150     +1400, - 250
20:23:50.608 ->      +1350, - 300     + 500, -1150     + 550, -1150     + 500, -1200
20:23:50.690 ->      + 500, -1150     + 500, -1200     +1300, - 350     + 500
20:23:50.770 -> unsigned int  rawData[23] = {1350,300, 1350,300, 500,1150, 1400,250, 1350,300, 500,1150, 550,1150, 500,1200, 500,1150, 500,1200, 1300,350, 500};  // UNKNOWN 143226DB
20:23:50.930 -> 
# OSC
20:23:51.311 -> Encoding  : UNKNOWN
20:23:51.345 -> Code      : 39D41DC6 (32 bits)
20:23:51.378 -> Timing[23]: 
20:23:51.378 ->      +1350, - 300     +1300, - 300     + 550, -1150     +1350, - 300
20:23:51.446 ->      +1300, - 350     + 500, -1150     + 550, -1150     +1300, - 350
20:23:51.514 ->      + 500, -1150     + 550, -1150     + 550, -1100     + 550
20:23:51.582 -> unsigned int  rawData[23] = {1350,300, 1300,300, 550,1150, 1350,300, 1300,350, 500,1150, 550,1150, 1300,350, 500,1150, 550,1150, 550,1100, 550};  // UNKNOWN 39D41DC6
20:23:51.752 -> 
# TIMER
20:23:52.248 -> Encoding  : UNKNOWN
20:23:52.281 -> Code      : E0984BB6 (32 bits)
20:23:52.315 -> Timing[23]: 
20:23:52.315 ->      +1350, - 300     +1350, - 300     + 550, -1100     +1350, - 300
20:23:52.383 ->      +1350, - 300     + 500, -1150     + 550, -1150     + 550, -1100
20:23:52.460 ->      +1350, - 300     + 550, -1150     + 550, -1150     + 500
20:23:52.540 -> unsigned int  rawData[23] = {1350,300, 1350,300, 550,1100, 1350,300, 1350,300, 500,1150, 550,1150, 550,1100, 1350,300, 550,1150, 550,1150, 500};  // UNKNOWN E0984BB6
20:23:52.700 -> 
```

bengtmartensson, 2020-06-11:
The given signals are not NEC, as you infer. Scrutinizing them with IrScrutinizer yields
```
F12_relaxed: {D=3,F=129,S=1}, beg=0, end=23, reps=1
F12_relaxed: {D=3,F=65,S=1}, beg=0, end=23, reps=1
F12_relaxed: {D=3,F=9,S=1}, beg=0, end=23, reps=1
F12_relaxed: {D=3,F=17,S=1}, beg=0, end=23, reps=1
```
so they are a variant of the F12 protocol, often found in home automation. The present project does not support F12, so you have to send them as raw.

AnalysIR, 2020-06-11:
Here is an example, with timings 'cleaned'...

```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : LASKO12AC :  : D90 : 12 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as necessary to the correct carrier frequency

uint16_t Signal_0_0[] = {1254,418,1254,418,418,1254,1254,418,1254,418,418,1254,418,1254,1254,418,418,1254,418,1254,418,1254,418}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: LASKO12AC, Key:  
```

kylejohnson, 2020-06-11:
Thank you both very much for your replies; this is my first foray in to the world of IR.

My code now looks like this, however I'm still not getting any activity from the fan:

```c
#include <IRremote.h>

IRsend irsend;

void setup(){
}

void loop() {
  int khz = 38;

  uint16_t Signal_0_0[] = {1254,418,1254,418,418,1254,1254,418,1254,418,418,1254,418,1254,1254,418,418,1254,418,1254,418,1254,418};

  irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); 

  delay(5000);  // Cycle power every 5 seconds
```

Am I doing anything obviously incorrect?

Also, I could use a little clarification:  For the `POWER` that I initially intercepted, the `rawData` was `{1350,300, 1350,300, 500,1150, 1350,300, 1350,300, 500,1150, 550,1150, 550,1100, 550,1150, 550,1150, 500,1150, 1350}`, but that isn't what we're sending in the signal.  How did you get your values, from the values that I intercepted?

kylejohnson, 2020-06-11:
I see that `rawData` changes even when `Code` stays the same when pressing the `POWER` button on the remote a few times, so I suspect `rawData` isn't what you actually feed in to `sendRaw`.

AnalysIR, 2020-06-11:
IR receivers will distort the signal timings by up to +/- 200uSecs for each mark or space(pulse).
In addition, IRremote works in only 50 uSecs increments (by default), which can make it worse again.
You will not get identical values from the same signal via a standard IR receiver. This is normal behaviour.

So there is lots of opportunity for the timings to drift.

I 'cleaned' these timings for you using a feature of AnalysIR.

I am not sure if the signal I used was the power one (wasnt the fort on your list).

So try altering the timings on your Power signal to the closest of 1254 or 418 and see how it goes.
{1350,300, ...  ======> {1254,418...., etc

If you still have issues, post some detaisl on your IR emitter circuit (schematic / photo etc)

kylejohnson, 2020-06-11:
Thank you again for your help.  Sounds like I might need to pick up a copy of AnalysIR once I get the hang of this, as this is only the first of a few IR projects that I have planned for my house.

Knowing that you didn't "clean" the `POWER` code, (the fourth on the list was the `TIMER` code), I manually turned on the fan, which enables timer functionality, and then pointed my arduino at it - to no effect - using the code above in 
https://github.com/z3t0/Arduino-IRremote/issues/692#issuecomment-642675412

I've also tried the modified `POWER` code (`1254,418, 1254,418, 418,1254, 1254,418, 1254,418, 418,1254, 418,1254, 418,1254, 418,1254, 418,1254, 418,1254, 1254`), to no effect.

[image]

kylejohnson, 2020-06-11:
I've just done another small test - intercepted the `POWER` code for my Vizio tv
```
11:17:35.882 -> Encoding  : NEC
11:17:35.922 -> Code      : 20DF10EF (32 bits)
11:17:35.962 -> Timing[67]: 
11:17:35.962 ->      +9100, -4350     + 700, - 400     + 700, - 450     + 650, -1600
11:17:36.042 ->      + 650, - 450     + 700, - 450     + 650, - 450     + 700, - 400
11:17:36.122 ->      + 700, - 450     + 700, -1550     + 650, -1600     + 650, - 450
11:17:36.202 ->      + 700, -1550     + 700, -1550     + 650, -1600     + 650, -1550
11:17:36.242 ->      + 700, -1550     + 700, - 450     + 650, - 450     + 700, - 450
11:17:36.322 ->      + 700, -1550     + 650, - 450     + 700, - 400     + 700, - 450
11:17:36.402 ->      + 650, - 450     + 700, -1550     + 700, -1550     + 700, -1550
11:17:36.482 ->      + 650, - 450     + 700, -1550     + 700, -1550     + 700, -1550
11:17:36.562 ->      + 650, -1600     + 650
11:17:36.562 -> unsigned int  rawData[67] = {9100,4350, 700,400, 700,450, 650,1600, 650,450, 700,450, 650,450, 700,400, 700,450, 700,1550, 650,1600, 650,450, 700,1550, 700,1550, 650,1600, 650,1550, 700,1550, 700,450, 650,450, 700,450, 700,1550, 650,450, 700,400, 700,450, 650,450, 700,1550, 700,1550, 700,1550, 650,450, 700,1550, 700,1550, 700,1550, 650,1600, 650};  // NEC 20DF10EF
11:17:36.962 -> unsigned int  data = 0x20DF10EF;
```

And then tried sending that code - again, to no effect.  I'd say that perhaps the KY-005 module is broken, but I've tried three of them at this point, so that seems unlikely.

```c
#include <IRremote.h>

IRsend irsend;

void setup()
{
}

void loop() {
 irsend.sendNEC(0x20DF10EF, 32);
	delay(5000); //5 second delay between each signal burst
}
```

AnalysIR, 2020-06-11:
It could be your emitter. Is there a transistor on the module (photo/link?).  If not the power will be very weak.

Try moving the module closer to the receiver on your TV / AC unit & point the round top of the IR LED at the receiver.

kylejohnson, 2020-06-11:
Looks like the module only has a resistor on it: https://arduinomodules.info/wp-content/uploads/KY-005_Fritzing_custom_part_image-108x240.png
I did try moving the module close - up to about 2 feet away (as long as the wires will reach atm).

AnalysIR, 2020-06-11:
link doesnt work(for me), but I found it on that site drectly.

It doesnt even have a resitor => really poor. 

So try a resistor (100R or 68R) between the module & pin 3.

OK, I found a data sheet....  [link](https://datasheetspdf.com/pdf-file/1402020/Joy-IT/KY-005/1)

There is no 5V connection at all. You may have blown the IR LEd by connecting 5V&GND???? not 100% sure.

Follow the wiring via the link above & using hte resitors I mentioned above will get you longer range, but 220R will also work.

So it looks like your issue is Hardware based & not related to the library.

kylejohnson, 2020-06-11:
You're right, no resistor.  The board does say R1, which I mistake for a resistor, but it is there if YOU want to add your own resistor.  Looks like we're making progress.  After using a standard IR LED w/ a resistor, I can use the NEC code to turn on and off my TV.

When I go back to `sendRaw` and try to turn on and off my fan, though, no luck.

I've grabbed 6 samples of pressing the `POWER` button, averaged them, and then tried sending the averaged data, to no effect.
```
1200,400,1250,400,400,1300,1200,400,1250,400,400,1300,400,1250,400,1300,400,1250,450,1250,400,1250,1250
1350,300,1350,300,500,1150,1350,300,1350,300,500,1200,500,1150,550,1150,500,1150,550,1150,550,1150,1350
1350,300,1350,300,500,1150,1350,300,1350,300,550,1100,550,1150,550,1100,550,1150,500,1200,550,1100,1350
1300,350,1250,350,500,1200,1300,300,1300,350,500,1200,500,1200,450,1200,500,1200,500,1150,500,1200,1250
1300,300,1350,300,500,1200,1300,300,1350,300,500,1200,500,1200,500,1150,500,1200,500,1150,550,1150,1350
1300,350,1250,400,450,1200,1300,350,1250,400,450,1250,450,1200,450,1250,500,1150,500,1200,500,1200,1250

1300,333,1300,342,475,1200,1300,325,1308,342,483,1208,483,1192,483,1192,492,1183,500,1183,508,1175,1300
```

AnalysIR, 2020-06-11:
Did you try replacing the numbers as I suggested earlier?
What value resistor did you use?

kylejohnson, 2020-06-11:
Yup, just tried those numbers again.  The `TIMER` only works when the fan is already on, so I power on the fan manually.

Went with a 330ohm resistor per [this sparkfun tutorial](https://learn.sparkfun.com/tutorials/ir-communication/all)

```c
#include <IRremote.h>

IRsend irsend;

void setup(){
}

void loop() {
  int khz = 38;

  uint16_t modified_timer[] = {1254,418,1254,418,418,1254,1254,418,1254,418,418,1254,418,1254,1254,418,418,1254,418,1254,418,1254,418}; //AnalysIR Batch Export (IRremote) - RAW

  irsend.sendRaw(modified_timer, sizeof(modified_timer)/sizeof(int), khz); 

  delay(5000);
}
```

This codes DOES work to turn my TV on and off:
```c
#include <IRremote.h>

IRsend irsend;

void setup(){
}

void loop() {
  irsend.sendNEC(0x20DF10EF, 32);
  delay(5000);
}
```

bengtmartensson, 2020-06-12:
Not that I want to stop you from buying AnalysIR, but you can achieve this (and much more) using free software, [IrScrutinizer](http://www.harctoolbox.org/IrScrutinizer.html), [download](https://github.com/bengtmartensson/IrScrutinizer/releases/latest). I used it to automatically generate a [sketch for your fan](https://gist.github.com/bengtmartensson/ef8f12e070e37bb1047ef88410fbd86f) (export as Arduino Raw). It supports both the current project, IRLib, and my own [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino).

AnalysIR, 2020-06-12:
If you have more 330R resistors available, wire up 2 or 3 in parallel to get more IR power. (gets you a 165R or 110R resistor)
Some background info from our blog: [Driving an Infrared Led directly from an Arduino Pin](https://www.analysir.com/blog/2014/10/03/driving-infrared-led-directly-arduino-pin/)

If that still does not work, try the following sketch from our blog, which will get a much more acurate capture of your signals. _(and post the output)_
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

kylejohnson, 2020-06-12:
I threw in a 100R resistor, but that doesn't seem to help either.  I don't think it's a power issue - the arduino can control my TV 8 ft away, but not the fan that is 6 in away.

I ran the sketch that you linked to (thank you); here's the output from 5 separate, individual presses of the power button.  The TSOP4838 is connected to 5v, gnd, and pin 2 for signal, on an arduino uno rev 3.
```
Press the button on the remote now - once only

Raw: (143) 1332, -328, 1324, -328, 492, -1196, 1324, -332, 1320, -360, 468, -1224, 472, -1204, 488, -1204, 496, -1204, 492, -1204, 492, -1204, 1328, -6868, 1328, -328, 1324, -328, 492, -1196, 1324, -332, 1324, -340, 488, -1204, 492, -1204, 492, -1204, 492, -1204, 492, -1204, 492, -1204, 1328, -6852, 1328, -328, 1328, -328, 492, -1196, 1324, -328, 1300, -364, 492, -1200, 492, -1204, 492, -1204, 492, -1204, 492, -1204, 496, -1200, 1328, -6864, 1328, -328, 1328, -328, 492, -1196, 1324, -332, 1324, -336, 492, -1200, 492, -1204, 492, -1204, 492, -1204, 496, -1204, 492, -1200, 1328, -6856, 1324, -328, 1328, -328, 492, -1196, 1324, -328, 1328, -336, 492, -1200, 492, -1204, 492, -1204, 492, -1204, 492, -1204, 492, -1204, 1328, -6880, 1332, -324, 1328, -328, 492, -1196, 1324, -328, 1328, -332, 496, -1200, 492, -1204, 492, -1204, 492, -1204, 492, -1204, 496, -1200, 1328, 

Press the button on the remote now - once only

Raw: (97) 300, -7124, 1332, -324, 1332, -324, 496, -1196, 1324, -324, 1332, -332, 496, -1196, 500, -1196, 500, -1200, 496, -1200, 496, -1204, 496, -1200, 1328, -6868, 1332, -324, 1328, -324, 496, -1192, 1332, -324, 1332, -328, 496, -1200, 496, -1200, 496, -1200, 496, -1200, 500, -1200, 496, -1200, 1332, -6852, 1332, -324, 1328, -324, 496, -1192, 1328, -328, 1328, -332, 496, -1196, 500, -1196, 500, -1196, 500, -1200, 496, -1200, 496, -1200, 1332, -6872, 1332, -320, 1332, -324, 496, -1192, 1332, -324, 1328, -332, 496, -1200, 496, -1196, 500, -1200, 496, -1200, 496, -1200, 496, -1200, 1328, 

Press the button on the remote now - once only

Raw: (119) 1336, -320, 1332, -324, 496, -1192, 1328, -328, 1328, -332, 496, -1196, 500, -1196, 500, -1200, 496, -1200, 500, -1200, 496, -1200, 1328, -6868, 1332, -320, 1332, -324, 496, -1192, 1332, -324, 1328, -332, 500, -1196, 496, -1200, 496, -1200, 496, -1200, 500, -1200, 496, -1196, 1332, -6852, 1332, -324, 1332, -324, 496, -1192, 1328, -324, 1332, -332, 496, -1196, 496, -1200, 496, -1200, 500, -1196, 500, -1200, 496, -1200, 1328, -6868, 1328, -324, 1332, -320, 500, -1192, 1328, -328, 1328, -332, 496, -1196, 496, -1200, 496, -1200, 496, -1200, 500, -1196, 500, -1196, 1332, -6872, 1332, -320, 1332, -324, 496, -1192, 1328, -328, 1328, -332, 496, -1200, 496, -1196, 500, -1196, 500, -1196, 500, -1196, 500, -1196, 1332, 

Press the button on the remote now - once only

Raw: (119) 1336, -324, 1332, -320, 500, -1188, 1332, -324, 1328, -336, 496, -1196, 500, -1196, 500, -1200, 496, -1200, 496, -1204, 496, -1200, 1332, -6864, 1332, -324, 1328, -324, 500, -1188, 1332, -324, 1332, -332, 496, -1196, 496, -1200, 500, -1196, 500, -1200, 496, -1200, 500, -1196, 1332, -6852, 1332, -324, 1332, -320, 500, -1192, 1328, -324, 1332, -332, 496, -1196, 496, -1200, 496, -1200, 496, -1200, 500, -1200, 496, -1200, 1328, -6864, 1332, -324, 1332, -324, 496, -1188, 1332, -324, 1332, -332, 496, -1196, 496, -1200, 496, -1200, 496, -1200, 500, -1196, 500, -1196, 1332, -6872, 1332, -324, 1328, -324, 496, -1192, 1332, -324, 1328, -332, 496, -1196, 500, -1196, 496, -1200, 472, -1224, 476, -1220, 476, -1220, 1308, 

Press the button on the remote now - once only

Raw: (119) 1332, -324, 1328, -324, 500, -1192, 1328, -324, 1332, -332, 496, -1196, 500, -1196, 500, -1200, 496, -1204, 496, -1200, 496, -1200, 1332, -6864, 1332, -324, 1332, -324, 496, -1192, 1328, -328, 1328, -332, 496, -1200, 496, -1200, 496, -1200, 496, -1200, 500, -1200, 496, -1200, 1328, -6856, 1332, -320, 1332, -324, 496, -1192, 1328, -328, 1328, -332, 496, -1196, 500, -1196, 500, -1200, 496, -1200, 496, -1200, 496, -1200, 1332, -6864, 1332, -320, 1332, -324, 496, -1192, 1328, -328, 1328, -332, 496, -1196, 500, -1196, 500, -1196, 500, -1200, 496, -1200, 496, -1200, 1328, -6872, 1328, -324, 1332, -324, 496, -1192, 1328, -324, 1332, -332, 496, -1196, 496, -1200, 496, -1200, 496, -1200, 500, -1196, 500, -1196, 1332, 

Press the button on the remote now - once only
```

kylejohnson, 2020-06-12:
I "scrutinized" the first RAW signal with IrScrutinizer, then exported it as Arduino Raw, and got the same results as before.
I then exported it as Infrared4Arduino, and wham!  It works!
Here's the working arduino code with the `POWER` command:
```c
// This Arduino sketch was automatically generated by IrScrutinizer.

// It supports:
//
// * Infrared4Arduino https://github.com/bengtmartensson/Infrared4Arduino,
//                    http://www.harctoolbox.org/Infrared4Arduino.html

// For problems, bugs, and suggestions, please open an issue at
// https://github.com/bengtmartensson/IrScrutinizer/issues

// This file contains C identifiers which have been translated from command names.

#include <IrSenderPwm.h>

// Constants
static const long BAUD = 115200UL; // Change if desired

IrSender *irsend = IrSenderPwm::getInstance(true);

// Command #1: IrScrutinizer scrutinized signal
// Protocol: f12_relaxed, Parameters: D=3U F=129U S=1U
const microseconds_t repeat_IrScrutinizer_scrutinized_signal[] PROGMEM = { 1254U, 333U, 1254U, 333U, 491U, 1254U, 1254U, 333U, 1254U, 333U, 491U, 1254U, 491U, 1254U, 491U, 1254U, 491U, 1254U, 491U, 1254U, 491U, 1254U, 1254U, 6864U };

static void sendRaw(const microseconds_t intro_P[], size_t lengthIntro, const microseconds_t repeat_P[],
             size_t lengthRepeat, frequency_t frequency, unsigned times) {
    microseconds_t intro[lengthIntro];
    microseconds_t repeat[lengthRepeat];
    memcpy_PF(intro, (uint_farptr_t) intro_P, sizeof(microseconds_t) * lengthIntro);
    memcpy_PF(repeat, (uint_farptr_t) repeat_P, sizeof(microseconds_t) * lengthRepeat);

    IrSignal irSignal(intro, lengthIntro, repeat, lengthRepeat, NULL, 0U, frequency);
    irsend->sendIrSignal(irSignal, times);
}

void setup() {
    Serial.begin(BAUD);
    Serial.setTimeout(60000UL);
}

// A pretty silly main loop; feel free to replace it by something smarter.
void loop() {
    Serial.println(F("Enter number of signal to send (1 .. 1)"));
    long commandno = Serial.parseInt();
    Serial.println(F("Enter number of times to send it"));
    long times = Serial.parseInt();
    switch (commandno) {
    case 1L:
        sendRaw(NULL, 0U, repeat_IrScrutinizer_scrutinized_signal, 24U, 38000U, times);
    default:
        Serial.println(F("Invalid number entered, try again"));
        break;
    }
}
```

kylejohnson, 2020-06-16:
I'd like to thank everyone for all of their help with this.
In the end, for reasons unknown to me, I am able to get Infrared4Arduino to control the fan, but not IRremote

closed 2020-06-16 by kylejohnson

kylejohnson, 2020-06-17:
Just to circle back a little bit for completeness sake, I went ahead and setup a a GIRS client for IrScrutinizer.  Captured the `POWER` signal for the fan, and then exported it as raw, infrared4arduino (i4a), and irremote.  The signal worked for each exported format, except for irremote, with a small modification.  I need to send the signal at least twice for the fan to respond.

The exported raw and i4a arduino sketches prompt you for the number of times to send the signal, while the irremote sketch does not, so I just added in a second `sendRaw` call in the sketch.

tl;dr: This was a classic PEBKAC issue.  Learned a lot.  Below is the sketch for a Lasko M16950 fan to toggle the POWER.

```c
// This Arduino sketch was automatically generated by IrScrutinizer.

// It supports:
//
// * IRremote https://github.com/z3t0/Arduino-IRremote
//            http://z3t0.github.io/Arduino-IRremote/

// For problems, bugs, and suggestions, please open an issue at
// https://github.com/bengtmartensson/IrScrutinizer/issues

// This file contains C identifiers which have been translated from command names.

/////////////////////////////////////////////////////

#include <IRremote.h>

// Constants
static const long BAUD = 115200UL; // Change if desired

// Global variables
IRsend irsend;

// Command #1: IrScrutinizer scrutinized signal
// Protocol: f12_relaxed, Parameters: D=3U F=129U S=1U
const uint16_t repeat_IrScrutinizer_scrutinized_signal[] PROGMEM = { 1255U, 405U, 1255U, 405U, 405U, 1255U, 1255U, 405U, 1255U, 405U, 405U, 1255U, 405U, 1255U, 405U, 1255U, 405U, 1255U, 405U, 1255U, 405U, 1255U, 1255U, 6888 };

static void sendRaw(const uint16_t data_P[], size_t length, uint16_t frequency) {
    uint16_t data[length];
    memcpy_PF(data, (uint_farptr_t) data_P, sizeof(uint16_t) * length);
    irsend.sendRaw(data, length, frequency);
}

void setup() {
    Serial.begin(BAUD);
    Serial.setTimeout(60000UL);
}

// A pretty silly main loop; feel free to replace it by something smarter.
void loop() {
    Serial.println(F("Enter number of signal to send (1 .. 1)"));
    long commandno = Serial.parseInt();
    switch (commandno) {
    case 1L:
        sendRaw(repeat_IrScrutinizer_scrutinized_signal, 24U, 38U);
        sendRaw(repeat_IrScrutinizer_scrutinized_signal, 24U, 38U);
        break;
    default:
        Serial.println(F("Invalid number entered, try again"));
        break;
    }
}```