Arduino-IRremote/Arduino-IRremote #696: Samsung Soundbar Remote does not decode (no output from any example)

zaneclaes, 2020-06-26:
**Board:** ARDUINO NANO
**Library Version:** 2.2.3
**Protocol:** RC6 (??)

**Code Block:**
I have only used examples.

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant 

I have tried all of the RX examples. Two of my remotes work as-expected. The remotes for my 2x Samsung Sound Bars (two different models) both appear to not decode at all. However, [this forked IR library](https://github.com/NicoHood/IRLremote) does detect them if I use the IRHash protocol. However, it is not capable of sending, so it does not meet my needs. 

Hoping I can find a way to clone my Samsung sound bar remote, so I can TX to it.

AnalysIR, 2020-06-26:
post the output you get from the example sketch **IRrecvDumpV2** for the unknown signal.

Specific model numbers for the devices & remote is also useful.

zaneclaes, 2020-06-26:
Thanks for the exceptionally fast reply.

I don't see anything from that example, either. I can press any button on the remote and see the lights flashing on the receiver. But nothing is printed to the console. Again, it does work with other remotes.

Actually, that's not entirely true. Sometimes I do see output. But only rarely, and if I press the remote right after using a different remote. I suspect that this is actually just noise, but here it is anyways:
```
12:56:54.925 -> Encoding  : UNKNOWN
12:56:54.925 -> Code      : CA31DA45 (32 bits)
12:56:54.960 -> Timing[77]: 
12:56:54.995 ->      +4400, -4450     + 500, - 500     + 450, - 550     + 400, - 600
12:56:55.064 ->      + 400, - 550     + 450, -1550     + 400, -1550     + 450, - 550
12:56:55.132 ->      + 450, - 550     + 400, -1550     + 450, -1550     + 400, -1550
12:56:55.200 ->      + 450, -1550     + 400, - 600     + 400, - 550     + 450, - 550
12:56:55.267 ->      + 450, - 550     + 400, -4500     + 500, - 500     + 450, - 550
12:56:55.368 ->      + 400, - 600     + 400, - 550     + 450, - 550     + 450, - 550
12:56:55.435 ->      + 400, - 600     + 400, - 550     + 450, -1550     + 400, -1550
12:56:55.503 ->      + 450, -1550     + 450, - 550     + 400, -1550     + 450, -1550
12:56:55.570 ->      + 400, -1550     + 450, -1550     + 400, - 550     + 450, - 550
12:56:55.637 ->      + 450, - 550     + 400, -1550     + 450
12:56:55.707 -> unsigned int  rawData[77] = {4400,4450, 500,500, 450,550, 400,600, 400,550, 450,1550, 400,1550, 450,550, 450,550, 400,1550, 450,1550, 400,1550, 450,1550, 400,600, 400,550, 450,550, 450,550, 400,4500, 500,500, 450,550, 400,600, 400,550, 450,550, 450,550, 400,600, 400,550, 450,1550, 400,1550, 450,1550, 450,550, 400,1550, 450,1550, 400,1550, 450,1550, 400,550, 450,550, 450,550, 400,1550, 450};  // UNKNOWN CA31DA45
12:56:56.114 -> 
```

The model number on the bottom of the remote itself is `AH59-02631A`. One of the sound bars is a `HW-J550` (2015 model). I believe the other bar is the same model but ~2 years newer.

bengtmartensson, 2020-06-26:
Try [this](https://gist.github.com/bengtmartensson/a8a10bc6adb969ce5d59c5f2697b095d).

ArminJo, 2020-06-26:
Or [this ](https://github.com/ukw100/IRMP#allprotocol-example)😀

zaneclaes, 2020-06-26:
> Try [this](https://gist.github.com/bengtmartensson/a8a10bc6adb969ce5d59c5f2697b095d).

I'm guessing this is supposed to work on the TX side without needing to record the RX codes, which seems great. I have tried using power on, volume up, volume down, etc. All of them show a light on the actual IR breakout, indicating transmission, but the sound bar does not do anything. I have not tried switching from IRRemote to a different library, though.

> Or this 😀

@ArminJo same results as with the RX on IRremote. I see output from other remotes, but not from the soundbar remote. Despite seeing the light on the RX breakout.

zaneclaes, 2020-06-26:
Huh — @ArminJo I just saw the same behavior above, where suddenly I did get one lonely line of output from the Samsung remote. I can't seem to repro it, but here it is, nestled in the middle:

```
13:29:20.070 -> P=NEC  A=0x3000 C=0x4E
13:29:20.342 -> P=NEC  A=0x3000 C=0x4E
13:29:20.581 -> P=NEC  A=0x3000 C=0x4E
13:32:54.643 -> P=SAMSUNG  A=0xF30 C=0x70
13:33:28.981 -> P=SIEMENS  A=0x512 C=0x295
13:33:29.049 -> P=SIEMENS  A=0x512 C=0x295 R
```

ArminJo, 2020-06-26:
This looks like a weak or disturbed input signal. The NEC recording is stable 😁 .
Try to put the remote just in front of the receiver, and / or remove all protocols except NEC and Samsung from the include, see [OneProtocol example](https://github.com/ukw100/IRMP/tree/master/examples/OneProtocol).

zaneclaes, 2020-06-26:
Hah. Wow. Bingo. I replaced the battery and it all works fine now _sigh_.

Thanks, all!

closed 2020-06-26 by zaneclaes