Arduino-IRremote/Arduino-IRremote #697: Remote sends out signal twice

XGamerYU, 2020-06-28:
**Board:** ARDUINO Pro Micro
**Library Version:** 2.5.0
**Protocol:** none

**Code Block:**
```c

#include <IRremote.h> 
void loop() {
  if (irrecv.decode(&results)) 
    {
      Serial.println(results.value, DEC); // Print the Serial 'results.value'
      delay(40);
        irrecv.resume();   // Receive the next value
    }
     
}
     
.....

```

So, my remote sends out the same signal twice on a single button press, which means that my function (currently - emulating HID keyboard keys) will run twice, which is not really good for me. 
I solved it by adding a small delay in between the beginning of the IF statement, and the irrecv.resume() lines, now I recieve only one of the codes. 

But, since I'm emulating keyboard keys, the actions have to be fast, and the delay just slows my program down, not really wanted when using remotes. 

Is there a way to ignore the second signal that is sent? 
(note: the code above is just a part of the code - posting my whole code would be pointless since most of it is for the keyboard emulation. 

Thanks in advance.

AnalysIR, 2020-06-28:
A number of options:
- just put some code in your sketch to ignore a 2nd signal withing a certain time window like 500mSecs
- Change the 40mSec delay to 500 or 1000 etc
- You could poll the IR Rx pin and wait until it is high for a similar period as above...before resuming IR rx.
- In your HID code just ignore any key press that happens too quickly...similar to above
- etc etc etc

ArminJo, 2020-06-28:
Use [IRMP](https://github.com/ukw100/IRMP/blob/master/pictures/AllProtocol_SerialMonitor.png) and check the repeat flag.

closed 2020-07-03 by ArminJo