Arduino-IRremote/Arduino-IRremote #698: esp32 core - M5stickC support

supaplextor, 2020-07-03:
Vendor: https://m5stack.com/collections/m5-core/products/stick-c?variant=17203451265114

[image]

### Board
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [x ] ESP32 board - M5stickC unit
* [ ] Teensy board
* [ ] Other - please specify

### Protocol
* [x ] TBD

**Code Block:**

https://github.com/m5stack/M5StickC/blob/master/examples/Unit/IR/IR.ino

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant

supaplextor, 2020-07-03:
I think pin9 is transmit for the internal IR. See also https://github.com/m5stack/M5StickC/issues/13 - the first example uses the grove port next to the usb-c connector. I'd rather default to gpio9 and save the grove and other gpios for other things. Thanks!

ArminJo, 2020-07-03:
You are welcome.

closed 2020-07-03 by ArminJo