Arduino-IRremote/Arduino-IRremote #700: All IR decoding is broken in the master branch

pvdbrand, 2020-07-07:
The commit 76f233faddfbaba9bf27d318bc22a0d7fb71e772 (13 days ago) seems to have completely broken all IR decoding, on all boards and for all protocols.

Old code in MATCH_MARK() in IRremote.cpp:
```
bool passed = ((measured_ticks >= TICKS_LOW (desired_us + MARK_EXCESS))
                && (measured_ticks <= TICKS_HIGH(desired_us + MARK_EXCESS)));
```
New code since that commit:
```
    int compensated_ticks = measured_ticks - MARK_EXCESS; // compensate for marks exceeded by demodulator hardware
    bool passed = ((compensated_ticks >= TICKS_LOW(desired_us)) && (compensated_ticks <= TICKS_HIGH(desired_us)));
```

This code is not equivalent and it will make all matches fail. There are several more changes in that commit that I have not checked. They may be wrong too.

ArminJo, 2020-07-07:
Why is it not equivalent? Please prove.

AnalysIR, 2020-07-07:
Is **MARK_EXCESS** defined as  uSecs?
Is compensated_ticks  defined as ticks?

ArminJo, 2020-07-07:
Thanks, thats the fault!

ArminJo, 2020-07-07:
@pvdbrand 🥇  
Thank you very much for reporting this bug!!!

commit 2bce3d69 (2020-07-07): Renamed MARK_EXCESS and USECPERTICK, fixes #700
commit 0268a0e7 (2020-07-07): Renamed MARK_EXCESS and USECPERTICK, fixes #700
commit 07e9bbad (2020-07-07): Renamed MARK_EXCESS and USECPERTICK, fixes #700
commit a8a403cf (2020-07-07): Renamed MARK_EXCESS and USECPERTICK, fixes #700
closed 2020-07-07 by ArminJo