Arduino-IRremote/Arduino-IRremote #701: Difference between IRremote library on Arduino IDE and on github.com/z3t0/Arduino-IRremote

NDC-SC, 2020-07-08:
There are differences in the contents of the IRremote folders resulting from downloading the IRremote library from within the Arduino IDE or when uploading the zip file from z3t0/Arduino-IRremote.
With an ESP32 microcontroller and the z3t0/Arduino-IRremote version of the IRremote library, the example sketch IRrecvDemo compiles.
In contrast, with an ESP32 microcontroller and the downloaded Arduino IDE IRremote library, the example sketch IRrecvDemo fails to compile with the message "SEND_PIN was not declared in this scope"
Is the IRremote library downloaded within the Arduino IDE out-of-date ?

ArminJo, 2020-07-08:
As you can see when you look at the version number.

closed 2020-07-08 by ArminJo

NDC-SC, 2020-07-08:
In that case, it would be useful to update the version on the Arduino IDE.

ArminJo, 2020-07-08:
Indeed?
But you do not want a halfway built release, or do you?
But thanks for testing 👍