Arduino-IRremote/Arduino-IRremote #704: NEC Repeat Implementation

decrazyo, 2020-07-23:
**Board** Arduino ATmega328* board (Nano)
**Library Version:** 2.5.0
**Protocol** NEC

I'm trying to use the IRrecord example to control an HDMI switch which uses the NEC protocol.
From what I can tell, my HDMI switch will only respond to a received code if it is followed by at least 1 repeat.
That gave me some problems since ```sendNEC``` doesn't seem to implement NEC-style repeats.
That might also explain things like issue #624.

Code snippet from IRrecord.ino
```c
void sendCode(int repeat) {
    if (codeType == NEC) {
        if (repeat) {
            irsend.sendNEC(REPEAT, codeLen);
            Serial.println("Sent NEC repeat");
        } else {
            irsend.sendNEC(codeValue, codeLen);
            Serial.print("Sent NEC ");
            Serial.println(codeValue, HEX);
        }
```

Code snippet from IRremote.h
```c
/**
 * Decoded value for NEC when a repeat code is received
 */
#define REPEAT 0xFFFFFFFF
```

I took oscilloscope readings of my original remote and the arduino.
You can see that the repeats are significantly different.
https://docs.google.com/spreadsheets/d/e/2PACX-1vSvfhoar-ez4OvHhL8twuyzsnmfHhVmasBg8zXqmSPthDOjRnfU_ik1-G3iwPxZ_jx46KDKbHV0Hw8y/pubhtml?gid=1144927176&single=true

As a test, I modified IRrecord.ino to send repeats as described [here](https://www.sbprojects.net/knowledge/ir/nec.php) and everything started working correctly.

I'd suggest adding NEC-style repeats and a ```repeat``` parameter to ```sendNEC``` just like what ```sendJVC``` has.

ArminJo, 2020-07-23:
Can you please test the version just commited to the repo?
Thanks

decrazyo, 2020-07-24:
That manged to fix the issue that I was having

closed 2020-08-03 by ArminJo