Arduino-IRremote/Arduino-IRremote #708: DisableIRIn() command doesn't work Attiny85

kidx9, 2020-08-07:
Board: Attiny85
Protocol: Not Sure

Code Block:
`#include < IRremote.h >

const int RECV_PIN = PB2; // Recving pin for IR light
volatile IRrecv irrecv(RECV_PIN); // Setting up the IR decoding
volatile decode_results results; // More set up

void setup() {
	// put your setup code here, to run once:
	pinMode(PB3, OUTPUT);
	irrecv.enableIRIn();
	irrecv.disableIRIn();
}

void loop() {

	unsigned long i = millis();

	delay(100);

	if (millis() > i) {
		digitalWrite(PB3, HIGH);
	}
}`

I wrote this simple code to test if millis() worked properly with IRremote.h. I soon discovered that it doesn't work if I do enableIRIn(). I bellieve this is because the receiving software uses the same clock as millis(). I was wondering if I could use the disableIRIn() command so that I could only have the receiver on when I need it so that I could use millis() the rest of the time. I ran the test above to see if the disable command allowed me to use millis(). It doesn't let me use millis(). I belleive this is a bug because I can use millis() if I never enable, so disabling it should bring me back to the state I was in before enabling and I should be able to use it like before. That does not happen leading me to think this is a bug. Please let me know if I am just doing something wrong.

BTW, when millis() didn't work the if statement condition returned false and the led did not light.

ArminJo, 2020-08-07:
There is not such simple thing than a bug here 😁 .
I depends if you use the [ATTinyCore](https://github.com/SpenceKonde/ATTinyCore) or the [Digispark core](https://github.com/ArminJo/DigistumpArduino). they use different timers for millis(). The Digispark uses timer 1 for millis(), so your example should work with it.
Timer 1 us currently not supported as IRremote timer for ATtin85, try [IRMP](https://github.com/ukw100/IRMP), there ATtiny85 is supported better.

kidx9, 2020-08-07:
I am not sure what core I am using but it still stands that the disable command isn't functioning as expected. It does not undo all of the affects of the enable command. I will try IRMP though thanks!

kidx9, 2020-08-07:
I found the issue. I looked through the source code and discovered that there are two bits changed by enableIRIn() that are not reset by disableIRIn(). I changed these two bits back to their original value whenever I ran disableIRIn() and it fixed it. Thanks so much for the help! 

New Code:
`
#include <IRremote.h>

const int RECV_PIN = PB2;        // Recving pin for IR light
volatile IRrecv irrecv(RECV_PIN);// Setting up the IR decoding
volatile decode_results results; // More set up

void setup() {
  // put your setup code here, to run once:
  pinMode(PB3, OUTPUT); 
  irrecv.enableIRIn();
  irrecv.disableIRIn();
  TCCR0A &= ~(1 << WGM01);  // These two lines are rewriting bits that are changed in enableIRIn() but not reset in disableIRIn()
  TCCR0B &= ~(1 << CS00);
}

void loop() {
  
unsigned long i = millis();

delay(100);

if (millis() > i) { 
  digitalWrite(PB3, HIGH);
}
}
`
I think there are a couple of other things done by enableIRIn() that are not undone by disableIRIn() and I think it would be helpful if they were changed back.

enableIRIn() code:
`void IRrecv::enableIRIn() {
// Interrupt Service Routine - Fires every 50uS
    cli();
    // Setup pulse clock timer interrupt
    // Prescale /8 (16M/8 = 0.5 microseconds per tick)
    // Therefore, the timer interval can range from 0.5 to 128 microseconds
    // Depending on the reset value (255 to 0)
    timerConfigNormal();

    // Timer2 Overflow Interrupt Enable
    TIMER_ENABLE_INTR;

    TIMER_RESET;

    sei();
    // enable interrupts

    // Initialize state machine variables
    irparams.rcvstate = IR_REC_STATE_IDLE;
    irparams.rawlen = 0;

    // Set pin modes
    pinMode(irparams.recvpin, INPUT);
}`

disableIRIn() code:
`void IRrecv::disableIRIn() {
    cli();
    TIMER_DISABLE_INTR;
    sei();
    // enable interrupts
}`

As you can see there are a couple things in enableIRIn() that are not disabled by disableIRIn(), like the command timerConfigNormal(). Thanks again for the help!

ArminJo, 2020-08-08:
Best is to save the TCCR0A , TCCR0B etc registers before enabling. But it is seldom required, since you want to receive IR signals continuously.
Everyone is invited to add support for ATtiny Timer1. 🥇

closed 2020-08-08 by ArminJo