Arduino-IRremote/Arduino-IRremote #710: Compilation error IRRemote with Tone...

JulesTopart, 2020-08-15:
I can't manage to compile IRRemote with Tone library on LilypadUSB (32u4)

I got : 
`Tone.cpp : In function timer0_pin_port:`
`Multiple definition of __vector_32`
`libraries\IRremote\IRremote.cpp.o (symbol from plugin):(.text+0x0): first defined here`

AnalysIR, 2020-08-15:
It looks like both libraries are using the Timer0 interrupt vector - which would cause a problem like this.

So search for a tone librbry that doesnt use Timer0....like **TimerFreeTone**

ArminJo, 2020-08-16:
Try to enable another timer in IRremoteBoardDefs.h line 198 ff.

JulesTopart, 2020-08-16:
Timer change fixed it obviously. But it was working few months ago...
I need to change this on my production library so I'll try to modify my lib instead of IRremoteBoardDefs.

Thank's for your help !

closed 2020-08-16 by JulesTopart