Arduino-IRremote/Arduino-IRremote #711: Reverse bits of the results.value get the correct key number

shenghongcao, 2020-08-17:
The data received by the infrared receiver is the low bit first, each byte from the receiver  need to be reversed to get the correct key value.

### Board
* [*] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP32 board
* [ ] Teensy board
* [ ] Other - please specify

### Protocol
* [ ] Unknown
* [ ] Aiwa
* [ ] BoseWave
* [ ] Denon
* [ ] Dish
* [ ] JVC
* [ ] Lego
* [ ] LG
* [ ] Mitsubishi
* [ ] Panasonic
* [*] RC5, RC6
* [ ] Samsung
* [ ] Sanyo
* [ ] Sharp
* [ ] Sony
* [ ] Whynter
* [ ] Other - please specify

**Code Block:**
```c

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
    byte tmp = results.value >> 8,result=0;
    for (int i = 0; i <= 7; i++) {
      if (bitRead(tmp, i)) {
        result = result | (1 << (7 - i));
      }
      else {
        result = result & (~(1 << (7 - i)));
      }
    }
    Serial.println(result, HEX);
    irrecv.resume(); // Receive the next value
  }
  delay(100);

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [*] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

ArminJo, 2020-08-17:
Which version do you use? The current one processes MSB first!

shenghongcao, 2020-08-17:
I use example program IRreceDemo.ino of the latest version 2.6.1,   one of the return code is 0xFF847B，0xFF is user code,  0x84 is key code.  I know the right key code is 0x21.   0x84= 0b10000100   and  0x21= 0b00100001 ， I think  reverse the return code to get the correct key code.

The attach files is the detail description.

[IRremote_key_number.pdf](https://github.com/z3t0/Arduino-IRremote/files/5085186/IRremote_key_number.pdf)

ArminJo, 2020-08-17:
Thanks for the explanation of your problem.
Of course you can interprete the codes as you want, but almost the whole world interprete RC5 + 6 as [MSB first](https://www.sbprojects.net/knowledge/ir/rc5.php) + [here ](https://www.sbprojects.net/knowledge/ir/rc6.php).

shenghongcao, 2020-08-17:
I check the signal in oscilloscope, the wave looks like NEC , this is the same with RC5+6.

[image]

shenghongcao, 2020-08-17:
Thanks for your advise.
I just check the [NEC protocol](https://www.sbprojects.net/knowledge/ir/nec.php) , in the protocol the LSB is transmitted first.

ArminJo, 2020-08-17:
Ok you seems to be right. the Nec implementation with MSB first is questionable. I have to do more research.

AnalysIR, 2020-08-17:
>  MSB first is questionable.

Not really...it has been done this way in IRremote from the very beginning & there are lots of users out there using this existing  approach/format. For anyone needing the reverse bit order, its a simple matter of adding the logic for this to their 'new' application. That way there would be no implications for the many exiting implementations. Also, for the majority of existing users the bit order is essentially irrelevant - IMHO. A sub-set of protocols operate on 8-bit/byte boundaries....many do not. 

For my own purposes, I treat the decoded signal as a bit-stream & if I need to extract a particular bit order or pattern, I would typically do that in post processing after the initial decode. (using IRremote & elsewhere)

shenghongcao, 2020-08-17:
The key codes of the infrared remote control I customized are numbered according to a two-dimensional array to facilitate students' memory. If I use an existing program, just write the reverse number during production. The problem is that 5,000 have been produced.

ArminJo, 2020-08-18:
I will prepare a version with LSB first the next days.

bengtmartensson, 2020-08-18:
When Ken Shirriff wrote the original IRremote in 2009 he did not give protocol parametrization much attention.  Just the bits delivered from the decoder/to the encoder, turned out that MSB first was slightly simpler, so MSB-first was used exclusively. 

Possbly this was a bad decision -- at least to call it the same name as some established protocols. It regularly causes confusion in forums like [JP1](http://www.hifi-remote.com/forums/index.php) and [RemoteCentral](http://www.remotecentral.com/cgi-bin/mboard/forums.cgi).

However, there is a lot of codes around, Youtube videos, HowTos etc that all depend on the original parametrization. So please do not "fix" it, you will break an unknown amount of legacy code.  At most, "complement" it with alternative implementations.
 
For implementation of the NEC1 and RC5 protocols using the parametrization the rest of the world uses, see [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino), classes `Nec1Decoder`, `Nec1Renderer`, `RC5Decoder`, `RC5Renderer`. Note that these not just one parameter but several, some having a default value, some toggling, having different lengths.

There is a machine readable data base for IR protocols contained within [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier) (which in its turn is contained within  IrScrutinizer). Automatically generating Arduino codes for these protocols was discussed [here](https://github.com/z3t0/Arduino-IRremote/issues/536). Possibly a new project can be created, consisting of protocols using IRremote as its foundation? Anyone interested?

ArminJo, 2020-09-16:
The NEC version with LSB first is now available. Please activate line //#define USE_NEC_STANDARD

closed 2020-09-16 by ArminJo