Arduino-IRremote/Arduino-IRremote #715: Press and Hold

philspitlerSF, 2020-10-05:
### Board
* [ x] Arduino ATmega328* board (UNO, Nano)

### Protocol

* [x ] Other -NEC

I'm trying to simulate the long press on my projector power button. 

In my arduino code I tried many things and the latest iteration of my code is here.
https://gist.github.com/philspitlerSF/b183236bdfe73f0dea0f7834a568d319

Any ideas?

Thanks

I have done a dump and it looks like this.

```
Protocol=NEC Data=0x1FEFA05 (32 bits)
Timing[67]: 
     +8950, -4450     + 600, - 500     + 600, - 550     + 550, - 550
     + 600, - 550     + 550, - 550     + 600, - 500     + 600, - 550
     + 600, -1600     + 600, -1650     + 550, -1650     + 600, -1600
     + 600, -1650     + 600, -1600     + 600, -1650     + 550, -1650
     + 600, - 500     + 600, -1650     + 600, -1600     + 600, -1650
     + 550, -1650     + 600, -1600     + 600, - 550     + 600, -1600
     + 600, - 550     + 550, - 550     + 600, - 550     + 550, - 550
     + 600, - 500     + 600, - 550     + 600, -1600     + 600, - 550
     + 600, -1600     + 600
unsigned int  rawData[67] = {8950,4450, 600,500, 600,550, 550,550, 600,550, 550,550, 600,500, 600,550, 600,1600, 600,1650, 550,1650, 600,1600, 600,1650, 600,1600, 600,1650, 550,1650, 600,500, 600,1650, 600,1600, 600,1650, 550,1650, 600,1600, 600,550, 600,1600, 600,550, 550,550, 600,550, 550,550, 600,500, 600,550, 600,1600, 600,550, 600,1600, 600};  // Protocol=NEC Data=0x1FEFA05
unsigned int  data = 0x1FEFA05;

Pronto Hex: 0000 006D 0022 0000 0158 00AB 0017 0013 0017 0015 0015 0015 0017 0015 0015 0015 0017 0013 0017 0015 0017 003E 0017 003F 0015 003F 0017 003E 0017 003F 0017 003E 0017 003F 0015 003F 0017 0013 0017 003F 0017 003E 0017 003F 0015 003F 0017 003E 0017 0015 0017 003E 0017 0015 0015 0015 0017 0015 0015 0015 0017 0013 0017 0015 0017 003E 0017 0015 0017 003E 0017 0806 

Protocol=NEC Data=0xFFFFFFFF R (0 bits)
Timing[3]: 
     +8950, -2200     + 600
unsigned int  rawData[3] = {8950,2200, 600};  // Protocol=NEC Data=0xFFFFFFFF R
unsigned int  data = 0xFFFFFFFF;

Pronto Hex: 0000 006D 0002 0000 0158 0055 0017 0806 

```

bengtmartensson, 2020-10-05:
A "long" NEC-signal has to be more than one repeat. Try with 10 or more. Each repeat accounts for 108 ms, and "long" probably means at least one second.

But that is not all. IRremote does not do proper delays between the sends, so you have to add them yourself. A receiving device can be picky about this, and reject the signal, so you have to add appropriate delay()-s.

For a better solution, consider [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino), available in the library manager as "Infrared". The example `Nec1Renderer` does almost exactly what you want; use Nex1Renderer::newIrSignal(128,95) instead. )

philspitlerSF, 2020-10-05:
@bengtmartensson thanks for your help,  to repeat the command 10x would it be something like this:

```
IrSender.sendNEC(0x1FEFA05, 32); //power off
IrSender.sendNEC(0xFFFFFFFF, 0); //repeat
IrSender.sendNEC(0xFFFFFFFF, 0); //repeat
IrSender.sendNEC(0xFFFFFFFF, 0); //repeat
IrSender.sendNEC(0xFFFFFFFF, 0); //repeat
IrSender.sendNEC(0xFFFFFFFF, 0); //repeat
IrSender.sendNEC(0xFFFFFFFF, 0); //repeat
IrSender.sendNEC(0xFFFFFFFF, 0); //repeat
IrSender.sendNEC(0xFFFFFFFF, 0); //repeat
IrSender.sendNEC(0xFFFFFFFF, 0); //repeat
IrSender.sendNEC(0xFFFFFFFF, 0); //repeat
IrSender.sendNEC(0xFFFFFFFF, 0); //repeat

```

Also, if I switched to the other Infrared library, is it straight forward to send the hexcodes I have figured out for my other functions?

Thanks again

bengtmartensson, 2020-10-05:
As I wrote previously, you have to insert `delay()`s  between the sends. Try with the first one `delay(40)` and the remainders `delay(96)`.

> Also, if I switched to the other Infrared library, is it straight forward to send the hexcodes I have figured out for my other functions?

Yes.

ArminJo, 2020-10-05:
Another way ist to use the new function `sendNECStandard(uint16_t aAddress, uint8_t aCommand, uint8_t aNumberOfRepeats)`.
I does the delays for you. 
To decode your signal / to get the Address and command of your power button,  you must comment out the line 63 `#define USE_NEC_STANDARD` in IRremote.h.

philspitlerSF, 2020-10-05:
@bengtmartensson adding the delays worked great... thanks so much.

Phil

philspitlerSF, 2020-10-07:
One other last question.... This is really a hardware question but figured you would know the answer..

Currently my IR LED needs to be within about 6" of the projector to work....Does that seem like a side effect of using a resistor that is too big?

ArminJo, 2020-10-07:
Of course the intensity depends on the current through the IR LED. since IR LED's have a low forward voltage, you can use 2 pieces in series doubling the intensity with the same current. But you must then adjust the resistor too to supply the same current, since the drop voltage on the resistor changes when using 2 LED's.

closed 2020-10-12 by ArminJo