Arduino-IRremote/Arduino-IRremote #720: ATtiny85 Doesn't seem to stop bursting signal

landtree, 2020-10-20:
### Board
* [X ] Other - ATtiny85 (TinyCore)

### Protocol
* [ X] Other - Roomba

I tested this code to create a virtual wall for the Roomba. It seems to work just fine with a Leonardo or a Uno. I have hooked up an osiclscope to monitor the PWM to compare it to the ATtiny85. The Uno has a more even break between pulses versus the 85 which seems to send a solid stream. I have tried switching clocks to help solve the problem. I am at a loss. 

**Code Block:**

#include <IRremote.h>
IRsend irsend;

void setup()
{
  irsend.enableIROut(38);

}

void loop()
{
  irsend.mark(1000);
  irsend.space(1000);
}

ArminJo, 2020-10-21:
Which timer do you have enabled? 0 or 1 ?
see line 249 in IRremoteBoardDefs.h
```
#define IR_USE_TIMER_TINY0   // tx = pin 1
//#define IR_USE_TIMER_TINY1   // tx = pin 4
```

landtree, 2020-10-24:
My timer for the ATtiny85 reads:

//ATtiny85
#elif defined(__AVR_ATtiny85__)
#  if !defined(IR_USE_TIMER_TINY0)
#define IR_USE_TIMER_TINY0   // tx = pin 1
#  endif

This was located starting at line 235.

landtree, 2020-10-24:
I just noticed the version Arduino updates to is 2.7 vs githubs 2.8. 

I updated and reuploaded and I still to be having the same problem. I am hooking up the oscilscope right now.

New board def:
`//ATtiny85
#elif defined(__AVR_ATtiny85__)
#  if !defined(IR_USE_TIMER_TINY0) && !defined(IR_USE_TIMER_TINY1)
#define IR_USE_TIMER_TINY0   // tx = pin 1
//#define IR_USE_TIMER_TINY1   // tx = pin 4
#  endif`

ArminJo, 2020-10-24:
What happens if you enable IR_USE_TIMER_TINY1?

landtree, 2020-10-24:
If I disable TINY0 I get a compiling error:

```

In file included from ....\Arduino\libraries\IRremote\src\IRremote.h:25,
                 from .....\Arduino\libraries\IRremote\src\irSend.cpp:1:
......\Arduino\libraries\IRremote\src\irSend.cpp: In member function 'void IRsend::space(unsigned int)':
.....\Arduino\libraries\IRremote\src\private/IRremoteBoardDefs.h:864:59: error: expected ';' before ')' token
 #define TIMER_DISABLE_SEND_PWM    (GTCCR &= ~(_BV(PWM1B))))
                                                           ^
.....\Arduino\libraries\IRremote\src\irSend.cpp:145:5: note: in expansion of macro 'TIMER_DISABLE_SEND_PWM'
     TIMER_DISABLE_SEND_PWM; // Disable pin 3 PWM output
     ^~~~~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board ATtiny25/45/85 (No bootloader).
```

ArminJo, 2020-10-24:
Sorry, my fault.
I updated the IRremoteBoardDefs.h in the repo.

landtree, 2020-10-24:
Alright, updated with the new IRremoteBoardDefs.h siwtched to tiny1.

I can compile now but I still can't get a response from the Roomba and on the ociliscope I am just getting a flat line. Hmmm, but the IR led is on confirmed via camera. 

I tried to simple blink code with microsecond delay and the scope shows the pulse. Granted I also just might not fully understand how that translates with the 38khz signal.

ArminJo, 2020-10-25:
OK you won.
I got my ATtiny and reproduced and fixed the error.
you have to replace your library with the new content from the repo, then it will work.

landtree, 2020-10-25:
Success! Thank you!

closed 2020-10-25 by landtree