Arduino-IRremote/Arduino-IRremote #721: Panasonic codes are calculated correctly

SanZamoyski, 2020-10-26:
### Board
* [ ] Arduino ATmega328* board (UNO, Nano)

### Protocol
* [ ] Panasonic

**Code Block:**
IRreceiveDumpV2.ino

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

I own Panasonic Viera TV and universal remote for decoder (original remote is somewhere but do not know where). It works fine with TV (at last for On/off, input and volume. I want to build simple remote with only four buttons, so i wanted to dump codes, obviously. 
Here are my dumps: https://gist.github.com/SanZamoyski/1179a4e9ad0b249b17f229fa25ca8b43

But usually there is nothing useful in those dumps. Sometimes i get something useful, i.e.:
```
unsigned int  rawDataOn[99] = {3500,1700, 500,400, 450,1250, 500,400, 500,400, 450,400, 500,350, 500,450, 450,400, 500,350, 500,400, 500,350, 500,400, 500,400, 500,1250, 450,400, 500,400, 450,400, 500,400, 500,350, 500,400, 500,350, 500,400, 500,400, 450,1300, 450,400, 500,400, 450,400, 500,400, 450,400, 500,400, 500,350, 500,400, 500,1250, 450,400, 500,1250, 500,1250, 450,1250, 500,1250, 500,350, 500,400, 500,1250, 450,400, 500,1250, 500,1250, 450,1250, 500,1250, 450,400, 500,1250, 500};  // Protocol=PANASONIC Data=0x40100BC
unsigned int  rawDataSrc[99] = {3550,1700, 450,400, 500,1250, 500,350, 500,400, 500,350, 500,400, 500,400, 450,400, 500,400, 500,400, 450,400, 500,350, 500,400, 500,1250, 500,350, 500,400, 500,350, 500,400, 500,400, 500,400, 450,400, 500,350, 500,400, 500,1250, 500,350, 500,400, 500,350, 500,400, 500,400, 450,400, 500,400, 500,400, 450,1250, 500,350, 500,1300, 450,400, 500,350, 500,400, 500,400, 450,400, 500,1250, 500,350, 500,1250, 500,400, 450,400, 500,400, 500,400, 450,1250, 500};  // Protocol=PANASONIC Data=0x40100A0 Address=0x4004
unsigned int  rawDataVolUp[99] = {3500,1700, 500,400, 450,1300, 450,400, 500,400, 450,400, 500,400, 450,450, 450,400, 500,400, 450,400, 500,400, 450,450, 450,400, 450,1300, 450,400, 500,400, 450,450, 450,400, 450,450, 450,400, 500,400, 450,400, 500,400, 450,1300, 450,400, 500,400, 450,400, 500,400, 450,450, 450,400, 500,350, 500,400, 500,1250, 450,400, 500,1250, 500,400, 450,400, 500,400, 450,400, 500,400, 450,1300, 450,400, 500,1250, 450,400, 500,400, 450,400, 500,400, 500,1250, 450};  // Protocol=PANASONIC Data=0x0
unsigned int  rawDataVolDown[99] = {3600,1700, 450,400, 500,1250, 450,400, 500,400, 450,400, 500,400, 500,350, 500,400, 500,400, 450,400, 500,400, 500,350, 500,400, 500,1250, 450,400, 500,400, 500,350, 500,400, 500,350, 500,400, 500,400, 450,400, 500,400, 500,1250, 450,400, 500,400, 450,400, 500,400, 500,400, 450,400, 500,400, 450,450, 450,400, 500,400, 450,400, 500,400, 450,400, 500,1250, 500,350, 500,400, 500,350, 500,400, 500,400, 450,450, 450,400, 500,1250, 450,400, 500,1250, 500};  // Protocol=PANASONIC Data=0x0 Address=0x4004
```
but after adding it into my code as `IrSender.sendPanasonic(PanasonicAddress,PanasonicVolumeDown);` i get back somehow cutted output (with the same receiver).

ArminJo, 2020-10-26:
What are the difference between the IR sender Input and the DumpV2 output?
And please use the latest library version from yesterday!

SanZamoyski, 2020-10-26:
Dumps with 2.8 library are here: https://gist.github.com/SanZamoyski/90ec1e72b112bbe7a6a21b4012abcd31
Unfortunately, I can't find "old" dumps, but it was something like 0x100BCBD (send) 40100BC or 04100BC (receive).

ArminJo, 2020-10-26:
#define PANASONIC_ADDRESS_BITS    16
#define PANASONIC_DATA_BITS       32
so you need no increased RAW BUFFER

As an alternative you can try [IRMP](https://github.com/ukw100/IRMP) and the [Kaseyko ](https://www.mikrocontroller.net/articles/IRMP_-_english#KASEIKYO)protocol,

SanZamoyski, 2020-10-26:
Yup! 
Tried IRMP and it tells me it is Kaseyko protocol. Thanks!

closed 2020-10-26 by SanZamoyski