Arduino-IRremote/Arduino-IRremote #722: Latest release, ESP32,Pin35, does not decode Sony anymore

FrankBoesing, 2020-10-31:
### Board
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [x] ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266
* [ ] Teensy board
* [ ] Other - please specify

### Protocol
* [ ] Unknown
* [ ] BoseWave
* [ ] Denon
* [ ] Dish
* [ ] JVC
* [ ] Lego
* [ ] LG
* [ ] NEC
* [ ] Panasonic
* [ ] RC5, RC6
* [ ] Samsung
* [ ] Sanyo
* [ ] Sharp
* [x] Sony
* [ ] Whynter
* [ ] Other - please specify

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

The example sketch does not decode Sony anymore. Instead it shows the RAW values.
It worked with the version before (2.7.0)

ArminJo, 2020-10-31:
Hello Frank
The new version now checks the header space.
Can you send me the new RAW result and the result of the old version?
Armin

FrankBoesing, 2020-10-31:
Sure, here are both:
`Key "1" on the remote:

New Version:
21:59:13.330 -> START e:\temp\arduino_modified_sketch_291386\IRreceiveDump.ino from Oct 31 2020
21:59:13.330 -> Ready to receive IR signals at pin 35
21:59:28.168 -> 
21:59:28.168 -> Protocol=UNKNOWN Data=0x207BEF0C (32 bits)
21:59:28.168 -> Raw [26]: -1702500 2450 -550 650 -550 650 -550 650 -550 650 -500 700 -500 700 -500 700 -500 1300 -500 700 -500 700 -500 650 -550 1250 
21:59:28.215 -> 
21:59:28.215 -> Protocol=UNKNOWN Data=0x207BEF0C (32 bits)
21:59:28.215 -> Raw [26]: -26900 2500 -500 700 -500 700 -500 700 -500 650 -550 650 -550 650 -550 650 -550 1250 -550 650 -550 650 -550 650 -550 1250 
21:59:28.268 -> 

Old Version:
2:02:11.659 -> START e:\temp\arduino_modified_sketch_459514\IRreceiveDump.ino from Oct 31 2020
22:02:11.713 -> Ready to receive IR signals at pin 35
22:02:16.071 -> 
22:02:16.071 -> Protocol=SONY Data=0x11 (12 bits)
22:02:16.071 -> Raw [26]: -4381500 2450 -550 650 -550 650 -550 650 -550 650 -500 700 -500 700 -500 700 -500 1250 -550 700 -500 650 -550 650 -550 1250 
22:02:16.124 -> 
22:02:16.124 -> Protocol=SONY Data=0x11 (12 bits)
22:02:16.124 -> Raw [26]: -26900 2500 -500 700 -500 700 -500 700 -500 650 -550 650 -550 650 -550 650 -550 1250 -550 650 -550 650 -550 650 -550 1250 
22:02:16.171 -> 
`

ArminJo, 2020-10-31:
Yep, you were right. I made a mistake on cleanup of the ir_Sony.cpp.
Can you please test and give feedback, then I will make a new release.

closed 2020-10-31 by ArminJo

FrankBoesing, 2020-11-01:
Yes, it works now. Good support!

Thank you, 
Frank

commit b49d2339 (2020-10-31): Closes #722