Arduino-IRremote/Arduino-IRremote #724: Why 33% duty cycle

ArminJo, 2020-11-05:
With IR signals & batteries the key thing is to always run the IR carrier @ 33.3% duty cycle (which IRremote does by default).  There is interesting maths behind this for anyone who wants to research further.

_Originally posted by @AnalysIR in https://github.com/z3t0/Arduino-IRremote/issues/674#issuecomment-586267368_

AnalysIR, 2020-11-05:
> AFAIK the volume of fundamental is most at 50%, with other duty cycles you generate more harmonics.

Yes....but **_relatively_** more of the harmonics is wasted @505 vs 33%.  When designing with batteries or for low power you always want to eliminate any power waste. These days it isprobably a good idea to avoid needless waste regardless. ......see below for more info....

That was a while ago  & I can't remember where it was shown. At the time I was looking into the maths behind FFTs and square waves etc.. From memory, it was some IR related article on the  topic - so may still be available via google.

The harmonics differ between a 33.3% duty and  a 50% duty cycle wave. With 33% duty **_relatively_** more of the energy is encapsulated within the fundamental than with 50%.

(In the receiver, I believe the filter isolates the fundamental sine wave and discards the harmonics...so any power in the harmonics is essentially wasted)

Another way of looking at it is that @50% duty cycle **_relatively_** more power is wasted in the harmonics that at 33.33333% and thus if you increase the power at 33% to be the equivalent of 50% there is beter efficiency.

example:
sending a signal @50% duty with a current of 100mA is the same as sending a signal @33% with a current of 150mA...**_in terms of nominal battery usage/drain_**. In the case of 33% duty more range is achieved per signal for the same battery drain. This approach can be leveraged to extend battery life or improve performance/range or a combination of both. 

Another way....more energy is in the 33% duty fundamental @ 150mA current than a 50% duty fundamental @ 100mA with nominal battery drain the same in both cases.

_Caveat: the above is meant as casual commentary and should be treated as such. If any reader needs a 'scientific' explanation/proof or wants to debate it please do the research and post your own findings._

ArminJo, 2020-11-15:
@AnalysIR Can you mak me a favor and please check the [new paragraph](https://github.com/z3t0/Arduino-IRremote#theory-of-pwm-and-ir-power) if there are no faults, english errors or misunderstandings.
Thanks
Armin

AnalysIR, 2020-11-15:
**General Comment**: I think the explanation is too detailed for 90%+ of IRremote users.

As mentioned before, IR users should be pointed at 50% duty cycle or the optimum for battery use/low power design 33% duty cycle.

> This can easily be done by reducing the series resistor by the same factor.

The factor may not be linear for all IR driver circuits. so...

> This can easily be done by adjusting the value of series resistor.

...or similar may be better text

_(Note: I didn't study/verify all the graphs or test etc in any detail)_

ArminJo, 2020-11-15:
Thanks!
> General Comment: I think the explanation is too detailed for 90%+ of IRremote users.

99% is my estimation, but I want to write down the math I could never find elesewhere.

closed 2020-11-15 by ArminJo