Arduino-IRremote/Arduino-IRremote #730: irrecv not work after irsend

Andr3g2, 2020-11-28:
### Board
* [x ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266
* [ ] Teensy board
* [ ] Other - please specify

### Protocol
* [ ] Unknown
* [ ] BoseWave
* [ ] Denon
* [ ] Dish
* [ ] JVC
* [ ] Lego
* [ ] LG
* [ ] NEC
* [ ] Panasonic
* [ ] RC5, RC6
* [ x] Samsung
* [ ] Sanyo
* [ ] Sharp
* [ ] Sony
* [ ] Whynter
* [ ] Other - please specify

**Code Block:**
```

#include <IRremote.h> 

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

Hallo,

I try use irsend & irrec.

irrec works until i use irsend.
after i use irsend, irrec  no longer works.
arduino works generally.

what do I do wrong?

regards Andre
BTU-C-S

tested Arduino-IRremote-master 2.7.0,2.8.0

Code:

IRrecv irrecv(IR_RECV_PIN);
IRsend IrSender;
...
void loop() {
  unsigned long tData = 0xE0E0E02F;
  tnows=(millis()/1000);
  if (irrecv.decode(&results)) {
   Serial.println(results.value, HEX);
   if (results.value == 0xE0E0E01F) {   //Wenn der Infrarotempfänger die Zahl 
    anstoss=1;
   }
  irrecv.resume();
  }
 if (anstoss==1){
  anstoss=0;
  delay(5000);
  IrSender.sendSAMSUNG(tData, 4);
  Serial.print(F("sendSAMSUNG(0x")); Serial.print(tData,HEX);  Serial.println(F(
", 4)"));
  //irrecv.resume();
  }
 if (tmemread<=tnows) {
  tmemread=(tnows+tintread);
  Serial.println(String("heartbeat ")+tnows);
 }
 delay(250);
} // loop

monitor:
- send voldown
- send volup
- send voldown
irfb v04 201128
heartbeat 0
E0E0D02F
E0E0D02F
E0E0D02F
E0E0E01F
sendSAMSUNG(0xE0E0E02F, 4)
heartbeat 30
heartbeat 60

ArminJo, 2020-11-28:
Try to use        
 IrReceiver.enableIRIn(); // Re-enable receiver
after sending

Andr3g2, 2020-11-29:
problem solved

Thx Andre

closed 2020-11-29 by Andr3g2