Arduino-IRremote/Arduino-IRremote #733: Suggested changes to ir_Template.cpp

Jim-2249, 2020-12-10:
I used ir_Template.cpp to add a new protocol. Alas... it is the same protocol as LG. So, a lot of unnecessary work.
I did find a few issues. 
- Function names and parameter orders appear have been changed.
- Function usage appears to have changed.
- Instructions in header comments a bit off.

Thank you for making this code available! I'd never have gotten this far without it! Hopefully, this Issue is not a mess.
I attached ir_Template.zip It compiles, but is otherwise untested. My version is a mess, so I didn't include it.
[ir_Template.zip](https://github.com/z3t0/Arduino-IRremote/files/5669830/ir_Template.zip)

==========================================================
Line 41;
Current:
```
 B. In the section "enumerated list of all supported formats", add:
 SHUZU,
```
Suggested change:
```
 B. In the section "An enum consisting of all supported formats", add:
 SHUZU,
```
==========================================================
Line 47:
Missing function declaration: bool decodeShuzu ();
Current:
```
 #if DECODE_SHUZU
 bool  decodeShuzu (decode_results *aResults) ;
 #endif
```
Suggest change:
```
 #if DECODE_SHUZU
 bool  decodeShuzu ();
 bool  decodeShuzu (decode_results *aResults) ;
 #endif
```
The function is defined about line 138:
```
 bool IRrecv::decodeShuzu() {
```
==========================================================
Line 59:
#ifdef in error? Change DECODE_NEC to DECODE_SHUZU
Current:
```
 2. Now open irRecv.cpp and make the following change:

 A. In the function IRrecv::decode(), add:
 #ifdef DECODE_NEC
 DBG_PRINTLN("Attempting Shuzu decode");
 if (decodeShuzu(results))  return true ;
 #endif
```
Suggested change:
```
 2. Now open irReceive.cpp and make the following change:

 A. In the function IRrecv::decode(), add:
 #ifdef DECODE_SHUZU
 DBG_PRINTLN("Attempting Shuzu decode");
 if (decodeShuzu(results))  return true ;
 #endif
```
==========================================================
Line 69:
Missing encoding() function?
Current:
```
 You will probably want to add your new protocol to the example sketch

 3. Open MyDocuments\Arduino\libraries\IRremote\examples\IRrecvDumpV2.ino

 A. In the encoding() function, add:
 case SHUZU:    Serial.print("SHUZU");     break ;
```
No "encoding() function" in IRrecvDumpV2.ino nor IRremote library

==========================================================
Line 127
function name not declared in IRremote.h Similar function in ir_LG.cpp
Current:
```
  sendPulseDistanceData(data, nbits,  SHUZU_BIT_MARK, SHUZU_ONE_SPACE,SHUZU_BIT_MARK, SHUZU_ZERO_SPACE);
```
Suggested change:
```
  sendPulseDistanceWidthData(SHUZU_BIT_MARK, SHUZU_ONE_SPACE,SHUZU_BIT_MARK, SHUZU_ZERO_SPACE, data, nbits);
```
Another example in ir_LG.cpp:
```
  sendPulseDistanceWidthData(LG_BIT_MARK, LG_ONE_SPACE, LG_BIT_MARK, LG_ZERO_SPACE, data, nbits);
```

==========================================================
Line 139
This isn't used in ir_LG.cpp The result 'data' isn't used in ir_Template.cpp Remove?
Current:
```
  unsigned long data = 0;  // Somewhere to build our code
```
Suggested change:
remove line

==========================================================
Line 158 
  parameters appear to be out of order and result is assigned to unsigned long data which is not used
Current:
```
  data = sendPulseDistanceWidthData(results, SHUZU_BITS, offset, SHUZU_BIT_MARK, SHUZU_ONE_SPACE, SHUZU_ZERO_SPACE);
```
Suggested change:

```
  if (!decodePulseDistanceData(SHUZU_BITS, offset, SHUZU_BIT_MARK, SHUZU_ONE_SPACE, SHUZU_ZERO_SPACE)) {
    return false;
  }
```

ArminJo, 2020-12-10:
Thanks for the valuable and elaborated issue 🥇

ArminJo, 2020-12-10:
I included all of your changes in the last commit an added you to the list of collaborators.

closed 2020-12-10 by ArminJo