Arduino-IRremote/Arduino-IRremote #736: may  cause an endless loop

jhao-x, 2020-12-17:
https://github.com/Arduino-IRremote/Arduino-IRremote/blob/563ec5bccee8b1330370015685fe4bcdddf32ad4/src/irReceive.cpp#L235
if irparams.rawlen not reset to 0, in ISR it will set the status IR_REC_STATE_STOP, and alway decode the old  ir data

ArminJo, 2020-12-17:
Did you test it?

jhao-x, 2020-12-17:
> Did you test it?

yes, i only press a key of panasonic remote, uart print the data all the time, never stop

ArminJo, 2020-12-17:
Thanks. Can you please send the code for me to reproduce it?

jhao-x, 2020-12-17:
void IRrecv::resume() {
    irparams.rawlen = 0;
    irparams.rcvstate = IR_REC_STATE_IDLE;
}

ArminJo, 2020-12-17:
And with this code, the error can be reproduced?

jhao-x, 2020-12-17:
> And with this code, the error can be reproduced?

i test it the same error not happen again

ArminJo, 2020-12-17:
So you can close the issue?

closed 2020-12-17 by ArminJo