Arduino-IRremote/Arduino-IRremote #737: Ethernet.h does not work well when IRremote.h is added

Realgigabyte, 2020-12-20:
Hi,

I am trying to write a software that uses Ethernet shield with (Ethernet.h library) and infrared emitter

The network part works fine but it seems memory gets corrupted as soon as I just add IRremote.h

Am I missing something?

Thanks you

AnalysIR, 2020-12-20:
> Am I missing something?

Probably...and providing more details is likely to get you more useful assistance
e.g.  Error messages from the IDE, MCU platform & devices you are using & anything you think might help.

Realgigabyte, 2020-12-21:
> > Am I missing something?
> 
> Probably...and providing more details is likely to get you more useful assistance
> e.g. Error messages from the IDE, MCU platform & devices you are using & anything you think might help.

I didn't add any detail because I was wondering if you guys are aware of those 2 library don't work together (using same memory address, ports, clock...)
I developed a TCP/IP server that would send some infrared codes when it receive a command.
The TCP/IP part works perfect (feedback to client and feedback to Serial).
There is no Infrared related code yet.
The moment I just add #include <IRremote.h>  the TCP/IP routine stop working and I get some corrupted stings in my Serial.println feedback.

Any help would be much appreciated.
Thank.

bengtmartensson, 2020-12-21:
> It's an Arduino Uno with a Network shield...

It _may_ be a memory issue, Uno + Ethernet does not leave that much RAM left... Try with a Mega 2560 if possible.

FWIW, [AGirs](https://github.com/bengtmartensson/Infrared4Arduino)  (which sits on top of [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino)) works fine with IR + Ethernet (TCP sockets), at least on a Mega...

Realgigabyte, 2020-12-21:
Sketch uses 15090 bytes (46%) of program storage space. Maximum is 32256 bytes.
Global variables use 1006 bytes (49%) of dynamic memory, leaving 1042 bytes for local variables. Maximum is 2048 bytes.

This is the report I get when compiled with both libraries.

Realgigabyte, 2020-12-21:
Also sometimes when I compile I get those messages (but then they disappear is I compile again)
In file included from C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Ethernet\src\Dns.cpp:8:0:
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Ethernet\src\Dns.cpp: In member function 'uint16_t DNSClient::BuildRequest(const char*)':
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Ethernet\src\utility/w5100.h:457:25: warning: result of '(256 << 8)' requires 18 bits to represent, but 'int' only has 16 bits [-Wshift-overflow=]
 #define htons(x) ( (((x)<<8)&0xFF00) | (((x)>>8)&0xFF) )
                      ~~~^~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Ethernet\src\Dns.cpp:164:18: note: in expansion of macro 'htons'
  twoByteBuffer = htons(QUERY_FLAG | OPCODE_STANDARD_QUERY | RECURSION_DESIRED_FLAG);
                  ^~~~~

Realgigabyte, 2020-12-21:
> It _may_ be a memory issue, Uno + Ethernet does not leave that much RAM left... Try with a Mega 2560 if possible.

I stripped the sketch to the bone and I can confirm that using less variables both ethernet and ir emitter work.

Will try to use a Mega 2560.

Thank you for the help.

ArminJo, 2020-12-21:
To check the remaining heap and RAM you can use the functions:
```
printFreeHeap(Print *aSerial);
initStackFreeMeasurement();
and printStackFreeMinimumBytes(Print *aSerial);
```
from https://github.com/ArminJo/Arduino-Utils/blob/master/src/AVRUtils.cpp

AnalysIR, 2020-12-21:
FYI: The Nano Every is another option....Faster @20MHz(no crystal) & much more SRAM, Flash, IO & peripherals vs  UNO. Of course, the Mega has much more again.

bengtmartensson, 2020-12-22:
> The Nano Every is another option.

> It's an Arduino Uno with a Network shield...

... so a Nano whatever will not be compatible with that shield.

Realgigabyte, 2020-12-22:
Ordered a Mega just now so I can reuse the shield.
But I'll consider a Nano as I can see there are some Ethernet shields on eBay for the Nano.

closed 2021-01-15 by ArminJo