Arduino-IRremote/Arduino-IRremote #740: ATtiny85 no longer supported on latest version?

darkmetalman, 2020-12-22:
The latest version (2.8.0) wont compile for ATtiny85... 
Previous versions worked fine...

### Board
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266
* [ ] Teensy board
* [X] Other - ATtiny85

### Protocol
* [ ] Unknown
* [ ] BoseWave
* [ ] Denon
* [ ] Dish
* [ ] JVC
* [ ] Lego
* [ ] LG
* [ ] NEC
* [ ] Panasonic
* [ ] RC5, RC6
* [ ] Samsung
* [ ] Sanyo
* [ ] Sharp
* [ ] Sony
* [ ] Whynter
* [ ] Other - please specify

**Code Block:**
```

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [X] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [X] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

ArminJo, 2020-12-22:
Are you sure?
We can not reproduce it, see: https://github.com/Arduino-IRremote/Arduino-IRremote/runs/1542951668?check_suite_focus=true#step:3:84

Montecri, 2020-12-31:
Hi,

Can confirm something is wrong. Digispark clone here, ATTiny85, IRsendRawDemo.ino and any other send example will not work. Blinks garbage. Same sketches works perfectly on Uno R3.

Thanks.

ArminJo, 2020-12-31:
OK, i currently do some optimisations...
But I just compiled the (new) IRsendDemo.cpp on my digispark with [this core](https://github.com/ArminJo/DigistumpArduino).
And everything is quite fine.
With send raw I can see one unexpected long pulse, and I am digging for the reason (tomorrow).

Montecri, 2020-12-31:
Hi. Thank you for the core tip. Installed and using this one now, as well as an updated micronucleus.

It worked now but had to set the clock to 16MHz since 16.5MHz is not supported.

Also, think I was missing the code below, since I started from IRSendPronto example and it didn't include it:

#if defined(__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__) || defined(__AVR_ATtiny87__) || defined(__AVR_ATtiny167__)
#include "ATtinySerialOut.h"
#endif

Thank you!

Regards,

- Cristiano

closed 2021-01-02 by ArminJo