Arduino-IRremote/Arduino-IRremote #745: Panasonic Decode output offset by two spaces using IRreceiveDumpV2

Boostedk20, 2021-01-01:
### Board
* [x] Arduino ATmega328* board (UNO, Nano)

### Protocol
* [x] Panasonic

When using the included IRreceiveDumpV2 program, decoding Panasonic IR was resulting in incorrect data values via the Serial Monitor.  For example, "OK" was returning:
```
uint16_t address = 0x4004;
uint16_t data = 0x4010092;

```
I noticed all were starting with '40' and missing the last two HEX values.  It appeared as the offset was two places to the left (grabbing the last 4 in the address and the leading 0).  The correct output should be:
```
uint16_t address = 0x4004;
uint16_t data = 0x1009293;

```

I was able to trace this back to the ir_Panasonic.cpp file in decodePanasonic. I made the following change to produce the proper data output:

```cpp
//    if (!decodePulseDistanceData(PANASONIC_DATA_BITS, offset + PANASONIC_ADDRESS_BITS, PANASONIC_BIT_MARK,
//    PANASONIC_ONE_SPACE, PANASONIC_ZERO_SPACE)) {
//        return false;
//    }
	
	// added + 8 to PANASONIC_DATA_BITS to show correct value
    if (!decodePulseDistanceData(PANASONIC_DATA_BITS + 8, offset + PANASONIC_ADDRESS_BITS, PANASONIC_BIT_MARK,
    PANASONIC_ONE_SPACE, PANASONIC_ZERO_SPACE)) {
        return false;
    }

```

I noticed that the ir_Panasonic.cpp file is no longer in this github, so I was unable to suggest any changes (or have someone familiar with CPP suggest proper code).  I am not sure the reason behind the removal of the file, but this change helped with decoding Panasonic IR for my Viera Plasma TV.

ArminJo, 2021-01-01:
Thanks for your contribution.
Can you please do me a big favor and use the new repository content and the IRreceiveDump program (formerly DumpV2) and check again. I did a mayor overhaul of the panasonic part, it is now in ir_Kaseikyo.cpp.

Boostedk20, 2021-01-03:
Thank you for the reply.  I used the 2.8.2 version and loaded the IRreceieveDump program.  I see the following in teh Serial Monitor when pressing "OK":
```
Protocol=PANASONIC Address=0x80 Command=0x49 Raw-Data=0x0 (48 bits)
```
It seems to be in a different format, but the pronto command looks the same as it was with the prior release.  Is this the expected result?

ArminJo, 2021-01-03:
Thanks a lot!
Yes the format is now the standard format with address and command for the new sendPanasonicStandard(uint16_t aAddress, uint8_t aCommand, uint8_t aNumberOfRepeats) function.

closed 2021-01-03 by ArminJo