Arduino-IRremote/Arduino-IRremote #746: Denon ir dump no data

Insomniac80, 2021-01-01:
### Board
* [x] Arduino ATmega328* board (Nano, old bootloader)

### Protocol
* [x] Denon

Code is the actual example
"IRreceiveDump.cpp"

There is only a TSOP34156 on pin 11 powered by 5v, nothing else.
I could copy paste the example code here, but it is 1:1 from the examples.

I have a Denon RC-1167 remote for my Denon AVR-2313. Before, I have used "send panasonic" twice with 65ms delay, which was also read by the receiver demonstration. 
Actually I get Denon data when dumping, problem is with no data (0x0). sendDenon expects data and bits. 

See example below:

```
Protocol=Denon Address=0x3140 Command=0x2D Raw-Data=0x0 (48 bits)

Result in internal ticks (50 us) - with leading gap
rawData[100]: 
     -46631
     +  66,-  34     +  8,-  8     +  8,-  9     +  8,-  25
     +  8,-  9     +  8,-  25     +  8,-  9     +  8,-  25
     +  8,-  8     +  8,-  9     +  8,-  25     +  8,-  9
     +  8,-  8     +  8,-  26     +  8,-  25     +  8,-  8
     +  8,-  9     +  8,-  9     +  8,-  8     +  8,-  9
     +  8,-  8     +  8,-  9     +  8,-  9     +  8,-  25
     +  8,-  8     +  8,-  26     +  8,-  8     +  8,-  9
     +  8,-  8     +  8,-  26     +  8,-  25     +  8,-  8
     +  9,-  8     +  8,-  25     +  8,-  9     +  8,-  25
     +  8,-  25     +  8,-  9     +  8,-  25     +  8,-  9
     +  8,-  8     +  8,-  9     +  8,-  8     +  9,-  25
     +  8,-  25     +  8,-  25     +  8,-  9     +  8,-  25
     +  8,-  8     +  9
Result in microseconds - with leading gap
rawData[100]: 
     -37790
     +3300,-1700     + 400,- 400     + 400,- 450     + 400,-1250
     + 400,- 450     + 400,-1250     + 400,- 450     + 400,-1250
     + 400,- 400     + 400,- 450     + 400,-1250     + 400,- 450
     + 400,- 400     + 400,-1300     + 400,-1250     + 400,- 400
     + 400,- 450     + 400,- 450     + 400,- 400     + 400,- 450
     + 400,- 400     + 400,- 450     + 400,- 450     + 400,-1250
     + 400,- 400     + 400,-1300     + 400,- 400     + 400,- 450
     + 400,- 400     + 400,-1300     + 400,-1250     + 400,- 400
     + 450,- 400     + 400,-1250     + 400,- 450     + 400,-1250
     + 400,-1250     + 400,- 450     + 400,-1250     + 400,- 450
     + 400,- 400     + 400,- 450     + 400,- 400     + 450,-1250
     + 400,-1250     + 400,-1250     + 400,- 450     + 400,-1250
     + 400,- 400     + 450

Result as internal ticks (50 us) array - always without leading gap
uint8_t rawTicks[99] = {66,34, 8,8, 8,9, 8,25, 8,9, 8,25, 8,9, 8,25, 8,8, 8,9, 8,25, 8,9, 8,8, 8,26, 8,25, 8,8, 8,9, 8,9, 8,8, 8,9, 8,8, 8,9, 8,9, 8,25, 8,8, 8,26, 8,8, 8,9, 8,8, 8,26, 8,25, 8,8, 9,8, 8,25, 8,9, 8,25, 8,25, 8,9, 8,25, 8,9, 8,8, 8,9, 8,8, 9,25, 8,25, 8,25, 8,9, 8,25, 8,8, 9};  // Protocol=Denon Address=0x3140 Command=0x2D Raw-Data=0x0 (48 bits)

Result as microseconds array - with leading gap for IRremote versions 2.*
uint16_t rawData[100] = {37790, 3300,1700, 400,400, 400,450, 400,1250, 400,450, 400,1250, 400,450, 400,1250, 400,400, 400,450, 400,1250, 400,450, 400,400, 400,1300, 400,1250, 400,400, 400,450, 400,450, 400,400, 400,450, 400,400, 400,450, 400,450, 400,1250, 400,400, 400,1300, 400,400, 400,450, 400,400, 400,1300, 400,1250, 400,400, 450,400, 400,1250, 400,450, 400,1250, 400,1250, 400,450, 400,1250, 400,450, 400,400, 400,450, 400,400, 450,1250, 400,1250, 400,1250, 400,450, 400,1250, 400,400, 450};  // Protocol=Denon Address=0x3140 Command=0x2D Raw-Data=0x0 (48 bits)

uint16_t address = 0x3140;
uint16_t command = 0x2D;
uint32_t data = 0x0;

Pronto Hex as string
char ProntoData[] = "0000 006D 0032 0000 007F 0041 000F 000F 000F 0011 000F 0030 000F 0011 000F 0030 000F 0011 000F 0030 000F 000F 000F 0011 000F 0030 000F 0011 000F 000F 000F 0032 000F 0030 000F 000F 000F 0011 000F 0011 000F 000F 000F 0011 000F 000F 000F 0011 000F 0011 000F 0030 000F 000F 000F 0032 000F 000F 000F 0011 000F 000F 000F 0032 000F 0030 000F 000F 0011 000F 000F 0030 000F 0011 000F 0030 000F 0030 000F 0011 000F 0030 000F 0011 000F 000F 000F 0011 000F 000F 0011 0030 000F 0030 000F 0030 000F 0011 000F 0030 000F 000F 0011 0806 "

```

ArminJo, 2021-01-02:
1. Take the new version from the repo.
2. Use the new sendDenonStandard() function as in IRsendDemo example.
3. Close issue if all works 😀

bengtmartensson, 2021-01-02:
> There is only a TSOP34156

The TSOPxxx56 is optimized for a 56 kHz carrier, while the protocols of the used remote uses 37 and 38 kHz. I recommend using a TSOPxxx38, optimized for 38kHz, but covers 36-40 kHz with good performance.

Insomniac80, 2021-01-02:
> 1. Take the new version from the repo.
> 2. Use the new sendDenonStandard() function as in IRsendDemo example.
> 3. Close issue if all works 😀

Thanks for the reply, I have now downloaded the code from the site and manually updated the library.
I have chenged to a TSOP4838 (was written on the package from ebay) and uploaded the IReceiveDump.ino. The output is same then the other TSOP, maybe I was wrong in the beginning about the xx56.

Output is:

```
Ready to receive IR signals at pin 11

Protocol=DENON Address=0x5140 Command=0x0 Raw-Data=0x0 (48 bits)

Result in internal ticks (50 us) - with leading gap
rawData[100]:
     -63626
     +  66,-  34     +  8,-  8     +  8,-  9     +  8,-  25
     +  8,-  9     +  8,-  25     +  8,-  9     +  8,-  25
     +  8,-  8     +  8,-  9     +  8,-  25     +  8,-  9
     +  8,-  8     +  8,-  26     +  8,-  25     +  8,-  8
     +  8,-  9     +  8,-  9     +  8,-  8     +  8,-  9
     +  8,-  8     +  8,-  9     +  8,-  9     +  8,-  25
     +  8,-  8     +  8,-  26     +  8,-  8     +  8,-  9
     +  8,-  8     +  8,-  26     +  8,-  8     +  8,-  25
     +  8,-  9     +  8,-  9     +  8,-  8     +  8,-  9
     +  8,-  8     +  8,-  9     +  8,-  9     +  8,-  8
     +  8,-  9     +  8,-  25     +  8,-  9     +  8,-  8
     +  8,-  9     +  8,-  25     +  8,-  9     +  8,-  8
     +  8,-  9     +  8
Result in microseconds - with leading gap
rawData[100]:
     -3181300
     +3300,-1700     + 400,- 400     + 400,- 450     + 400,-1250
     + 400,- 450     + 400,-1250     + 400,- 450     + 400,-1250
     + 400,- 400     + 400,- 450     + 400,-1250     + 400,- 450
     + 400,- 400     + 400,-1300     + 400,-1250     + 400,- 400
     + 400,- 450     + 400,- 450     + 400,- 400     + 400,- 450
     + 400,- 400     + 400,- 450     + 400,- 450     + 400,-1250
     + 400,- 400     + 400,-1300     + 400,- 400     + 400,- 450
     + 400,- 400     + 400,-1300     + 400,- 400     + 400,-1250
     + 400,- 450     + 400,- 450     + 400,- 400     + 400,- 450
     + 400,- 400     + 400,- 450     + 400,- 450     + 400,- 400
     + 400,- 450     + 400,-1250     + 400,- 450     + 400,- 400
     + 400,- 450     + 400,-1250     + 400,- 450     + 400,- 400
     + 400,- 450     + 400

Result as internal ticks (50 us) array - always without leading gap
uint8_t rawTicks[99] = {66,34, 8,8, 8,9, 8,25, 8,9, 8,25, 8,9, 8,25, 8,8, 8,9, 8,25, 8,9, 8,8, 8,26, 8,25, 8,8, 8,9, 8,9, 8,8, 8,9, 8,8, 8,9, 8,9, 8,25, 8,8, 8,26, 8,8, 8,9, 8,8, 8,26, 8,8, 8,25, 8,9, 8,9, 8,8, 8,9, 8,8, 8,9, 8,9, 8,8, 8,9, 8,25, 8,9, 8,8, 8,9, 8,25, 8,9, 8,8, 8,9, 8};  // Protocol=DENON Address=0x5140 Command=0x0 Raw-Data=0x0 (48 bits)

Result as microseconds array - with leading gap for IRremote versions 2.*
uint16_t rawData[100] = {35572, 3300,1700, 400,400, 400,450, 400,1250, 400,450, 400,1250, 400,450, 400,1250, 400,400, 400,450, 400,1250, 400,450, 400,400, 400,1300, 400,1250, 400,400, 400,450, 400,450, 400,400, 400,450, 400,400, 400,450, 400,450, 400,1250, 400,400, 400,1300, 400,400, 400,450, 400,400, 400,1300, 400,400, 400,1250, 
400,450, 400,450, 400,400, 400,450, 400,400, 400,450, 400,450, 400,400, 400,450, 400,1250, 400,450, 400,400, 400,450, 400,1250, 400,450, 400,400, 400,450, 400};  // Protocol=DENON Address=0x5140 Command=0x0 Raw-Data=0x0 (48 bits)

uint16_t address = 0x5140;
uint16_t command = 0x0;
uint32_t data = 0x0;

Pronto Hex as string
char ProntoData[] = "0000 006D 0032 0000 007F 0041 000F 000F 000F 0011 000F 0030 000F 0011 000F 0030 000F 0011 000F 0030 000F 000F 000F 0011 000F 0030 000F 0011 000F 
000F 000F 0032 000F 0030 000F 000F 000F 0011 000F 0011 000F 000F 000F 0011 000F 000F 000F 0011 000F 0011 000F 0030 000F 000F 000F 0032 000F 000F 000F 0011 000F 000F 000F 0032 000F 000F 000F 0030 000F 0011 000F 0011 000F 000F 000F 0011 000F 000F 000F 0011 000F 0011 000F 000F 000F 0011 000F 0030 000F 0011 000F 000F 000F 0011 000F 0030 000F 0011 000F 000F 000F 0011 000F 0806 "

```

Then I uploaded the IReceiveDemo.ino, which outputs (both monitor speed was 115200):

```
Enabling IRin
Ready to receive IR signals at pin 11
Protocol=DENON Address=0x5140 Command=0x0 Raw-Data=0x0 (48 bits)
[... 52 lines cut ...]
	delay(myDelay);
    irsend.sendSharpStandard(sAddress, sCommand, sRepeats);
    delay(1000);

    // sRepeats++;	
	myDelay += 1;

delay(1000);

```

I have once commented out the "myDelay" and second call, and increased the repeats. Then I checked with zero repeats and increased myDelay by one ms, from 50 up to 100.
No luck.

I have once read out the same button (was 2018 and in hours I was not able to find the library I used ). However, it said it was a panasonic protocol. When using:

```
#define PanasonicAddress 0x2A4C
#define PanasonicPower 0x28A0088

	irsend.sendPanasonic(PanasonicAddress, payld);
	delay(65);
	irsend.sendPanasonic(PanasonicAddress, payld);
```
it works.

ArminJo, 2021-01-02:
Sorry, I forgot to mention, that you must comment out line 41 https://github.com/Arduino-IRremote/Arduino-IRremote/blob/8b1d47cebb92aa6a3c4d1967ea2ba4084b657722/src/IRremote.h#L41
And please check if line 194 of your local version of irDenon.cpp is the same as https://github.com/Arduino-IRremote/Arduino-IRremote/blob/75f4f7ef0a9871d1d9096b31baa95741b4c869ac/src/ir_Denon.cpp#L194
For me it looks like you are compiling an old version.

Insomniac80, 2021-01-02:
> Sorry, I forgot to mention, that you must comment out line 41
>
> https://github.com/Arduino-IRremote/Arduino-IRremote/blob/8b1d47cebb92aa6a3c4d1967ea2ba4084b657722/src/IRremote.h#L41
>
>
> And please [...]

Normally I am using Platform.io. 
To ensure I am using the newest library, I have clean removed my old Arduino IDE, deleted the "documents/Arduino" folder, newly installed 1.8.13 . All projects are deleted, only library is the actual (Arduino-IRremote-master.zip), copied by hand. From readme: "Version 2.8.3". Checked library list, only shows two pre-installed ones, clean remove was successful.
 
I have checked both mentioned code lines: 
line 41 of "Arduino-IRremote-master/src/IRremote.h" has already been commented out. 
line 194 of "Arduino-IRremote-master/src/ir_Denon.cpp" is same as in your answer.

I have again read the code from IR, it shows the same output as above.

I then have uploaded "IRSendDemo" from the examples within the folder (always using the Arduino IDE atm^^). Only change is:
```
uint16_t sAddress = 0x5340;
uint16_t sCommand = 0x0;
```
to the read values (and data type size), and
```
// sAddress += 0x0101;
// sCommand += 0x11;
```
is now commented out, so only the repeats are increasing.

I have an ir receiver right above the AVR, which shows (some) data is sended, while the serial monitor shows it sends the right data.
The command is still not recognized by the AVR.

ArminJo, 2021-01-02:
> line 41 of "Arduino-IRremote-master/src/IRremote.h" has already been commented out.

You got the latest version from the repo and like a miracle the line is commented out???
I do not know what you are doing, but blease check that miracle.
I recently tested the current version and It looks like:
```
Protocol=Denon Address=0x6 Command=0x5A Raw-Data=0x1968 (15 bits)
Protocol=Denon Address=0x6 Command=0x5A Auto-repeat gap=45350us Raw-Data=0x1A97 (15 bits)
```

Insomniac80, 2021-01-02:
https://github.com/Arduino-IRremote/Arduino-IRremote -> Button "Code" -> Download Zip
Unpacked, opened "IRemote.h", line 41 reads:
`//#define USE_STANDARD_DECODE // remove comment to have the standard NEC and other decoders available.`
Am I downloading at the wrong place?

ArminJo, 2021-01-02:
Ok now i got it. I am not a native english speaker, so how do you call it if you comment and uncomment some code.
In German we say "auskommentiert" for commenting and and I translated it to comment out 😞.
Can I use comment and uncomment or is comment and remove comment better?

AnalysIR, 2021-01-02:
> comment and remove comment

is probably best, particularly for non-native speakers

_for fun/reference here is a line from some of my own wiki docs on using IRremote with KontroLIR._

> _You need to comment out the line with the "#error" statement and un-comment the line defining "IR_USE_TIMER1"_

ArminJo, 2021-01-02:
@AnalysIR 
Thanks,
but remove comment is abiguous for lines like
`//#define USE_PIN_CHANGE_INTERRUPT_D0_TO_D7  // using PCINT2_vect - PORT D`
But I have no good idea either.
Any suggestions are welcome!!!

Insomniac80, 2021-01-02:
I´m also from Germany and know it as comment out (add slashes) and uncomment (remove slashes) :D
So I have now removed the slashes, and ensured the file was re-read when compiling. Sadly, it did not change anything. Output of the IR receiver is the same, loop does same as before - nothing. As soon as I upload my old Panasonic-workaround, receiver responds..

ArminJo, 2021-01-02:
Can you please post the output again?
It should start with
```
START ../src/IRreceiveDump.cpp from Jan  2 2021
Using library version 2.8.2
Ready to receive IR signals at pin 11

Protocol=DENON Address=0x6 Command=0x5A Raw-Data=0x1968 (15 bits)
Result in internal ticks (50 us) - with leading gap
```
after reset

Insomniac80, 2021-01-02:
Sure, again used Arduino Nano and TSOP4838

IReceiveDump:

```
START C:\Users\Inso\Documents\Arduino\libraries\Arduino-IRremote-master\examples\IRreceiveDump\IRreceiveDump.ino from Jan  2 2021
Ready to receive IR signals at pin 11

Protocol=DENON Address=0x5140 Command=0x0 Raw-Data=0x0 (48 bits)

Result in internal ticks (50 us) - with leading gap
rawData[100]: 
     -65535
     +  66,-  34     +  8,-  8     +  8,-  9     +  8,-  25
     +  8,-  9     +  8,-  25     +  8,-  8     +  8,-  26
     +  8,-  8     +  8,-  9     +  8,-  25     +  8,-  9
     +  8,-  8     +  8,-  25     +  8,-  26     +  8,-  8
     +  8,-  9     +  8,-  8     +  8,-  9     +  8,-  9
     +  8,-  8     +  8,-  9     +  8,-  8     +  8,-  26
     +  8,-  8     +  8,-  25     +  8,-  9     +  8,-  9
     +  8,-  8     +  8,-  25     +  8,-  9     +  8,-  25
     +  8,-  9     +  8,-  8     +  8,-  9     +  8,-  9
     +  8,-  8     +  8,-  9     +  8,-  8     +  8,-  9
     +  8,-  9     +  8,-  25     +  8,-  8     +  8,-  9
     +  8,-  9     +  8,-  25     +  8,-  8     +  8,-  9
     +  8,-  9     +  8
Result in microseconds - with leading gap
rawData[100]: 
     -3276750
     +3300,-1700     + 400,- 400     + 400,- 450     + 400,-1250
     + 400,- 450     + 400,-1250     + 400,- 400     + 400,-1300
     + 400,- 400     + 400,- 450     + 400,-1250     + 400,- 450
     + 400,- 400     + 400,-1250     + 400,-1300     + 400,- 400
     + 400,- 450     + 400,- 400     + 400,- 450     + 400,- 450
     + 400,- 400     + 400,- 450     + 400,- 400     + 400,-1300
     + 400,- 400     + 400,-1250     + 400,- 450     + 400,- 450
     + 400,- 400     + 400,-1250     + 400,- 450     + 400,-1250
     + 400,- 450     + 400,- 400     + 400,- 450     + 400,- 450
     + 400,- 400     + 400,- 450     + 400,- 400     + 400,- 450
     + 400,- 450     + 400,-1250     + 400,- 400     + 400,- 450
     + 400,- 450     + 400,-1250     + 400,- 400     + 400,- 450
     + 400,- 450     + 400

Result as internal ticks (50 us) array - always without leading gap
uint8_t rawTicks[99] = {66,34, 8,8, 8,9, 8,25, 8,9, 8,25, 8,8, 8,26, 8,8, 8,9, 8,25, 8,9, 8,8, 8,25, 8,26, 8,8, 8,9, 8,8, 8,9, 8,9, 8,8, 8,9, 8,8, 8,26, 8,8, 8,25, 8,9, 8,9, 8,8, 8,25, 8,9, 8,25, 8,9, 8,8, 8,9, 8,9, 8,8, 8,9, 8,8, 8,9, 8,9, 8,25, 8,8, 8,9, 8,9, 8,25, 8,8, 8,9, 8,9, 8};  // Protocol=DENON Address=0x5140 Command=0x0 Raw-Data=0x0 (48 bits)

Result as microseconds array - with leading gap for IRremote versions 2.*
uint16_t rawData[100] = {65486, 3300,1700, 400,400, 400,450, 400,1250, 400,450, 400,1250, 400,400, 400,1300, 400,400, 400,450, 400,1250, 400,450, 400,400, 400,1250, 400,1300, 400,400, 400,450, 400,400, 400,450, 400,450, 400,400, 400,450, 400,400, 400,1300, 400,400, 400,1250, 400,450, 400,450, 400,400, 400,1250, 400,450, 400,1250, 400,450, 400,400, 400,450, 400,450, 400,400, 400,450, 400,400, 400,450, 400,450, 400,1250, 400,400, 400,450, 400,450, 400,1250, 400,400, 400,450, 400,450, 400};  // Protocol=DENON Address=0x5140 Command=0x0 Raw-Data=0x0 (48 bits)

uint16_t address = 0x5140;
uint16_t command = 0x0;
uint32_t data = 0x0;

Pronto Hex as string
char ProntoData[] = "0000 006D 0032 0000 007F 0041 000F 000F 000F 0011 000F 0030 000F 0011 000F 0030 000F 000F 000F 0032 000F 000F 000F 0011 000F 0030 000F 0011 000F 000F 000F 0030 000F 0032 000F 000F 000F 0011 000F 000F 000F 0011 000F 0011 000F 000F 000F 0011 000F 000F 000F 0032 000F 000F 000F 0030 000F 0011 000F 0011 000F 000F 000F 0030 000F 0011 000F 0030 000F 0011 000F 000F 000F 0011 000F 0011 000F 000F 000F 0011 000F 000F 000F 0011 000F 0011 000F 0030 000F 000F 000F 0011 000F 0011 000F 0030 000F 000F 000F 0011 000F 0011 000F 0806 "
```

IReceiveDemo:
```
START C:\Users\Inso\Documents\Arduino\libraries\Arduino-IRremote-master\examples\IRreceiveDemo\IRreceiveDemo.ino from Jan  2 2021
Enabling IRin
Ready to receive IR signals at pin 11
Protocol=DENON Address=0x5140 Command=0x0 Raw-Data=0x0 (48 bits)

Protocol=DENON Address=0x5140 Command=0x0 Repeat gap=73700us Raw-Data=0x0 (48 bits)
```

ArminJo, 2021-01-02:
OK now i got it.
You have the Denon variant of the Kaseikyo protocol 😀 
Since the Kaseiko protocol has 48 bits it has now raw data which is limited to 32 bits.
I see I have to change the protocol names!

ArminJo, 2021-01-02:
> I see I have to change the protocol names!

Done.
You have to use     
`    IrSender.sendKaseikyoStandard(0x5140 , 0, DENON_VENDOR_ID_CODE, sRepeats);`

AnalysIR, 2021-01-02:
> @AnalysIR
> Thanks,
> but remove comment is abiguous for lines like
> `//#define USE_PIN_CHANGE_INTERRUPT_D0_TO_D7 // using PCINT2_vect - PORT D`
> But I have no good idea either.
> Any suggestions are welcome!!!

For me it would not be ambiguous (in this context)

However, in this case un-comment the line.....should do fine.

(referencing 'the line' should remove any potential ambiguity.)

_FYI: I have never noticed any issue understanding anything you posted (wrt English). I wouldn't be too concerned about it. _

Insomniac80, 2021-01-02:
> > I see I have to change the protocol names!
> 
> Done.
> You have to get again the latest version from the repo 😢 and use
> ` IrSender.sendKaseikyoStandard(0x5140 , 0, DENON_VENDOR_ID_CODE, sRepeats);`

I have tried with several IR codes now, and all I can say is:
Works like a charm! :D
Thank you very much for your help and work! (Und ein schönes Restwochenende^^)

closed 2021-01-02 by Insomniac80