Arduino-IRremote/Arduino-IRremote #747: IRremote sending works only once with Arduino Nano

trueToastedCode, 2021-01-03:
Board: Arduino Nano with ATmega328P
Protocol: Unknown

I am trying to transmit a color code to an RGBW Controller, wich is not working. I have the right codes, i know because i found the same on other peoples project on github. I believe i have to send it at least three times, probably really close, but i does not work either.

- I tried several lib versions including the newest
- "Send" message keeps printing in the console
- Only works again when resetting
- Tried sending raw and with nec
- Tried 2 different LED's

```
#include <IRremote.h>

IRsend irsend;

void setup() {
  Serial.begin(9600);
}

const static unsigned int raw[68] = {8950,4450,550,550,600,550,600,500,600,550,550,550,600,500,600,550,600,500,600,1600,600,1600,600,1650,550,1650,550,550,600,1600,600,1600,600,1650,550,550,600,1600,600,1600,600,550,550,550,600,500,600,550,550,550,600,1600,600,550,550,550,600,1600,600,1600,600,1600,600,1600,600,1650,550};

void loop() {  
  Serial.println("Send");
  for (int i=0; i<3; i++) {
    irsend.sendRaw(raw,68,38);
    delay(40);
  }
  delay(5000);
}
```

ArminJo, 2021-01-03:
> Only working on PIN13 / In the Doku it sais it should be 3 and 9

This means, it works.

>     * Only works again when resetting

This means, it works at least one time 😀 

> "Send" message keeps printing in the console

As expected 👍 .

trueToastedCode, 2021-01-03:
> > Only working on PIN13 / In the Doku it sais it should be 3 and 9
> 
> This means, it works.
> 
> > ```
> > * Only works again when resetting
> > ```
> 
> This means, it works at least one time 😀
> 
> > "Send" message keeps printing in the console
> 
> As expected 👍 .

closed 2021-01-03 by trueToastedCode