Arduino-IRremote/Arduino-IRremote #748: ESP32 - BLE callback

tomaculum, 2021-01-05:
### Board
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [x] ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266
* [ ] Teensy board
* [ ] Other - please specify

### Protocol
* [ ] Unknown
* [ ] BoseWave
* [ ] Denon
* [ ] Dish
* [ ] JVC
* [ ] Lego
* [ ] LG
* [x] NEC
* [ ] Panasonic
* [ ] RC5, RC6
* [ ] Samsung
* [ ] Sanyo
* [ ] Sharp
* [ ] Sony
* [ ] Whynter
* [ ] Other - please specify

**Code Block:**

[gist](https://gist.github.com/tomaculum/b53c23a31aa112eb8052427877e15896)

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant 

--------

Hi all,

I'm struggling with my implementation of a bluetooth controlled esp32 remote control.

Everything seems to work as far as I expect it to work but something strange is happening if I try to send the IR commands from the BLE callback in line 43 or 58.

If I send the power command from the "loop" function (line 25), it works as expected. If I call the same function from the callback class, it outputs the expected output in the serial console but does not send the ir command.

```
Send Logitech Command 128 0    // loop: works
Send Logitech Command 128 0    // loop: works
onRead POWER UUID
Send Logitech Command 128 0    // callback line 43: not working
new value: 10
DETECTED POWER UUID
Send Logitech Command 128 0    // callback line 58: not working
```

Did someone had any similar problem or knows how to call it from within a callback/ class properly?

ArminJo, 2021-01-05:
ESP32 uses `ledcWrite(LED_CHANNEL, IR_SEND_DUTY_CYCLE)` maybe this does not work in interrupt context.
Anyway, it is "bad" style to have a long running action in an Interrupt Callback, since it blocks all subsequent interrupts.
So store the values and check them in the loop and send it from there.

tomaculum, 2021-01-05:
Alright, thank you very much! (:

closed 2021-01-05 by tomaculum