Arduino-IRremote/Arduino-IRremote #749: Issues on IRremote 2.8.1 @ 4MHz ?

AnalysIR, 2021-01-05:
### Board
* [ X] Arduino ATmega328* board (UNO, Nano)

### Protocol
* [X ] NEC

- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used

I have a system that can be configured to run at 16Mhz, 8Mhz & 4Mhz depending on whether powered from USB or battery.

with the latest 2.8.1 release I have noticed some issues @ 4mHz (F_CPU)

Specifically, the signals sent are not valid & on first look seem much shorter and incomplete.

I don't have time to look into it further at the moment & will revert to 2.6.1 for now. Do you have any idea what might be the cause? There was nothing obvious to me after some investigation. The carrier freq seems fine - the issue is just with the pulse durations.

It doesn't appear (???) to be related to SYSCLOCK or F_CPU based on initial check - but I haven't kept track of all the changes in 2.8.x. 

UART is working fine @4MHz etc.

When I compile on 2.6.1 everything is fine @4MHz.

The only thing I can think of is all of the additional loops/function_call overhead related to **_decodePulseDistanceData_** and similar changes etc. This overhead might be material @4MHz vs previous versions.

The other candidates could include any new/updated delay functions (if any).

Happy to test out any Suggestions :)

ArminJo, 2021-01-06:
I have to setup for 4 MHz, maybe I test it on my Digisparks tomorrow.

AnalysIR, 2021-01-06:
Possible Cause in mark & space functions??

use of [delayMicroseconds()](https://www.arduino.cc/reference/en/language/functions/time/delaymicroseconds/)

> Currently, the largest value that will produce an accurate delay is 16383.

It is possible that @8&4MHz this limit is even less....(not specifically mentioned but could explain my issue)

Either way this is a likely bug if the largest mark or space is 16383uSecs (or less) at all clock frequencies.

I suggest you revert these delays to what was in 2.6.1/2.7 and earlier.

```
    if (time > 0) {
        custom_delay_usec(time);
    }
```

**UPDATE**: Just made the above change on my local system & it now works @ 4Mhz. _(time is now called timeMicros)_

So I suggest you create a v2.8.2 with the above change and remove 2.8.0 & 2.8.1 - as any mark or space above 16383uSecs will be broken in 2.8.0 & 2.8.1 (more likely to occur with spaces than marks)

AnalysIR, 2021-01-06:
FYI...forgot to mention earlier

you can set up 8MHz or 4MHz etc  on a UNO or Nano using the Mini Core...to select the clock from the IDE menu

....then just set the system clock divider in the **CLKPR** register as the first instruction in setup() 
e.g. DIV/4 to go to 4MHz, DIV/2 to go to 8MHz etc etc

Lower BAUD rates may be required at lower clocks (I only use 19,200bps @ 4MHz)

_(You may also need to upload the bootloader for the mini core first)_

ArminJo, 2021-01-06:
It is not as easy 😢 `custom_delay_usec(time); ` does not work on an ATtiny85 with 1 MHz. It results in a delay of 760 us instead of the requested 560 us.
And you are right, the delayMicroseconds() does not work for lower frequencies, since the arduino guys are reluctant to merge PR's which support lower frequencies.
Maybe is is is best to copy the function from such a PR ( but I do not remember which).
Or maybe you know a stable implementation?

ArminJo, 2021-01-06:
The reason for the 16383uSecs max limit of delayMicroseconds() is in wiring.c line 175 and herefore less critical for 8 and disappears for 4 Mhz.

> As it is now there is a significant bug for all Arduino (UNO/Nano/Leonardo/Micro) systems @16mhz, 8Mhz & 4Mhz in 2.8.0 & 2.8.1 and this should be corrected ASAP. (IMHO)

The bug can only appear for sendRaw.
You have more experience with the sendRaw function, please tell me for what purpose one need marks or spaces of more than 16  and less than 64 milliseconds. Maybe we had to add the 16383uSecs @16MHz check again.

**And if you can do me a favor, please check the latest commit. It should work now for 4 MHz. THANKS!**

AnalysIR, 2021-01-06:
> ... please tell me for what purpose one need marks or spaces of more than 16 and less than 64 milliseconds

I can't readily recall where I have seen a mark longer than 16383uSecs, but spaces are common...primarily as gaps between signals. e.g. the gaps between repeats on NEC signals, Sharp is another example.

It seems like there are lots of new function prototypes in the master, so using the new Master irSend.cpp doesn't compile.

So I just copied over the assembly code and got some compile issues:
```
C:\Users\xxxx\Documents\Arduino\libraries\IRremote\src\irSend.cpp: In member function 'void IRsend::mark_long(uint32_t)':
C:\Users\xxxx\Documents\Arduino\libraries\IRremote\src\irSend.cpp:148:46: error: 'us' was not declared in this scope
         "brne 1b" : "=w" (timeMicros) : "0" (us) // 2 cycles
                                              ^~
C:\Users\xxxx\Documents\Arduino\libraries\IRremote\src\irSend.cpp:148:46: note: suggested alternative: 'u8'
         "brne 1b" : "=w" (timeMicros) : "0" (us) // 2 cycles
                                              ^~
                                              u8
C:\Users\xxxx\Documents\Arduino\libraries\IRremote\src\irSend.cpp: In member function 'void IRsend::space_long(uint32_t)':
C:\Users\xxxx\Documents\Arduino\libraries\IRremote\src\irSend.cpp:189:46: error: 'us' was not declared in this scope
         "brne 1b" : "=w" (timeMicros) : "0" (us) // 2 cycles
                                              ^~
C:\Users\xxxx\Documents\Arduino\libraries\IRremote\src\irSend.cpp:189:46: note: suggested alternative: 'u8'
         "brne 1b" : "=w" (timeMicros) : "0" (us) // 2 cycles
                                              ^~
                                              u8
```
Note: obviously this '4MHz' code would only work with **AVRs**

Also, I should note that I have been compiling with another core @4Mhz recently for the ATmega328PB (by watterott). Maybe this core does have support for delayMicroseconds @ 4Mhz. I suspect the minicore could have it also, but not sure. It is understandable that Arduino official platforms don't support 4Mhz as they don't produce that option.

I may not have much time to follow up today, but will come back to it in the next few days.

ArminJo, 2021-01-06:
> spaces are common...primarily as gaps between signals. e.g. the gaps between repeats on NEC signals, Sharp is another example.

But receiving stops after _GAP microseconds (5 ms)!
Longer spaces were used for LEGO, but I changed the "strange" implementation.

> So I just copied over the assembly code and got some compile issues:

Sorry for this, I amended a change to my commit.

AnalysIR, 2021-01-06:
Seems like the latest ASM code now works (on Master) @4MHz ...tnx

> 
> But receiving stops after _GAP microseconds (5 ms)!
> Longer spaces were used for LEGO, but I changed the "strange" implementation.

This was all discussed years ago due to a series of bugs being opened, which is why we moved to the `custom_delay_usec()` approach. _(from memory)_
The 16383uSecs issue is not about receiving by IRremote, it is about sending. For more than a few protocols, there is a gap longer than 16383uSecs & thus the 2.8.x code breaks. 

It may be possible to control the implementation of protocols supported internally within IRremote to avoid all this and put the responsibility for sending longer gaps on the individual coder.
However, sendRAW implementations cover marks/spaces up to ~65kuSecs (uint16_t) long. The 2.8.x code is currently broken for this usage due to the limitation of delayMicroseconds of 16383uSecs.

In my opinion, the previous approach(2.6.1) was more elegant/cross-platform/generic etc & I suggest you consider it a special case for the tiny @1Mhz & similar as required.

ArminJo, 2021-01-15:
I added a check `if (timeMicros >= 0x4000) {` to be backwards compatible.
Thanks for reporting
Armin

closed 2021-01-15 by ArminJo

AnalysIR, 2021-01-15:
```
   if (timeMicros >= 0x4000) {
        // The implementation of Arduino delayMicroseconds() overflows at 0x4000 / 16.384 @16MHz (wiring.c line 175)
        // But for sendRaw() and external protocols values between 16.384 and 65.535 might be required
        // Use delay(), this in calls yield which is required on some platforms to work properly
        delay(timeMicros / 1000);
```
This is now only accurate to within 1,000 uSecs ???. Depending on when it is called there could be additional offsets within the 

A better approach for this  specific case (IMHO) would be to use:
custom_delay_usec(timeMicros)

For reference the delay() source is below:
```
void delay(unsigned long ms)
{
	uint32_t start = micros();
	while (ms > 0) {
		yield();
		while ( ms > 0 && (micros() - start) >= 1000) {
			ms--;
			start += 1000;
		}
	}
}
```

_Note: Of course reverting to the way it was done in 2.6.1 and earlier would still be my preferred method._

ArminJo, 2021-01-15:
Do not tell me that the delays between the commands are not more or less a guess, so nobody requires it to be more exact than 1.5% 😀

AnalysIR, 2021-01-15:
It is not only delays between individual signals. Some signals)protocols) will have long gaps embedded in them.

Inaccuracies will only come into play at the 'performance edge'. If all things are operating well, IR usually will work with inaccurate signal timing - provided the signal is well formed. 

Good quality IR receivers can distort timings by up to +/- 200uSecs. (usually less).
IR remote RAW captures have a granularity of +/- 50uSecs built-in by default.
.....it can all add up to poorer performance/range.

However, I don't really understand why less accurate timing should be designed in when it can easily be avoided.

ArminJo, 2021-01-15:
Really interesting.. do you have examples for protocols with such long gaps?
I hesitate to revert, because of the massive problems I saw with my digisparks.
Maybe I should test again...

AnalysIR, 2021-01-15:
NEC GAPs between repeats
XMP
NIKON (camera)
a few AIR Conditioner protocols.
...etc

Going forward, I will look into creating a fork which will just work seamlessly for our own use cases.

ArminJo, 2021-01-15:
I do not understand. 
Are you sending all these codes as raw data? 
And you know that the receivers will fail if you have -6% in long timings?
Do you want to have a compile switch for supporting these exact long timings? That is no problem!

AnalysIR, 2021-01-16:
> I do not understand.
> What is your use case?
> Are you sending all these codes as raw data?

We have several products that use or have the option of using IRremote to send IR signals.
As IRremote only supports a relatively small subset of protocols natively we will use sendRAW  sometimes and other times we will call the mark() & space() functions directly from user code.

> And you know that the receivers will fail if you have -6% in long timings? 

No, this will always depend on the receiving decoder, which for us is usually an unknown factor...see my earlier post about 'performance edge/range'

Do you want to have a compile switch for supporting these exact long timings? That is no problem!

You can park this issue for now as we will look into creating a fork for our own needs (if needed).

ArminJo, 2021-01-24:
I introduced a compile flag USE_CUSTOM_DELAY.