Arduino-IRremote/Arduino-IRremote #752: Update the repo URL

Neilblaze, 2021-01-13:
Currently, the repo link redirects to :arrow_right: `z3t0.github.io/arduino-irremote`, which returns 404. The link should be updated to :arrow_right: `https://arduino-irremote.github.io/Arduino-IRremote`

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The title of the issue is helpful and relevant

ArminJo, 2021-01-13:
There are a lot of links in this repo. Can you specify?

Neilblaze, 2021-01-13:
@ArminJo this one :point_down: 
[image]

ArminJo, 2021-01-13:
Thanks 👍 
I never used it and there is no website for this repo any more, so I deleted it.

closed 2021-01-13 by ArminJo