Arduino-IRremote/Arduino-IRremote #756: Command always is zero although raw shows otherweise

omercade, 2021-01-17:
Hello,
I was using the library before and with the last update ( and mandatory adaptations ) it stopped working, basically i get always this data:

Using library version 2.9.0
Enabling IRin
Ready to receive IR signals at pin 11
Protocol=SONY Address=0x0 Command=0x0 Raw-Data=0x2F0 (12 bits)

and Command is empty. I don't have any way to get Raw-Data into variable, as IrReceiver.decodedIRData.rawData ( or Raw-data or raw-data or rawdata ) is not existing.

Basically before i could manage to get the received command in HEX in a variable and now it seems impossible, as the only hint of it is in the  IrReceiver.printIRResultRawFormatted(&Serial, true); that prints

"Protocol=SONY Address=0x0 Command=0x0 Raw-Data=0xA70 (12 bits)
20
0)" 

but there is no way to get the "0xA70 " which i suppose it should be in command or raw-data field of IrReceiver.decodedIRData

Thanks 

### Board
* [X ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266
* [ ] Teensy board
* [ ] Other - please specify

### Protocol
* [ ] Unknown
* [ ] BoseWave
* [ ] Denon
* [ ] Dish
* [ ] JVC
* [ ] Lego
* [ ] LG
* [ ] NEC
* [ ] Panasonic
* [ ] RC5, RC6
* [ ] Samsung
* [ ] Sanyo
* [ ] Sharp
* [X] Sony
* [ ] Whynter
* [ ] Other - please specify

**Code Block:**
```

 if (IrReceiver.decode()) {
        // Print a short summary of received data
        IrReceiver.printIRResultShort(&Serial);
        if (IrReceiver.decodedIRData.protocol == UNKNOWN) {
            // We have an unknown protocol, print more info
            IrReceiver.printIRResultRawFormatted(&Serial, true);
        }
           
        
       Serial.println(IrReceiver.decodedIRData.protocol );
       Serial.println(IrReceiver.decodedIRData.command );
        Serial.println();

```

**checklist:**
- [X] I have **read** the README.md file thoroughly
- []X I have searched existing issues to see if there is anything I have missed.
- [X] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [X] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [X] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

ArminJo, 2021-01-17:
I see `Protocol=SONY Address=0x2 Command=0x34 Raw-Data=0x134 (12 bits)` for me.
Please use the lates download (not release) from the repo There were a few changes.

omercade, 2021-01-17:
thanks, i'll try to import it to arduino web coding as a custom project then to get the very latest.

omercade, 2021-01-17:
With latest version (master worked) so the not working one is the one published in arduino editor plugin manager.
Thanks for the support, very appreciated!

closed 2021-01-17 by omercade

Rhragnarok, 2021-01-23:
So which version should we be using as I am having a similar issue.  Can you give a version number for me to try?   I don't get what you mean by "use the latest download (not release)"...

ArminJo, 2021-01-23:
[This](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) is always the latest repo version.