Arduino-IRremote/Arduino-IRremote #760: IRSend NEC Argument/syntax?

Fffrank, 2021-01-19:
### Board
* [ X] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266
* [ ] Teensy board
* [ ] Other - please specify

### Protocol
* [ ] Unknown
* [ ] BoseWave
* [ ] Denon
* [ ] Dish
* [ ] JVC
* [ ] Lego
* [ ] LG
* [X ] NEC
* [ ] Panasonic
* [ ] RC5, RC6
* [ ] Samsung
* [ ] Sanyo
* [ ] Sharp
* [ ] Sony
* [ ] Whynter
* [ ] Other - please specify

**checklist:**
- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] The latest [repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) is used
- [X] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [X] The title of the issue is helpful and relevant 

Here's the data that I've captured:
```
Protocol=NEC Address=0x0 Command=0x58 Raw-Data=0xA758FF00 (32 bits)
```

Here's how I'm trying to send it through an IR led --- I have confirmed with my cell phone that the LED is lighting, but my device is not responding?

```
    IrSender.sendNEC(0xA758FF00, 32);
```

ArminJo, 2021-01-19:
Please use sendNECRaw() or sendNEC(uint16_t aAddress, uint8_t aCommand, uint8_t aNumberOfRepeats, bool aIsRepeat).

Fffrank, 2021-01-19:
That worked -- would be nice to add that to the examples file since there are no docs.  I appreciate the response!

NW93, 2021-01-31:
Hello
I'm trying to send NEC 32bit code but my device is not responding I have the same problem can you share a simple code how to do that please.

Greetings.

DMladenov1975, 2021-02-01:
With IRemote v3.0.0 try the new command sendNECMSB(0xXXXXXXXX,32);
I tested it and works.

NW93, 2021-02-01:
> With IRemote v3.0.0 try the new command sendNECMSB(0xXXXXXXXX,32);
> I tested it and works.

I'll try it. Thanks for responding.
Greetings.

NW93, 2021-02-01:
> With IRemote v3.0.0 try the new command sendNECMSB(0xXXXXXXXX,32);
> I tested it and works.

Unfortunately it doesn't work. can you show me a sample code please ?

ArminJo, 2021-02-01:
Have you looked in the IRsendDemo example?

DMladenov1975, 2021-02-02:
> > With IRemote v3.0.0 try the new command sendNECMSB(0xXXXXXXXX,32);
> > I tested it and works.
> 
> Unfortunately it doesn't work. can you show me a sample code please ?

Here is a simple example:

#include <IRremote.h>
void setup() {
	IrSender.begin(true); // Enable feedback LED
}
void loop() {
	IrSender.sendNECMSB(0x00FFFB12, 32); //old format
	IrSender.sendNECRaw(0xCC34FF00, 0); //new
	delay(5000);
}

closed 2021-02-02 by ArminJo

NW93, 2021-02-02:
> > > With IRemote v3.0.0 try the new command sendNECMSB(0xXXXXXXXX,32);
> > > I tested it and works.
> > 
> > Unfortunately it doesn't work. can you show me a sample code please ?
> 
> Your LED cathode must be wired to Arduino pin 3 (uno/nano) and anode to +5V via 300Ohm resistor !
> Here is a simple example:
> 
> #include <IRremote.h>
> void setup() {
> 	IrSender.begin(true); // Enable feedback LED
> }
> void loop() {
> 	IrSender.sendNECMSB(0x00FFFB12, 32); //old format MSB first
> // To convert one into the other, you must reverse the byte positions and then reverse all positions of each byte.
> 	IrSender.sendNECRaw(0x847BFF00, 0); //new
> 	delay(5000);
> }
> 
> 

Finally it works. Thank you so much man for your help.

Greetings.👍👍👍