Arduino-IRremote/Arduino-IRremote #762: Updated Library in previousl;y working sketch; error verify/compiling for NANO. Help?

itsjustme0661, 2021-01-19:
HI -  I m very new to this.  Had a sketch working well with previous version.
Updated to latest. Noted many more include lines
failed "verify compile" step for NANO/328
I dont understand the note below about if more than 30 lines code do "
### Board
* [X ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266
* [ ] Teensy board
* [ ] Other - please specify

### Protocol
* [X ] Unknown
* [ ] BoseWave
* [ ] Denon
* [ ] Dish
* [ ] JVC
* [ ] Lego
* [ ] LG
* [ ] NEC
* [ ] Panasonic
* [ ] RC5, RC6
* [ ] Samsung
* [ ] Sanyo
* [ ] Sharp
* [ ] Sony
* [ ] Whynter
* [ ] Other - please specify

**Code Block:**
Here is the error text from the compile log window, truncated to the area where the com[pile appeared to fail.

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Users\\glena\\Dropbox (Personal)\\Arduino_Sketches\\libraries\\arduino_189866\\src" "C:\\Users\\glena\\Dropbox (Personal)\\Arduino_Sketches\\libraries\\arduino_189866\\src\\ir_Whynter.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Users\\glena\\Dropbox (Personal)\\Arduino_Sketches\\libraries\\arduino_189866\\src" "C:\\Users\\glena\\Dropbox (Personal)\\Arduino_Sketches\\libraries\\arduino_189866\\src\\nRF5.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Users\\glena\\Dropbox (Personal)\\Arduino_Sketches\\libraries\\arduino_189866\\src" "C:\\Users\\glena\\Dropbox (Personal)\\Arduino_Sketches\\libraries\\arduino_189866\\src\\sam.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Users\\glena\\Dropbox (Personal)\\Arduino_Sketches\\libraries\\arduino_189866\\src" "C:\\Users\\glena\\AppData\\Local\\Temp\\arduino_build_339388\\sketch\\GFL_remote_0.6A.ino.cpp" -o "C:\\Users\\glena\\AppData\\Local\\Temp\\arduino_build_339388\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\glena\\AppData\\Local\\Temp\\arduino_build_339388\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Users\\glena\\Dropbox (Personal)\\Arduino_Sketches\\libraries\\arduino_189866\\src" "C:\\Users\\glena\\AppData\\Local\\Temp\\arduino_build_339388\\sketch\\GFL_remote_0.6A.ino.cpp" -o "C:\\Users\\glena\\AppData\\Local\\Temp\\arduino_build_339388\\sketch\\GFL_remote_0.6A.ino.cpp.o"
In file included from C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\GFL_remote_0.6A\GFL_remote_0.6A.ino:5:0:
C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\arduino_189866\src/TinyIRReceiver.h:87:8: error: redefinition of 'struct TinyIRReceiverStruct'
 struct TinyIRReceiverStruct {
        ^~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\arduino_189866\src/TinyIRReceiver.cpp.h:37:0,
                 from C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\GFL_remote_0.6A\GFL_remote_0.6A.ino:4:
C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\arduino_189866\src/TinyIRReceiver.h:87:8: note: previous definition of 'struct TinyIRReceiverStruct'
 struct TinyIRReceiverStruct {
        ^~~~~~~~~~~~~~~~~~~~
Using library arduino_189866 at version 2.8.1 in folder: C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\arduino_189866 
exit status 1
Error compiling for board Arduino Nano.

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

ArminJo, 2021-01-19:
You declared the structure `struct TinyIRReceiverStruct` in your source GFL_remote_0.6A.ino.
And please do not check 
`[x] The latest [repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) is used`
if you use the old version 2.8.1 !

itsjustme0661, 2021-01-19:
Hi, and thanks for the quick reply, just getting back to this.

First, I'm confused on both and need a little more clarification. 

1. I don't follow what you are saying about "tiny receiver structure". I declared nothing on my own. I do see an include of "tiny receiver"  - that was added automatically when i included the updated library ( and confused me too, but who am I to argue with the author?).  So what do i do exactly?

The following were automatically added when "included" the library. Previously it was only one "include" line
#include <digitalWriteFast.h>
#include <IRremote.h>
#include <LongUnion.h>
#include <TinyIRReceiver.cpp.h>
#include <TinyIRReceiver.h>

2. I am also confused on the version. I did my best to fill out the questionnaire.  When i added the library using the library manager there was a drop-down menu with versions. I let it default to the latest, which i think was 2.8.1. SO i indicated i was using the latest. Is there something later? Or ??????

I take it i need to remove something from my sketch. Is it both references to tiny receiver?

Thank you

G

ArminJo, 2021-01-19:
Which IDE do you use?
There was a flaw i corrected. Can you try the latest version from the repository https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip, maybe you must install it manually or try to remove and add the library.
Thanks for your patience and for helping to improve the usability.

itsjustme0661, 2021-01-19:
Hi -

I use Arduino 1.8.13. This version on Windows, but also have it on MacOS.

Sorry to be inexperienced, but how do i remove manually?  I have only added libraries using the library manager.

I did try the following: i removed all the "include" statements, except "#include <IRremote.h>". This was all that the original version had.

It now compiles and verifies. I have NOT, however, uploaded and run it since this is being plugged into a much larger project PCB with many "moving parts" and i cannot suffer a disruption / failure of the logic control right now.

Thanks again,

G

itsjustme0661, 2021-01-20:
Hi -I have located the libraries and presume its as simple as removing one library folder and replacing it with another.  Pls confirm.

Assuming that is correct, i it possible to have an older (known working) and new version installed?  And if so how do i distinguish between the two when i  "include" it?  

Thank you.

G

itsjustme0661, 2021-01-20:
Hi

I downloaded version 3.0 (pointed to above) sand tried to install.  I could not distinguish at include between various versions, but it selected the latest (confirmed during verbose compilation ).

When i "included" it still added a total of 5 include statements, including the offending "..tiny..."

This has not changed.

I have removed all include statements except the "include<IRremote.h>"

Correct?  How do i revert to an earlier version if this breaks my application?

TIA,

G

itsjustme0661, 2021-01-20:
... and it does not respond with my sketch. I will try to debug but it is very difficult since this is an embedded system and i don't have spare receivers handy to test locally.... Can you tell me (ver 3.0 which you pointed me to) if this has to do with the several include statements which i removed (so it would compile) or other?

At the moment i simply want to revert to what was working - current as of about 1.5 weeks ago. What version would that be and how do i ensure it is the only version in my library?

itsjustme0661, 2021-01-21:
When i leave all "#includes" it again fails to compile:

Here are the error messages.

My device is a NANO / 328. It is selected.

Arduino: 1.8.13 (Windows 10), Board: "Arduino Nano, ATmega328P"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\glena\AppData\Local\Temp\arduino_build_261025 -warnings=none -build-cache C:\Users\glena\AppData\Local\Temp\arduino_cache_556971 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\GFL_remote_0.6A\GFL_remote_0.6A.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\glena\AppData\Local\Temp\arduino_build_261025 -warnings=none -build-cache C:\Users\glena\AppData\Local\Temp\arduino_cache_556971 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\GFL_remote_0.6A\GFL_remote_0.6A.ino

Using board 'nano' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr

Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr

Detecting libraries used...

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "C:\\Users\\glena\\AppData\\Local\\Temp\\arduino_build_261025\\sketch\\GFL_remote_0.6A.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for digitalWriteFast.h: [arduino_189866@2.8.1 Arduino-IRremote-master@3.0.0]

ResolveLibrary(digitalWriteFast.h)

  -> candidates: [arduino_189866@2.8.1 Arduino-IRremote-master@3.0.0]

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Users\\glena\\Dropbox (Personal)\\Arduino_Sketches\\libraries\\Arduino-IRremote-master\\src" "C:\\Users\\glena\\AppData\\Local\\Temp\\arduino_build_261025\\sketch\\GFL_remote_0.6A.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Using cached library dependencies for file: C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\Arduino-IRremote-master\src\IRremote.cpp

Using cached library dependencies for file: C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\Arduino-IRremote-master\src\esp32.cpp

Using cached library dependencies for file: C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\Arduino-IRremote-master\src\irPronto.cpp

Using cached library dependencies for file: C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\Arduino-IRremote-master\src\irReceive.cpp

Using cached library dependencies for file: C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\Arduino-IRremote-master\src\irSend.cpp

Using cached library dependencies for file: C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\Arduino-IRremote-master\src\ir_BoseWave.cpp

[... cut ...]
C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\Arduino-IRremote-master\src/TinyIRReceiver.cpp.h:183: undefined reference to `handleReceivedTinyIRData(unsigned int, unsigned char, bool)'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "digitalWriteFast.h"

 Used: C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\Arduino-IRremote-master

 Not used: C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\arduino_189866

Using library Arduino-IRremote-master at version 3.0.0 in folder: C:\Users\glena\Dropbox (Personal)\Arduino_Sketches\libraries\Arduino-IRremote-master 

exit status 1

Error compiling for board Arduino Nano.

itsjustme0661, 2021-01-21:
... and if i remove the includes that reference  "tiny..."

1. it compiles and verifies
2. it uploads
3. it returns "0" as the HEX code for every key. Not even "0x0"
G

itsjustme0661, 2021-01-21:
OK, i can refine my question. I found where my problem lies.

In the sketch folder, there is a library folder. It has the added libraries, such as "IRremote".

It also has libraries with "Arduino - xxxcode". I assumed these were reserved Arduino files. Not entirely. They turned out to be older archived versions of IRremote.  If i erased the newer versions, the "newest old one" is selected at compile/include time.  But clearly these are made and named by Arduino and i don't entirely know what the consequences might be of me playing with them- for example re-naming so i recognize them.

Any wisdom welcome.

G

itsjustme0661, 2021-01-21:
I have finally figured out how to get the old version back, but really would like any pointers to learn, simply, how to update the old code.

ArminJo, 2021-01-21:
maybe it is a good idea if you can send me your (all files) code in an email to armin.arduino@gmail.com
Then I have the chance to improve the converting instructions und you get an updated code, which you can compare with your current one.
And maybe for deleting old versions you must clean also files from C:\Users\<NAME>\AppData\Local\Arduino15\staging\libraries

itsjustme0661, 2021-01-21:
> maybe it is a good idea if you can send me your (all files) code in an email to armin.arduino@gmail.com <mailto:armin.arduino@gmail.com>

Hi Armin, and thank you very much. I know you do this “volunteer” work adn then get guys like me who don't have the level of understanding you do.

Godo news:  I found out the problem with old versions. Arduino’s IDE renames them “Arduino-someinscrutablecode”. To me they looked like stem libraries that i should not touch, but when i five in, it was versions 2.8.0, 2.8.1 and 2.9. So it would always select the latest — 2.9 (although it lied and said “2.8.1 installed).

So my code runs again.

But i would like to be able to take advantage of improvements in IRlibrary going forward, so I’d like to learn the new syntax.

I am not familiar with the serial() function. I have printed out the reference sections and hope to read them very soon - maybe tonight.

A little later i will assemble the following and send it to you:

1. My old code
2. The example code, from you and Ken, that i tried to modify per your instructions.  Its a mess as i kept experimenting.

 I guess that is it.

I finally found the sample code you referenced in your readme.  While it was listed as a URL, you wrote it as the URL + a lin number concatenated. I finally figured this out after about 20 “404” errors. :-)

Watch this space and many thanks.

Grant

ArminJo, 2021-01-21:
> I finally figured this out after about 20 “404” errors. :-)

Sorry for this, and thanks for reporting so it is now fixed.

itsjustme0661, 2021-01-24:
HI Armin. I sent the requested code to the gmail address you indicated.

-G

ArminJo, 2021-01-25:
Hi
Thank you very much for the attached code.
I just extended the converting documentation.
If you like to make a Quality Assurance, you are welcome 😀 
Regards
Armin

itsjustme0661, 2021-01-26:
I read your updated instructions.  I understand most of what to do, but two things are still not clear:

1. What do i do with the setup line ""decode_results results;"?  I must create something to act on - print it, switch based on its case, etc.  Its unclear what object "decode()" creates.
2. When i include the library, do i include a 5 lines:
#include <digitalWriteFast.h>
#include <IRremote.h>
#include <LongUnion.h>
#include <TinyIRReceiver.cpp.h>
#include <TinyIRReceiver.h>

Or only "#include <IRremote.h>"?

Thank you

G

ArminJo, 2021-01-26:
- Also remove the line `decode_results results` or similar, since the decoded values are now in `IrReceiver.decodedIRData` and not in `results` any more.
- only "#include <IRremote.h>!!! I fixed it in the library.properties file.

itsjustme0661, 2021-01-26:
Re: OK, so I basically understood. But its not working that way.....really confused
#6
Today at 09:36 pm
Hi, Two "dumb" questions so that i can experiment and test v3.

1. Can i rename the folder in my library to something recognizable such as "IRremote_v2.8.0"? or "IRremote_V3.0.0"?

2. If so can I simply rename the "IRremote.h" file in "src" to "IRremote_280.h" and another "IRremote_300.h" and then use the appropriate "include" commands to easily switch between them (working and testing)?
TIA,

G

ArminJo, 2021-01-26:
>     1. Can i rename the folder in my library to something recognizable such as "IRremote_v2.8.0"? or "IRremote_V3.0.0"?

Don't know just test it
> If so can I simply rename the "IRremote.h" file in "src" to "IRremote_280.h" and another "IRremote_300.h" and then use the appropriate "include" commands to easily switch between them (working and testing)?

A brilliant idea! I👍  I guess this is working, since Arduino looks for the includes and then finds the appropriate library. I have never tested it, but it should work. Please reoport. (something new learned today 😀 )

itsjustme0661, 2021-01-27:
Your should never trust a newbie like me, but I'll create a dummy workspace and see how it works. Watch this space, might take a day or tow for me to find the hours i need to set it up....

Hey, I'm learning :-)

itsjustme0661, 2021-01-28:
making some progress with managing multiple libraries, but some work top go.

HOWEVER:

do i need to replace:
"    Serial.println(results.value, HEX);"

with something else, like (this is unclear to me):

"   Serial.println(IrReceiver.decodedIRData.decodedRawData, HEX);"
???

TIA, Grant

itsjustme0661, 2021-01-28:
Hi Armin,

to the above: "yes". Again, maybe to others its clear, but to me you need to say more. I think it would be bets to explain that the object one must reference to process, assign, print, "whatever" the old "results.value" data is now in IrReceiver.decodedIRData.decodedRawData, and therefore all instances of the former (results.value) must be replaced with the latter (IrReceiver.decodedIRData.decodedRawData).

MULTIPLE LIBRARIES:
_____________________

I'm amazed, but my hack seems to work.  I dont have full integration with the Arduino IDE, but I:

1. unzipped the master IRremote library
2. modified the folder to "IRremotev3.0"
3. opened the src folder, and renamed "IRremote.h" --> "IRremotev3.h"
4. manually placed the include line "include IRremotev3.h;" in the sketch

Now i can call both v2.8 and v3.0 and have them both available in my library folder.

Or so far I can.

Grant

ArminJo, 2021-01-28:
> I think it would be bets to explain that the object one must reference to process, assign, print, "whatever" the old "results.value" data is now in IrReceiver.decodedIRData.decodedRawData, and [...]
Is this not clear enough [here ](https://github.com/Arduino-IRremote/Arduino-IRremote#converting-your-program-to-the-3x-version) results.value moved to IrReceiver.decodedIRData.decodedRawData.?
Do you think I must say it clearer like you did above? Please answer frankly.

> Now i can call both v2.8 and v3.0 and have them both available in my library folder.

Fantastic 🚀 and thanks for reporting 🥇

itsjustme0661, 2021-01-29:
Hi Armin,

You ask:  
     "Is this not clear enough here results.value moved to IrReceiver.decodedIRData.decodedRawData.?"

For anyone who understands the structure of your library or has lots of experience with the "serial" function, what you wrote is probably sufficient. And after trying almost 20 times i finally got it too.  But it was not at all clear to me initially.  When you wrote "has moved to" i assumed you were speaking of a structural change that would not affect me.  When i kept returning "0" i dug deeper.

Assuming I understand correctly, perhaps  you could explain:

1. the new "decode()" command places several pieces the decoded data in a new [variable?] [array?] named "IrReceiver.decodedIRData.decodedRawData" rather than "results.value" and "results.decode_type".  Therefore you must reference the decoded data in this [whatever the specific data structure actually is].

The problem is that while i see example code, i cannot actually find any documentation that explains exactly how the library works, what the stored variables are, and how/what to call them, so i have to infer and guess from example code -- or your instructions.

Thanks for all the work you put into this.

Grant

ArminJo, 2021-01-29:
Thanks Grant 🥇 

I updated the converting instructions and improved comments in examples.
You helped many other people with your suggestions 👍 
Thats the true spirit of community 😀 

Armin

itsjustme0661, 2021-01-30:
Excellent. I'll read it over ( and keep a copy too).

Appreciate your attentiveness.

Grant

closed 2021-02-03 by ArminJo