Arduino-IRremote/Arduino-IRremote #767: Speed of receiver does not match speed of transmitter

MTRobin, 2021-01-20:
### Board
* [X] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP32 board - first check https://github.com/crankyoldgit/IRremoteESP8266
* [ ] Teensy board
* [ ] Other - please specify

### IDE
* [X] Arduino IDE
* [ ] Arduino Web Editor
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [ ] PlatformIO IDE
* [ ] Other - please specify

### Protocol
* [ ] Unknown
* [ ] BoseWave
* [ ] Denon
* [ ] Dish
* [ ] JVC
* [ ] Lego
* [ ] LG
* [X] NEC
* [ ] Panasonic
* [ ] RC5, RC6
* [ ] Samsung
* [ ] Sanyo
* [ ] Sharp
* [ ] Sony
* [ ] Whynter
* [ ] Other - please specify

**Code Block:**
```

<script `src="https://gist.github.com/MTRobin/38f45c1352e9ca6857294524b732aa71.js"></script>`

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] The latest [repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) is used
- [X] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [X] The title of the issue is helpful and relevant 

**I have one Arduino Uno sending a hex value every 30 milliseconds via NEC over IR. I have another Arduino Uno receiving those values and then changing the color of an RGB LED based on what hex value is sent. This is all working fine, except the receiving unit doesn't look like it is changing the LED every 30 milliseconds. It is slower. Is it possible to speed up the turnaround time between receiving the IR command and telling the LED what color to be?

On a similar note, I would like to send faster than 30 milliseconds, but if I go below 25 milliseconds the receiver doesn't show the right values and the protocol becomes unknown.

Thank you,
Maxwell**

ArminJo, 2021-01-20:
The answer to your questions is [here](https://github.com/Arduino-IRremote/Arduino-IRremote/blob/0d4aceb316f6f62f8ddb5d11e30c13f0d970ced9/src/ir_NEC.cpp#L61).

closed 2021-01-20 by ArminJo

MTRobin, 2021-01-20:
Thank you for the info. I changed the NEC_UNIT and NEC_AVERAGE_DURATION and my serial monitor showed a lot more unknown protocol errors.

Can you please advise what values I need to use to get my transmission and reception to happen 30 times per second? If this is even possible?

Thank you,
Maxwell

AnalysIR, 2021-01-20:
Many (most) common IR receiver will/may automatically shut off if you continuously send IR signals.

Thus you should not be expecting a fast data stream (without specialized solutions) outside the scope of IRremote.

An alternative approach is to create a series of LED sequences on the target device and send one short IR signal to kick off that series on the target device. Some of the more complex industry implementations use that approach.

If you have to send continuously, I would suggest only sending 1 signal every 120->150mSecs. Even with that some receivers may turn off automatically assuming it is noise. 

You may be better off creating a very fast decoder yourself to avoid the extra overhead of IRremote when decoding the signal, which I believe only starts after the complete signal is received.

MTRobin, 2021-01-20:
Ok. Thank you!

ArminJo, 2021-01-21:
You can invent your own protocol or try to use e.g. the LG protocol and interprete address and data as 3*8 bit for your colors.

MTRobin, 2021-01-22:
> You can invent your own protocol or try to use e.g. the LG protocol and interpret address and data as 3*8 bit for your colors.

Is the LG protocol faster than the NEC protocol? Currently I am able to receive the NEC 32 bit command and interpret it appropriately. The only issue I am trying to overcome is sending and receiving at least 30x per second.

Thanks.

ArminJo, 2021-01-22:
LG saves 4 bits and has a better parity.
If I were you, I would take LG,  take 250 as unit and use 8* unit as LG_HEADER_MARK and 4* unit as LG_HEADER_SPACE and  change  the line 
`#define RECORD_GAP_MICROS   5000 // Nec header space is 4500`
to
`#define RECORD_GAP_MICROS   1500 // LG header space is 1000`
in IRremoteint.h

Feedback is appreciated...

MTRobin, 2021-01-22:
Thank you for the advice. I changed the LG file and that seemed to speed things up, but I am getting errors showing it is an unknown protocol in the serial monitor.

I was unable to find the line `#define RECORD_GAP_MICROS 5000 // Nec header space is 4500`

I have a file called "IRremote.h" but not one that says "IRremoteint.h"

ArminJo, 2021-01-22:
Did you change the parameters on the receiver side also?

MTRobin, 2021-01-22:
I updated the ir_LG file and the IRremoteint.h file and then reuploaded both sketches to both the send and receive arduinos.

Are there parameters I need to change elsewhere in these files?

ArminJo, 2021-01-22:
Maybe MARK_EXCESS_MICROS, but this depends on your receiver circuit.
Please post the unknown timing output generated by `IrReceiver.printIRResultRawFormatted(&Serial, true)`.

MTRobin, 2021-01-22:
Here is everything I have configured. Thank you very much for your help.

Here is a screenshot of the serial monitor.
[image]

Here is the IR Sender Code
`#include <IRremote.h>

IRsend irsend;
unsigned long tData = 0x000000;

void setup() {
    pinMode(LED_BUILTIN, OUTPUT);
    Serial.begin(115200);
}

void loop() {
    tData = 0x060606;
    irsend.sendLG(tData, 28);

    delay(30);

    tData = 0x000000;
    irsend.sendLG(tData, 28);

    delay(30);

}`

Here is the IR Receive code
`#include <IRremote.h>

int IR_RECEIVE_PIN = 2;
IRrecv IrReceiver(IR_RECEIVE_PIN);
//decode_results IRResult;

int redPin = 5;
int greenPin = 9;
int bluePin = 6;
unsigned long color = 0x000000;

void setup() {
  Serial.begin(115200);
  IrReceiver.enableIRIn();
  IrReceiver.blink13(true);

  pinMode(redPin, OUTPUT);
  pinMode(greenPin, OUTPUT);
  pinMode(bluePin, OUTPUT);
}

void loop() {

    if (IrReceiver.decode()) {
        IrReceiver.printIRResultShort(&Serial);
        if (IrReceiver.decodedIRData.protocol == UNKNOWN) {
            IrReceiver.printIRResultRawFormatted(&Serial, true);
        }
        if (IrReceiver.decodedIRData.protocol == LG) {
          color = IrReceiver.results.value;
        }
        else{
          color = 0x000000;
        }
        analogWrite(redPin, color >> 16);
        analogWrite(greenPin, color >> 8 & 0xFF);
        analogWrite(bluePin, color & 0xFF);
        
        IrReceiver.resume();
    }
[... 373 lines cut ...]

//#define TICKS_LOW(us)   ((int)(((us)*LTOL/MICROS_PER_TICK)))
//#define TICKS_HIGH(us)  ((int)(((us)*UTOL/MICROS_PER_TICK + 1)))
#if MICROS_PER_TICK == 50 && TOLERANCE == 25           // Defaults
#define TICKS_LOW(us)   ((us)/67 )     // (us) / ((MICROS_PER_TICK:50 / LTOL:75 ) * 100)
#define TICKS_HIGH(us)  ((us)/40 + 1)  // (us) / ((MICROS_PER_TICK:50 / UTOL:125) * 100) + 1
#else
    #define TICKS_LOW(us)   ((uint16_t) ((long) (us) * LTOL / (MICROS_PER_TICK * 100) ))
    #define TICKS_HIGH(us)  ((uint16_t) ((long) (us) * UTOL / (MICROS_PER_TICK * 100) + 1))
#endif

//------------------------------------------------------------------------------
// IR receivers on a board with an external output transistor may have "inverted" output
#ifdef IR_INPUT_IS_ACTIVE_HIGH
// IR detector output is active high
#define MARK   1 ///< Sensor output for a mark ("flash")
#define SPACE  0 ///< Sensor output for a space ("gap")
#else
// IR detector output is active low
#define MARK   0 ///< Sensor output for a mark ("flash")
#define SPACE  1 ///< Sensor output for a space ("gap")
#endif

#endif
`

ArminJo, 2021-01-22:
- Please use sendLGRaw() function
- Please edit your post above to show only the changes you made, otherwise support is impossible for me.
- You claim to send 0x060606 but I do not see any long space (LG_ONE_SPACE            (3 * LG_UNIT) for a one bit) in the receiver timing.
- I tested it with IrSendDemo and IRReceive demo on two boards and it worked like a charm for me.
Here is my received timing
```
Protocol=LG Address=0x3 Command=0x45 Raw-Data=0x300459 (28 bits) MSB first
rawData[60]: 
     -1971950
     +2000,-1000     + 250,- 250     + 250,- 250     + 250,- 200
     + 300,- 250     + 250,- 250     + 250,- 250     + 250,- 750
     + 250,- 700     + 250,- 250     + 300,- 200     + 300,- 250
     + 300,- 200     + 250,- 250     + 250,- 250     + 250,- 250
     + 250,- 250     + 250,- 250     + 250,- 700     + 300,- 200
     + 300,- 250     + 250,- 250     + 250,- 700     + 300,- 200
     + 300,- 700     + 300,- 700     + 250,- 250     + 300,- 200
     + 300,- 700     + 300
```

MTRobin, 2021-01-22:
- Can you please explain what you mean by "use sendLGRaw() function"
- I have edited my post.
- The hex value is just for testing purposes now. The plan is to send the hex value of a color and then the receiver will turn an LED that color. Can you please explain the long space I need?
- Are you able to share the code you used so I can test? I am just looking for a simple solution to send a color value in hex to a receiver that then makes an LED that received color.

Thank you for your help and patience.

ArminJo, 2021-01-22:
- get the latest version from the Repo.
- change the 3 lines in ir_LG.cpp as you had done before
- get the IRSendDemo
- in the loop, strip every protocol except LG.
- compile and run it.
- get the IRReceiveDemo
- Compile and run it
- and voila...
- change the RECORD_GAP_MICROS as you done before
- Next modify the delay in send and test it again.
- Now you can try to send your rgb data using 16 bit address and  8 bit data, so you do not have to compute the checksum by yourself. Or use sendRaw and be smart enough to compute and add the checksum.
- Now you have what you want 😀 .

Good Night

MTRobin, 2021-01-23:
Thank you very much for your help. I got it all working except for one part that I am struggling with.

I am sending the following code
`IrSender.sendLGRaw(0x060606);`

I am receiving the following code
`Protocol=LG Address=0x0 Command=0x6060 Parity fail Raw-Data=0x60606 (28 bits) MSB first`

The last step is to get the hex value into a variable. The current command is
`color = IrReceiver.decodedIRData.command;`

and the value I am getting from that is
24672

What is the proper syntax to get the Raw-Data hex value?

Thank you!

ArminJo, 2021-01-23:
Please look in the implementation of the print function in irReceive.cpp, there you can see how to access the Raw-data.

MTRobin, 2021-01-23:
Found it and it works! Thank you very much for your time, patience, and help. It is much appreciated!