Arduino-IRremote/Arduino-IRremote #770: IRremote freezing due to a removed line of code

z3t0, 2021-01-24:
### Board
Unknown

### IDE
Unknown

### Protocol
Unknown

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [] The latest [repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

Hi, I am reporting a bug that I received via e-mail. Sorry, for leaving the other fields above empty, I don't know the answer to many of those. If this is a bug then I think it's a general one that applies to more than one board or protocol.
**The rest of this issue is from the email I received.**

i found bug in your otherwise great library IRremote.
In irReceive.cpp, line 168 in method resume()
In older version of library, there was this piece of code:
irparams.rawlen = 0; // not required
Now it seems, that this piece of code is removed.
This code is required, when two signals are received too soon after each other.
If this line is missing, IRremote library stops working until restart
of processor.

ArminJo, 2021-01-24:
This is a bug in the 2.8.0 version and already fixed in this [commit in src/irReceive.cpp](https://github.com/Arduino-IRremote/Arduino-IRremote/commit/3d27531c8380a5a8c73eac53ad6bb6ac224d6c77) and included in 2.8.1. 
The problem was that overflow was not handled correctly.
Which verision does he use?

z3t0, 2021-01-25:
I'm honestly not sure which version was being used, but I'll close the issue since its been fixed! Thanks!

closed 2021-01-25 by z3t0