Arduino-IRremote/Arduino-IRremote #777: Option / macros  USE_NO_SEND_PWM don't work as expected

DMladenov1975, 2021-01-30:
### Board
* [X] Arduino ATmega328* board (UNO, Nano)

### IDE
* [X] Arduino IDE

### Protocol
* [X] NEC

**checklist:**
- [X] I have **read** the README.md file thoroughly.
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] The latest [repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) is used.
- [X] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [X] I checked, if at least one of the examples was working.

I need to send IR codes without PWM (direct connect to wired IR-input)
When I try to use  #define USE_NO_SEND_PWM in my sketch or in IRremoteInt.h - not working.
But when I replace these lines In irSend.cpp.h:
TIMER_ENABLE_SEND_PWM to digitalWrite(sendPin, LOW) for IRsend::mark
and
TIMER_DISABLE_SEND_PWM to digitalWrite(sendPin, HIGH) for IRsend:space
everything is OK.
Also, when I insert delay(50); after digitalWrite(sendPin, LOW)  at IRsend::mark - also works.

DMladenov1975, 2021-01-31:
The problem is bug in irSend.cpp.h at IRsend::mark function.
My fix is to move the line #endif //  USE_SOFT_SEND_PWM at then bottom of IRsend::mark right after     TIMER_ENABLE_SEND_PWM;

Like that:
....
#elif defined(USE_NO_SEND_PWM)
    (void) timeMicros;
    digitalWrite(sendPin, LOW); // Set output to active low.
#else
    TIMER_ENABLE_SEND_PWM
    ; // Enable pin 3 PWM output
**#endif //  USE_SOFT_SEND_PWM -moved line !**
    setFeedbackLED(true);
.....

ArminJo, 2021-01-31:
Thanks for reporting 🥇

commit fac01904 (2021-01-31): fixed USE_NO_SEND_PWM bug. Closes #777
closed 2021-01-31 by ArminJo