Arduino-IRremote/Arduino-IRremote #783: Changing the IR_SEND_PIN

Niltonsf, 2021-02-02:
Hey @ArminJo 
It's me again, I'm working with your library and it's great. I just have one question, how can I change the PIN that is going to send. I'm using ESP32 at the moment and the GPIO23 from it. I attempted to change only the pinMode but it didn't work:
```
#include <IRremote.h>
#if defined(__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__) || defined(__AVR_ATtiny87__) || defined(__AVR_ATtiny167__)
#include "ATtinySerialOut.h"
#endif

// On the Zero and others we switch explicitly to SerialUSB
#if defined(ARDUINO_ARCH_SAMD)
#define Serial SerialUSB
#endif

void setup() {
    pinMode(23, OUTPUT);

    Serial.begin(115200);
#if defined(__AVR_ATmega32U4__) || defined(SERIAL_USB) || defined(SERIAL_PORT_USBVIRTUAL)  || defined(ARDUINO_attiny3217)
    delay(2000); // To be able to connect Serial monitor after reset or power up and before first printout
#endif
    // Just to know which program is running on my Arduino
    Serial.println(F("START " __FILE__ " from " __DATE__ "\r\nUsing library version " VERSION_IRREMOTE));
    Serial.print(F("Ready to send IR signals at pin "));
    Serial.println(IR_SEND_PIN);

    IrSender.begin(true); // Enable feedback LED,
}

void loop() {
    uint32_t test = 0x88C0051;

    delay(1000);
    
    Serial.println(F("Send LGRaw 0x88C0051 with \"parity error\""));
    IrSender.sendLGRaw(test, 0);
    delay(2000);
    
    delay(5000); // additional delay at the end of each loop
}
```

ArminJo, 2021-02-02:
I have to confess, that the documentation about this was missing in the readme. I added a few lines...

## Hardware specifications
The receiver sample interval is generated by a timer. 
On many boards this must be a hardware timer, on some a software timer is available and used.<br/>
The send PWM signal is generated by a hardware timer. The same timer as for the receiver is used.
Since each hardware timer has its dedicated output pins, you must change timer to change PWN output.<br/>
The timer and the pin usage can be adjusted in [IRremoteBoardDefs.h](src/private/IRremoteBoardDefs.h)

Niltonsf, 2021-02-02:
@ArminJo 
Oh Ok, I'm glad I can keep helping.
I entered in the IRremoteBoardDefs.h, should I change this line:
for example change the IR_USE_TIMER_ESP32 23 (23 represents the pin)
```
/*********************
 * OTHER CPU's
 *********************/
#elif defined(ESP32)
#  if !defined(IR_USE_TIMER_ESP32)
#define IR_USE_TIMER_ESP32
#  endif
```
Will I have to add something to the main page, or I don't need to add nothing to represent that I'm using esp32.

Thnk, Nilton

ArminJo, 2021-02-02:
Search for the esp32 send pin definition `#define IR_SEND_PIN` in IRremoteBoardDefs.h. That schould be all.

Niltonsf, 2021-02-02:
@ArminJo 
I have made some changes, only in the IrRemoteBoardDefs.h:
```
#elif defined(ESP32)
#  if !defined(IR_USE_TIMER_ESP32)
#define IR_USE_TIMER_ESP32 23
#  endif

and

#define IR_SEND_PIN 23
```
But I didn't obtain no success.

ArminJo, 2021-02-02:
> #define IR_SEND_PIN 23

Which line?

Niltonsf, 2021-02-02:
@ArminJo Line 441

ArminJo, 2021-02-02:
Wrong line. Look for defined(ESP32)

Niltonsf, 2021-02-02:
@ArminJo 
There is this line (line 367):
```
/*********************
 * OTHER CPU's
 *********************/
#elif defined(ESP32)
#  if !defined(IR_USE_TIMER_ESP32)
#define IR_USE_TIMER_ESP32 
#  endif
```
and this part (line 145):
```
#define USE_SOFT_SEND_PWM
// Define to use spin wait instead of delayMicros()
//#define USE_SPIN_WAIT
// Supply own enableIRIn()
#undef USE_DEFAULT_ENABLE_IR_IN

#elif defined(ESP32)
// No system LED on ESP32, disable blinking by NOT defining BLINKLED

// Supply own enableIRIn() and enableIROut()
#undef USE_DEFAULT_ENABLE_IR_IN
#undef USE_DEFAULT_ENABLE_IR_OUT

#elif defined(PARTICLE)
````

ArminJo, 2021-02-03:
There are more lines for ESP32. 
What do you think is `#define IR_USE_TIMER_ESP32` for?

Niltonsf, 2021-02-03:
@ArminJo 
Is it for setting the pin from arduino?
I have attempted to set:
```
#define IR_USE_TIMER_ESP32 23
```

ArminJo, 2021-02-03:
Wrong answer...

Niltonsf, 2021-02-03:
@ArminJo 
I'm not suppose to determine any timers...

ArminJo, 2021-02-03:
Hint: search `IR_USE_TIMER_ESP32`  in the file

Niltonsf, 2021-02-03:
@ArminJo 
I have found this:
```
// ESP32 has it own timer API and does not use these macros, but to avoid ifdef'ing
// them out in the common code, they are defined to no-op. This allows the code to compile
// (which it wouldn't otherwise) but irsend will not work until ESP32 specific code is written
//
// The timer code is in the esp32.cpp file
//
// An IRremote version for ESP8266 and ESP32 is available at https://github.com/crankyoldgit/IRremoteESP8266
#elif defined(IR_USE_TIMER_ESP32)

#if ! defined(IR_SEND_PIN)
#define IR_SEND_PIN 23 // can use any pin, no timer restrictions
#endif

#if ! defined(LED_CHANNEL)
#define LED_CHANNEL 0 // The channel used for PWM 0 to 7 are high speed PWM channels
#endif
````

ArminJo, 2021-02-03:
This is the right line 👍 I just checked it with my hardware and it worked as expected with the IRSendDemo example!
So start with a fresh library (mabe even a new download from repo) and do only this change and you will have success!

Niltonsf, 2021-02-03:
@ArminJo 
So strange. I have donwloaded the library again and fresh installed it. Made only that change on yout library, But I sill don't get any success with the sendLGRaw code.

ArminJo, 2021-02-03:
Is the pin 4 working, maybe your pins are numbered wrong or are defect. Test a few other pins. 
I am sorry but I cant help you furher, it worked for my ESP32 Devkit V1 with pin 4 and 23!

Niltonsf, 2021-02-03:
@ArminJo 
I have the same DOIT ESP32 DeVKIT V1, pin 4 is D4 and D23 is pin 23 right?

ArminJo, 2021-02-03:
Yes

Niltonsf, 2021-02-03:
@ArminJo 
It just doens't work, so strange. I grabbed my phone camera to see if it was sending but it wasn't. I made a sketch to see if the ir sender was broke but it turned on.

Niltonsf, 2021-02-03:
@ArminJo Funny part is thath if I use Serial.println(IR_SEND_PIN) I get 4 as the result.

ArminJo, 2021-02-03:
Good discovery 👍 You are likely working with anohter version of the library. 
Remove all IRremote versions until you get an error no library or so and then install again.

Niltonsf, 2021-02-03:
@ArminJo 
I just don't know what was happening hahah, I uninstalled it 3 times but on this last time it worked. Thx, great Work you have created.

closed 2021-02-03 by Niltonsf