Arduino-IRremote/Arduino-IRremote #789: ir codes when reading have changed

yura371, 2021-02-09:
### Board
* [] Arduino ATmega328* board (UNO, Nano)

### IDE
* [ ] Arduino IDE

### Protocol
* [ ] Samsung

### Current behavior

Good day. After upgrading to 3.0.0. codes when reading have changed. For example, the Samsung code 0xE0E0E01F became 0xF8070707 (0xE0E0D02F-> 0xF40B0707; 0xE0E0F00F-> 0xF00F0707). This is normal?

ArminJo, 2021-02-09:
Yes this is intended! Samsung is now LSB first, which is standard. Just do a bit reverse of the old codes and you have the new ones.
To get the old codes you must use version 2.x or replace the current 3.0.x version with the content from the repo and activate line 54 in IRremoteInt.h.

yura371, 2021-02-09:
This also applies to other codes and protocols.

closed 2021-02-09 by ArminJo